node-hid/node-hid #90: install.sh fails with node v0.12.0

alicee, 2015-02-24:
I'm trying to build a module that depends on node-hdi, but get the following errors on `sh install.sh` for node-hdi - getting various warnings and ultimately failing with his output: https://gist.github.com/alicee/9ef6fc81372391439661

And here's npm-debug.log: 
https://gist.github.com/alicee/3b5033147f5d0b3d6fd0

This occurs when using nodejs v0.12.0 - I didn't have any problems using v0.10.

weisjohn, 2015-02-26:
Here's a more concise gist: https://gist.github.com/weisjohn/701c0b6fecfa8df870b4

MylesCaley, 2015-03-04:
+1... if you run arch linux on the raspberry pi 2 (armv7 architecture), the default package in pacman is now v12 node.

martijndevrieze, 2015-03-04:
+1 for just generally running Arch. default package indeed is v12. I have the exact same issue as alicee, with the same module I am trying to get back to functioning.

nomasteryoda, 2015-03-07:
Here's my npm-debug.log in the hope that it will help solve this.
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'install', 'msi-keyboard' ]
2 info using npm@2.7.0
3 info using node@v0.12.0
4 silly cache add args [ 'msi-keyboard', null ]
5 verbose cache add spec msi-keyboard
6 silly cache add parsed spec { raw: 'msi-keyboard',
6 silly cache add   scope: null,
6 silly cache add   name: 'msi-keyboard',
6 silly cache add   rawSpec: '',
6 silly cache add   spec: '*',
6 silly cache add   type: 'range' }
7 verbose addNamed msi-keyboard@*
8 silly addNamed semver.valid null
9 silly addNamed semver.validRange *
10 silly addNameRange { name: 'msi-keyboard', range: '*', hasData: false }
11 silly mapToRegistry name msi-keyboard
12 silly mapToRegistry using default registry
13 silly mapToRegistry registry https://registry.npmjs.org/
14 silly mapToRegistry uri https://registry.npmjs.org/msi-keyboard
15 verbose addNameRange registry:https://registry.npmjs.org/msi-keyboard not in flight; fetching
16 verbose request uri https://registry.npmjs.org/msi-keyboard
17 verbose request no auth needed
18 info attempt registry request try #1 at 22:00:04
19 verbose request id 421d72817be9268f
20 verbose etag "D0S3SZP3UTKBFF3KEB75XT0B"
21 http request GET https://registry.npmjs.org/msi-keyboard
22 http 304 https://registry.npmjs.org/msi-keyboard
23 silly get cb [ 304,
23 silly get   { date: 'Sat, 07 Mar 2015 03:00:31 GMT',
23 silly get     server: 'Apache',
23 silly get     via: '1.1 varnish',
23 silly get     'last-modified': 'Sat, 07 Mar 2015 03:00:31 GMT',
23 silly get     'cache-control': 'max-age=60',
23 silly get     etag: '"D0S3SZP3UTKBFF3KEB75XT0B"',
23 silly get     age: '0',
23 silly get     'x-served-by': 'cache-iad2144-IAD',
23 silly get     'x-cache': 'MISS',
23 silly get     'x-cache-hits': '0',
23 silly get     'x-timer': 'S1425697231.174996,VS0,VE56',
23 silly get     vary: 'Accept',
23 silly get     'content-length': '0',
23 silly get     'keep-alive': 'timeout=10, max=50',
23 silly get     connection: 'Keep-Alive' } ]
24 verbose etag https://registry.npmjs.org/msi-keyboard from cache
25 verbose get saving msi-keyboard to /root/.npm/registry.npmjs.org/msi-keyboard/.cache.json
26 silly addNameRange number 2 { name: 'msi-keyboard', range: '*', hasData: true }
27 silly addNameRange versions [ 'msi-keyboard', [ '0.0.1', '0.0.2' ] ]
28 verbose addNamed msi-keyboard@0.0.2
29 silly addNamed semver.valid 0.0.2
30 silly addNamed semver.validRange 0.0.2
31 silly cache afterAdd msi-keyboard@0.0.2
32 verbose afterAdd /root/.npm/msi-keyboard/0.0.2/package/package.json not in flight; writing
33 verbose afterAdd /root/.npm/msi-keyboard/0.0.2/package/package.json written
34 silly install resolved [ { name: 'msi-keyboard',
34 silly install resolved     description: 'MSI Keyboard LED Controller',
34 silly install resolved     version: '0.0.2',
34 silly install resolved     homepage: 'http://github.com/wearefractal/msi-keyboard',
34 silly install resolved     repository:
34 silly install resolved      { type: 'git',
34 silly install resolved        url: 'git://github.com/wearefractal/msi-keyboard.git' },
34 silly install resolved     author:
34 silly install resolved      { name: 'Fractal',
34 silly install resolved        email: 'contact@wearefractal.com',
34 silly install resolved        url: 'http://wearefractal.com/' },
34 silly install resolved     main: './index.js',
34 silly install resolved     dependencies: { 'node-hid': '*' },
34 silly install resolved     devDependencies: { mocha: '*', should: '*' },
34 silly install resolved     scripts: { test: 'mocha' },
[... 210 lines cut ...]
150 error Tell the author that this fails on your system:
150 error     sh install.sh
150 error You can get their info via:
150 error     npm owner ls node-hid
150 error There is likely additional logging output above.
151 verbose exit [ 1, true ]
152 verbose unbuild node_modules/msi-keyboard/node_modules/node-hid
153 info preuninstall node-hid@0.3.2
154 info uninstall node-hid@0.3.2
155 verbose unbuild rmStuff node-hid@0.3.2 from /home/joe/node_modules
156 verbose unbuild rmStuff in /home/joe/node_modules/msi-keyboard/node_modules
157 info postuninstall node-hid@0.3.2
158 verbose gentlyRm vacuuming /home/joe/node_modules/msi-keyboard/node_modules/node-hid
159 silly gentlyRm purging /home/joe/node_modules/msi-keyboard/node_modules/node-hid
160 silly gentlyRm removing /home/joe/node_modules/msi-keyboard/node_modules
161 silly gentlyRm quitting because other entries in /home/joe/node_modules/msi-keyboard
162 verbose unbuild node_modules/msi-keyboard
163 info preuninstall msi-keyboard@0.0.2
164 info uninstall msi-keyboard@0.0.2
165 verbose unbuild rmStuff msi-keyboard@0.0.2 from /home/joe/node_modules
166 info postuninstall msi-keyboard@0.0.2
167 verbose gentlyRm vacuuming /home/joe/node_modules/msi-keyboard
168 silly gentlyRm purging /home/joe/node_modules/msi-keyboard
169 silly gentlyRm removing /home/joe/node_modules
170 silly gentlyRm finished vacuuming up to /home/joe

rjsteinert, 2015-03-08:
There's a pull request working on 0.11.x, not yet 0.12.x https://github.com/node-hid/node-hid/pull/69

farin, 2015-03-16:
error from console output
```
/home/farin/.node-gyp/0.12.0/deps/v8/include/v8.h:167:37: error: cannot convert ‘v8::Primitive*’ to ‘v8::Object* volatile’ in assignment
     *(static_cast<T* volatile*>(0)) = static_cast<S*>(0);      \
                                     ^
/home/farin/.node-gyp/0.12.0/deps/v8/include/v8.h:596:5: note: in expansion of macro ‘TYPE_CHECK’
     TYPE_CHECK(O, Primitive);
```

and inportant part od log
gyp ERR! System Linux 3.16.0-31-generic
gyp ERR! command "node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "configure" "build" "install"
gyp ERR! cwd /media/farin/storage/s/IPG-UI/node_modules/node-hid
gyp ERR! node -v v0.12.0
gyp ERR! node-gyp -v v1.0.2
gyp ERR! not ok 
npm ERR! Linux 3.16.0-31-generic
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "install"
npm ERR! node v0.12.0
npm ERR! npm  v2.5.1
npm ERR! code ELIFECYCLE

npm ERR! node-hid@0.3.2 install: `sh install.sh`
npm ERR! Exit status 1

natevw, 2015-04-03:
On OS X, `brew info node` showed that the latest 0.10.x node I had was .36, so I used `brew switch node 0.10.36` to downgrade.

Still a pain — does anyone know if using HID devices through the `usb` module would be simple enough? Now libusb makes it sounds like [it is a pain](https://github.com/libusb/libusb/wiki/FAQ#Does_libusb_support_HID_devices) on most platforms which was not the case with hidapi.

marcooliveira, 2015-04-11:
OS X here as well on `0.12.2`:

```
> node-hid@0.3.2 install .............node_modules/node-hid
> sh install.sh

  CC(target) Release/obj.target/hidapi/hidapi/mac/hid.o
../hidapi/mac/hid.c:255:20: warning: comparison of integers of different signs: 'CFIndex' (aka 'long') and 'size_t' (aka 'unsigned long') [-Wsign-compare]
                if (chars_copied == len)
                    ~~~~~~~~~~~~ ^  ~~~
../hidapi/mac/hid.c:295:20: warning: comparison of integers of different signs: 'CFIndex' (aka 'long') and 'size_t' (aka 'unsigned long') [-Wsign-compare]
                if (used_buf_len == len)
                    ~~~~~~~~~~~~ ^  ~~~
2 warnings generated.
  LIBTOOL-STATIC Release/hidapi.a
  CXX(target) Release/obj.target/HID/src/HID.o
../src/HID.cc:50:79: error: no member named 'New' in 'v8::String'
  virtual Handle<Value> asV8Exception() const { return ThrowException(String::New(message().c_str())); }
                                                                      ~~~~~~~~^
../src/HID.cc:57:12: error: expected class name
  : public ObjectWrap
           ^
../src/HID.cc:61:38: error: unknown type name 'Arguments'; did you mean 'v8::internal::Arguments'?
  static Handle<Value> devices(const Arguments& args);
                                     ^~~~~~~~~
                                     v8::internal::Arguments
/Users/foo/.node-gyp/0.12.2/deps/v8/include/v8.h:127:7: note: 'v8::internal::Arguments' declared here
class Arguments;
      ^
../src/HID.cc:76:34: error: unknown type name 'Arguments'; did you mean 'v8::internal::Arguments'?
  static Handle<Value> New(const Arguments& args);
                                 ^~~~~~~~~
                                 v8::internal::Arguments
/Users/foo/.node-gyp/0.12.2/deps/v8/include/v8.h:127:7: note: 'v8::internal::Arguments' declared here
class Arguments;
      ^
../src/HID.cc:77:35: error: unknown type name 'Arguments'; did you mean 'v8::internal::Arguments'?
  static Handle<Value> read(const Arguments& args);
                                  ^~~~~~~~~
                                  v8::internal::Arguments
/Users/foo/.node-gyp/0.12.2/deps/v8/include/v8.h:127:7: note: 'v8::internal::Arguments' declared here
class Arguments;
      ^
../src/HID.cc:78:36: error: unknown type name 'Arguments'; did you mean 'v8::internal::Arguments'?
  static Handle<Value> write(const Arguments& args);
                                   ^~~~~~~~~
                                   v8::internal::Arguments
/Users/foo/.node-gyp/0.12.2/deps/v8/include/v8.h:127:7: note: 'v8::internal::Arguments' declared here
class Arguments;
      ^
../src/HID.cc:79:36: error: unknown type name 'Arguments'; did you mean 'v8::internal::Arguments'?
  static Handle<Value> close(const Arguments& args);
                                   ^~~~~~~~~
                                   v8::internal::Arguments
/Users/foo/.node-gyp/0.12.2/deps/v8/include/v8.h:127:7: note: 'v8::internal::Arguments' declared here
class Arguments;
      ^
../src/HID.cc:80:45: error: unknown type name 'Arguments'; did you mean 'v8::internal::Arguments'?
  static Handle<Value> setNonBlocking(const Arguments& args);
                                            ^~~~~~~~~
                                            v8::internal::Arguments
/Users/foo/.node-gyp/0.12.2/deps/v8/include/v8.h:127:7: note: 'v8::internal::Arguments' declared here
class Arguments;
      ^
../src/HID.cc:81:47: error: unknown type name 'Arguments'; did you mean 'v8::internal::Arguments'?
  static Handle<Value> getFeatureReport(const Arguments& args);
                                              ^~~~~~~~~
                                              v8::internal::Arguments
/Users/foo/.node-gyp/0.12.2/deps/v8/include/v8.h:127:7: note: 'v8::internal::Arguments' declared here
class Arguments;
[... 31 lines cut ...]
      Context::GetCurrent()->Global()->Get(String::New("Buffer") )
      ~~~~~~~~~^
../src/HID.cc:200:44: error: no member named 'New' in 'v8::String'; did you mean simply 'New'?
      Context::GetCurrent()->Global()->Get(String::New("Buffer") )
                                           ^~~~~~~~~~~
                                           New
../src/HID.cc:76:24: note: 'New' declared here
  static Handle<Value> New(const Arguments& args);
                       ^
../src/HID.cc:200:56: error: reference to type 'const v8::internal::Arguments' could not bind to an lvalue of type 'const char [7]'
      Context::GetCurrent()->Global()->Get(String::New("Buffer") )
                                                       ^~~~~~~~
../src/HID.cc:76:45: note: passing argument to parameter 'args' here
  static Handle<Value> New(const Arguments& args);
                                            ^
../src/HID.cc:203:68: error: too few arguments to function call, expected 2, have 1
    Handle<Value> nodeBufferArgs[1] = { Integer::New(message.size()) };
                                        ~~~~~~~~~~~~               ^
/Users/foo/.node-gyp/0.12.2/deps/v8/include/v8.h:2012:3: note: 'New' declared here
  static Local<Integer> New(Isolate* isolate, int32_t value);
  ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make: *** [Release/obj.target/HID/src/HID.o] Error 1
```

MylesCaley, 2015-04-11:
@marcooliveira a bit off topic here, but you can use nvm on OSX to install an older version of node if you wish. a nice walkthrough here: http://flnkr.com/2014/11/install-nvm-and-node-js-on-os-x/

rowanc, 2015-04-11:
Also seeing the same error with `v0.12.2` of node running on `3.13.0-24-generic #47-Ubuntu SMP Fri May 2 23:30:00 UTC 2014 x86_64 x86_64 x86_64 GNU/Linux`

marcooliveira, 2015-04-11:
@MylesCaley thanks. I've been using `nave` to downgrade to `0.10` and go around the issue, but ideally this would be fixed on `0.12` :smile:

skilbjo, 2015-04-28:
+1, will be following this following

nomasteryoda, 2015-04-30:
Still tracking as my keyboard is stuck white... needs node-hid and node-gyp
to make msi-keyboard nodejs work.

edasque, 2015-05-12:
+1, also an issue for me.

bleiddyn, 2015-05-12:
For the person who asked about using libusb, I've used that in other projects in OS X in the past and it entails removing the device you want to connect to from the HID subsystem (You have to write a dummy kext for that) and until you undo that the device you were looking for is of course not available to the normal HID subsystem. HIDAPI is probably still the best idea.

Pecacheu, 2015-05-19:
Are the developers working on this issue? I can't find a single gamepad API for Node.JS that doesn't use this API, forcing me to use the HTML5 Gamepad API, which is non-functional in Chrome or Safari on Mac OS X (works in Firefox, though), and iff-y on Windows.

marcooliveira, 2015-05-20:
Has the project been abandoned?

hanshuebner, 2015-05-20:
I am not maintaining it right now.  Patches are accepted, as are volunteers who use it and would like to maintain it.

natevw, 2015-05-20:
I'd be interested in helping to maintain it, as a off-and-on client project needs it and constantly finding it broken has been a big liability, but currently overcommitted on other modules already. I do already have a fork that simplifies the build process so it doesn't break under atom-shell/Electron, but haven't had the cycles to catch the actual bindings up with 0.12.x and don't want to make any promises.

marcooliveira, 2015-05-20:
@natevw would be awesome if you would fix that! Do you accept payments in beer? haha :smile:

hanshuebner, 2015-05-20:
Just let me know if anyone wants to become a member of the node-hid
organization.  I have not used the library in over two years and will not
spend any significant amount of time with it in the near future, so please
don't expect me to fix problems that you have.

marcooliveira, 2015-05-20:
@hanshuebner ok! It's currently impossible for me to work on this, but hopefully someone will come forward. I'm betting on @natevw there! Hehe

Cheers for the work you' already done, @hanshuebner! If it serves any good, I've used `node-hid` for some cool projects. Here's a talk I did with a drone & gamepad hack https://www.youtube.com/watch?v=qU40CJg27kI

edasque, 2015-05-20:
So many node.js packages depend on node-hid and so many other packages depend on these packages, the community will clamor for this to be fixed

nomasteryoda, 2015-05-20:
I'm clamoring... my white keyboard back lights are burning my eyes out at
night... I so miss the ability to change it to another color and the nodejs
msi-keyboard was how I did that.

natevw, 2015-05-21:
Please see if https://github.com/node-hid/node-hid/pull/95 works for you (see my comment with link to package tarball that may be necessary if you don't want to manually build my branch via instructions in its README). Somewhat surprisingly, there was much to fix for whatever reason beyond the build simplification I'd already done and another contributor's patch to the gyp file.

natevw, 2015-05-21:
@hanshuebner I'd be happy to help maintain this at least for a time, if you could add me as a collaborator here and on npm. I'm no USB expert, nor can I promise this module will always be useful to me either, but it's an interest given my embedded/hardware direction lately, and at least for a compatibility release and a bit of the issues backlog here I'd like to help out.

hanshuebner, 2015-05-22:
@natevw Cool, thanks!  I've sent you an invite.  Feel free to merge
anything that you think may improve the current situation.  If you have
specific questions regarding the code, don't hesitate to contact me.

natevw, 2015-05-26:
@hanshuebner Thanks, I've got my recent PR merged and the package.json version updated. Can you `npm publish`, or `npm owner add` me ("natevw") to the package there too?

hanshuebner, 2015-05-26:
@natevw I've made you an owner of the npm package, thank you for caring!

natevw, 2015-05-26:
@hanshuebner Thank you for sharing :-)

@all — published as `node-hid@0.4.0`, please let me know if anyone has further trouble, but going to mark this one as fixed by https://github.com/node-hid/node-hid/pull/95.

closed 2015-05-26 by natevw

marcooliveira, 2015-05-26:
Woohoo! You deserve a beer, @natevw! :beer: Let me know if you ever come to Portugal! Hehe. And thanks @hanshuebner for being OS friendly!

MylesCaley, 2015-05-27:
Thanks @natevw . I work for GE FirstBuild and we have a [“hacker” module](http://market.firstbuild.com/products/greenbean) that plugs into a lot of GE Appliances on one end and shows up as an HID device on the other. We have been forcing our customers to use node 0.10.x which has been an increasing problem with most distros upgrading their node release to 0.12.x. Thanks @hanshuebner for creating this originally.