node-hid/node-hid #93: stream.Duplex

chrisdewar, 2015-04-12:
Is there any technical reason this hasn't been implemented as a `stream.Duplex`? The first thing I wanted to do after downloading this module was `device.pipe(protocol).pipe(application)`, just wondering if there are any technical reasons why this isn't the case before I take a crack at converting it over.

My use case:
```javascript
var hid = require('node-hid'),
    Protocol = require('./Protocol'),
    protocol = module.exports = new Protocol(),
    devices = hid.devices();

devices
    .filter(function (device) {
        // filter devices down to those that match our interest
        return device.vendorId === Protocol.VENDOR
            && device.productId === Protocol.PRODUCT;
    })
    .map(function (device) {
        // map HID event emitter objects over devices
        return new hid.HID(device.path);
    })
    .forEach(function (device) {
        // pipe device data into protocol transformer
        device.pipe(protocol);
    });
```

todbot, 2015-10-15:
USB HID is fundamentally a packet protocol (of multiple types, called "reports" in USB parlance) with no clean mapping to a transmit/receive stream, so I think that's what that informed the architecture.

Have any good docs on implementing `stream.Duplex`? If I can add it without distorting the current API, I'll definitely look into it.

chrisdewar, 2015-12-01:
This was a long time ago now, but if I remember correctly, the API exposes the "reports" as events on an EventEmitter. But all the events on this emitter (write, read, data, pause, resume) already exist on a duplex stream (which is itself an EventEmitter).

If I had to guess, the API would change very little. The only difference would be that instead of implementing these things yourself, you use the native features of the stream, and glue those up to the functionality of the library.

Docs for duplex streams are here: https://nodejs.org/api/stream.html#stream_class_stream_duplex_1

It enables users to pipe the input from a usb directly into a file via `fs.createWriteStream` for example, or to `stdout` via `process.stdout`. In my case, I wanted to `pipe` the data from the USB through a `stream.Transform` such that I could convert it into a more useful format before `pipe`ing it into something that could consume that data.

bathos, 2017-02-22:
Just ran into this too. I’m glad I tried to pipe and learned about it though! HID replicates a lot of the stream interface —  `read`, `write`, `close`, etc — so it’s natural to assume it’s an actual stream if you don’t look at the source, yet that assumption might have led to subtle bugs down the line if I had expected HID to have the usual behaviors and semantics of a stream.

> USB HID is fundamentally a packet protocol (of multiple types, called "reports" in USB parlance) with no clean mapping to a transmit/receive stream, so I think that's what that informed the architecture.

I’m probably not reading that correctly, but if you mean it’s not a ‘continuous’ byte stream, node streams have "object mode" for atomic read/write units which are any arbitrary thing (as opposed to freely concat-able buffers or strings).

A duplex/transform stream is the natural interface for this in nodeland imo, but if this is undesirable, would throwing a warning on the readme that it’s not actually the node stream API be cool for a PR?

todbot, 2017-02-22:
Hi @bathos,
Yes, if you want to do a PR with the README warning, that would be awesome.  I know nothing of stream.Duplex yet but I'd be amenable to adding it. node-hid's was written before existed I think and I have no objection to adding to its API to make it work in a more modern way.

bathos, 2017-02-23:
I opened a PR for that. I have also been experimenting locally with a Duplex-based API and if this proves successful I will share the results, but it would represent significant breaking changes.

bminer, 2017-04-06:
I'm all for improving docs, but maybe we should change node-hid to implement the Duplex stream?