node-hid/node-hid #94: Symbol not found _IOHIDManagerCreate

masterakos, 2015-04-14:
Cloned the project then used npm install to execute all installation steps:
```npm install``` output
```
> node-hid@0.3.2 preinstall /Users/Tsalik/Projects/hidTest/node_modules/node-hid
> sh get-hidapi.sh

Cloning into 'hidapi'...
remote: Counting objects: 1965, done.
remote: Total 1965 (delta 0), reused 0 (delta 0), pack-reused 1965
Receiving objects: 100% (1965/1965), 2.79 MiB | 155.00 KiB/s, done.
Resolving deltas: 100% (1082/1082), done.
Checking connectivity... done.
\
> node-hid@0.3.2 install /Users/Tsalik/Projects/hidTest/node_modules/node-hid
> sh install.sh

  CC(target) Release/obj.target/hidapi/hidapi/mac/hid.o
../hidapi/mac/hid.c:255:20: warning: comparison of integers of different signs: 'CFIndex' (aka 'long') and 'size_t' (aka 'unsigned long') [-Wsign-compare]
                if (chars_copied == len)
                    ~~~~~~~~~~~~ ^  ~~~
../hidapi/mac/hid.c:295:20: warning: comparison of integers of different signs: 'CFIndex' (aka 'long') and 'size_t' (aka 'unsigned long') [-Wsign-compare]
                if (used_buf_len == len)
                    ~~~~~~~~~~~~ ^  ~~~
2 warnings generated.
  LIBTOOL-STATIC Release/hidapi.a
  CXX(target) Release/obj.target/HID/src/HID.o
  SOLINK_MODULE(target) Release/HID.node
  SOLINK_MODULE(target) Release/HID.node: Finished
0.12.2
nan@1.7.0 node_modules/nan
```
Then tried simply to ```node src/show-devices.js```
Got the following:
```
dyld: lazy symbol binding failed: Symbol not found: _IOHIDManagerCreate
  Referenced from: /Users/Tsalik/Projects/hidTest/node_modules/node-hid/build/Release/HID.node
  Expected in: dynamic lookup

dyld: Symbol not found: _IOHIDManagerCreate
  Referenced from: /Users/Tsalik/Projects/hidTest/node_modules/node-hid/build/Release/HID.node
  Expected in: dynamic lookup

Trace/BPT trap: 5
```

sw_vers -productVersion  ```10.9.5```
node -v  ```v0.12.2```

Also to mention that using npm repository I can't even install the module.
That's the ```npm install``` output:
```
> node-hid@0.3.2 install /Users/Tsalik/Projects/hidTest/node_modules/node-hid
> sh install.sh

  CC(target) Release/obj.target/hidapi/hidapi/mac/hid.o
../hidapi/mac/hid.c:255:20: warning: comparison of integers of different signs: 'CFIndex' (aka 'long') and 'size_t' (aka 'unsigned long') [-Wsign-compare]
                if (chars_copied == len)
                    ~~~~~~~~~~~~ ^  ~~~
../hidapi/mac/hid.c:295:20: warning: comparison of integers of different signs: 'CFIndex' (aka 'long') and 'size_t' (aka 'unsigned long') [-Wsign-compare]
                if (used_buf_len == len)
                    ~~~~~~~~~~~~ ^  ~~~
2 warnings generated.
  LIBTOOL-STATIC Release/hidapi.a
  CXX(target) Release/obj.target/HID/src/HID.o
../src/HID.cc:50:79: error: no member named 'New' in 'v8::String'
  virtual Handle<Value> asV8Exception() const { return ThrowException(String::New(message().c_str())); }
                                                                      ~~~~~~~~^
../src/HID.cc:57:12: error: expected class name
  : public ObjectWrap
           ^
[... 287 lines cut ...]
87 error node v0.12.2
88 error npm  v2.7.4
89 error code ELIFECYCLE
90 error node-hid@0.3.2 install: `sh install.sh`
90 error Exit status 1
91 error Failed at the node-hid@0.3.2 install script 'sh install.sh'.
91 error This is most likely a problem with the node-hid package,
91 error not with npm itself.
91 error Tell the author that this fails on your system:
91 error     sh install.sh
91 error You can get their info via:
91 error     npm owner ls node-hid
91 error There is likely additional logging output above.
92 verbose exit [ 1, true ]
93 verbose unbuild node_modules/node-hid
94 info preuninstall node-hid@0.3.2
95 info uninstall node-hid@0.3.2
96 verbose unbuild rmStuff node-hid@0.3.2 from /Users/Tsalik/Projects/hidTest/node_modules
97 info postuninstall node-hid@0.3.2
98 silly gentlyRm /Users/Tsalik/Projects/hidTest/node_modules/node-hid is being purged from base /Users/Tsalik/Projects/hidTest
99 verbose gentlyRm don't care about contents; nuking /Users/Tsalik/Projects/hidTest/node_modules/node-hid
100 silly vacuum-fs purging /Users/Tsalik/Projects/hidTest/node_modules/node-hid
101 silly vacuum-fs removing /Users/Tsalik/Projects/hidTest/node_modules
102 silly vacuum-fs finished vacuuming up to /Users/Tsalik/Projects/hidTest
```

ril3y, 2015-05-05:
Bump on this.  I have the exact same issue.

holocron, 2015-05-09:
same issue under mac os x both 10.9 and 10.10. looks like IOKit lookup problem

masterakos, 2015-05-16:
Indeed, problem solved by editing binding.gyp before ```npm install```. Thank you @holocron (Y)

natevw, 2015-05-21:
I included https://github.com/masterakos/node-hid/commit/fea83535e1d65bfd296f7c9558626ab19d29c53a in https://github.com/node-hid/node-hid/pull/95, works a treat!

natevw, 2015-05-26:
This should be fixed via #95 as of node-hid@0.4.0.

closed 2015-05-26 by natevw