node-hid/node-hid #96: Segmentation fault when there's an error opening HID device

bminer, 2015-05-28:
node-hid 0.4.0 crashes the Node process with a segmentation fault when trying to open a HID device without the appropriate permissions (i.e. there is an error opening the device).  The correct behavior is to throw an exception (as per the README).

This bug is confirmed in node-hid@0.4.0.  This bug is not present in node-hid@0.3.2.  Version 0.3.2 works as expected.

Sample node-hid 0.3.2 output:

```
cannot open device with path 0002:0004:00
```

Sample node-hid 0.4.0 output:

```
Segmentation fault
```

natevw, 2015-05-28:
Are you looking into the cause? If I had to hazard a guess, perhaps it's just a bug in bridging the error back to JS.

bminer, 2015-05-28:
@natevw - Haven't taken a good look yet, but I'd imagine you're right.  Do you have any pointers as to what changed between these versions?

natevw, 2015-06-01:
Sorry this is kind of broad (and a messy set of "experimental" commits to test the new build process) but AFAIK the only changes were all the ones here: https://github.com/node-hid/node-hid/pull/95, mostly 3cc5b0a / 6252525 / fea8353 were the ones changing the actual library code but don't take my word for it ;-)

First thing I usually do in cases like this are fire up a crashing script under gdb/lldb (`lldb node script.js` + `run`…there's probably a one-liner way…) and get a `bt` after it crashes.

ForestMist, 2015-06-02:
Not sure if related but I sometimes get segmentation fault when using device.close() with version 0.4.0.

bminer, 2015-06-03:
@natevw - Based on those commits that you listed, I don't see anything popping out as an issue... might be something different about hid-api?

todbot, 2015-10-12:
I think I fixed this issue with commit b8e7736c0d237830bd26915935cf3bf5a9c7f34e.  If anyone having this problem can test, let me know. Thanks!

bminer, 2015-11-24:
This has been fixed.  Confirmed running node-hid 0.5.1.

closed 2015-11-24 by bminer

natevw, 2015-11-24:
Awesome, thanks @todbot for fixing and @bminer for confirming!