node-hid/node-hid #97: node-hid not showing all usb devices

magic890, 2015-06-09:
I'm using `node-hid` with this call:
   
    HID.devices()

and I should get a list of all USB connected devices as [listed here in the guide](https://github.com/node-hid/node-hid#get-a-list-of-all-hid-devices-in-the-system).

As result I'm getting this devices list:

    [ { vendorId: 1452,
      productId: 33346,
      path: 'USB_05ac_8242_6500000',
      serialNumber: '',
      manufacturer: 'Apple Computer, Inc.',
      product: 'Apple IR',
      release: 22,
      interface: -1 },
    { vendorId: 1452,
      productId: 567,
      path: 'USB_05ac_0237_6300000',
      serialNumber: '',
      manufacturer: 'Apple Inc.',
      product: 'Apple Internal Keyboard / Trackpad',
      release: 144,
      interface: -1 },
    { vendorId: 1452,
      productId: 567,
      path: 'USB_05ac_0237_6300000',
      serialNumber: '',
      manufacturer: 'Apple Inc.',
      product: 'Apple Internal Keyboard / Trackpad',
      release: 144,
      interface: -1 },
    { vendorId: 0,
      productId: 0,
      path: '',
      serialNumber: '',
      manufacturer: '',
      product: 'Apple Mikey HID Driver',
      release: 0,
      interface: -1 },
    { vendorId: 1452,
      productId: 567,
      path: 'USB_05ac_0237_6300000',
      serialNumber: '',
      manufacturer: 'Apple Inc.',
      product: 'Apple Internal Keyboard / Trackpad',
      release: 144,
      interface: -1 },
    { vendorId: 1452,
      productId: 567,
      path: 'USB_05ac_0237_6300000',
      serialNumber: '',
      manufacturer: 'Apple Inc.',
      product: 'Apple Internal Keyboard / Trackpad',
      release: 144,
      interface: -1 } ]

which does not include the ACR122U-A9 device (NFC/RFID device provided by ACS Ltd) connected through USB to my Mac.

All the peripherals connected to the Mac are listed under `System Report -> Hardware -> USB`, here the screenshot: ![ACR122U connected to the Mac][1]

Can you figure out why is this happening and how to solve this issue?

  [1]: http://i.stack.imgur.com/AUL8Z.png

slooker, 2015-06-16:
I've got the same issue.  

[ { vendorId: 1133,
    productId: 16388,
    path: 'Unifying_046d_4004_14100300',
    serialNumber: '',
    manufacturer: 'Logitech',
    product: 'Unifying Device',
    release: 4609,
    interface: -1 },
  { vendorId: 1452,
    productId: 627,
    path: 'SPI_05ac_0273_1000000',
    serialNumber: '',
    manufacturer: '',
    product: 'Apple Internal Keyboard / Trackpad',
    release: 1570,
    interface: -1 },
  { vendorId: 1133,
    productId: 50475,
    path: 'USB_046d_c52b_14100000',
    serialNumber: '',
    manufacturer: 'Logitech',
    product: 'USB Receiver',
    release: 4609,
    interface: -1 },
  { vendorId: 1133,
    productId: 16388,
    path: 'Unifying_046d_4004_14100300',
    serialNumber: '',
    manufacturer: 'Logitech',
    product: 'Unifying Device',
    release: 4609,
    interface: -1 },
  { vendorId: 1133,
    productId: 4136,
    path: 'Unifying_046d_1028_14100100',
    serialNumber: '',
    manufacturer: 'Logitech',
    product: 'Unifying Device',
    release: 4609,
    interface: -1 },
  { vendorId: 1133,
    productId: 16388,
    path: 'Unifying_046d_4004_14100200',
    serialNumber: '',
    manufacturer: 'Logitech',
    product: 'Unifying Device',
    release: 4609,
    interface: -1 },
  { vendorId: 1133,
    productId: 16388,
    path: 'Unifying_046d_4004_14100200',
    serialNumber: '',
    manufacturer: 'Logitech',
    product: 'Unifying Device',
    release: 4609,
    interface: -1 },
  { vendorId: 1452,
    productId: 627,
    path: 'SPI_05ac_0273_1000000',
    serialNumber: '',
    manufacturer: '',
    product: 'Apple Internal Keyboard / Trackpad',
    release: 1570,
    interface: -1 },
  { vendorId: 1133,
    productId: 16388,
    path: 'Unifying_046d_4004_14100300',
    serialNumber: '',
    manufacturer: 'Logitech',
    product: 'Unifying Device',
    release: 4609,
    interface: -1 },
  { vendorId: 1133,
    productId: 16388,
    path: 'Unifying_046d_4004_14100300',
    serialNumber: '',
    manufacturer: 'Logitech',
    product: 'Unifying Device',
    release: 4609,
    interface: -1 },
  { vendorId: 1133,
    productId: 4136,
    path: 'Unifying_046d_1028_14100400',
    serialNumber: '',
    manufacturer: 'Logitech',
    product: 'Unifying Device',
    release: 4609,
    interface: -1 },
  { vendorId: 1452,
    productId: 627,
    path: 'SPI_05ac_0273_1000000',
    serialNumber: '',
    manufacturer: '',
    product: 'Apple Internal Keyboard / Trackpad',
    release: 1570,
    interface: -1 },
  { vendorId: 1452,
    productId: 627,
    path: 'SPI_05ac_0273_1000000',
    serialNumber: '',
    manufacturer: '',
    product: 'Apple Internal Keyboard / Trackpad',
    release: 1570,
    interface: -1 },
  { vendorId: 1133,
    productId: 16388,
    path: 'Unifying_046d_4004_14100200',
    serialNumber: '',
    manufacturer: 'Logitech',
    product: 'Unifying Device',
    release: 4609,
    interface: -1 },
  { vendorId: 1133,
    productId: 50475,
    path: 'USB_046d_c52b_14100000',
    serialNumber: '',
    manufacturer: 'Logitech',
    product: 'USB Receiver',
    release: 4609,
    interface: -1 },
  { vendorId: 1133,
    productId: 16388,
    path: 'Unifying_046d_4004_14100200',
    serialNumber: '',
    manufacturer: 'Logitech',
    product: 'Unifying Device',
    release: 4609,
    interface: -1 },
  { vendorId: 1133,
    productId: 16388,
    path: 'Unifying_046d_4004_14100300',
    serialNumber: '',
    manufacturer: 'Logitech',
    product: 'Unifying Device',
    release: 4609,
    interface: -1 },
  { vendorId: 1133,
    productId: 50475,
    path: 'USB_046d_c52b_14100000',
    serialNumber: '',
    manufacturer: 'Logitech',
    product: 'USB Receiver',
    release: 4609,
    interface: -1 },
  { vendorId: 1133,
    productId: 16388,
    path: 'Unifying_046d_4004_14100200',
    release: 4609,
    interface: -1 },
  { vendorId: 1133,
    productId: 16388,
    path: 'Unifying_046d_4004_14100300',
    serialNumber: '',
    manufacturer: 'Logitech',
    product: 'Unifying Device',
    release: 4609,
    interface: -1 },
  { vendorId: 1133,
    productId: 50475,
    path: 'USB_046d_c52b_14100000',
    serialNumber: '',
    manufacturer: 'Logitech',
    product: 'USB Receiver',
    release: 4609,
    interface: -1 },
  { vendorId: 1133,
    productId: 16388,
    path: 'Unifying_046d_4004_14100200',
    serialNumber: '',
    manufacturer: 'Logitech',
    product: 'Unifying Device',
    release: 4609,
    interface: -1 },
  { vendorId: 1452,
    productId: 627,
    path: 'SPI_05ac_0273_1000000',
    serialNumber: '',
    manufacturer: '',
    product: 'Apple Internal Keyboard / Trackpad',
    release: 1570,
    interface: -1 } ]

[image]

todbot, 2015-10-07:
Not a HID device, as discussed here: https://stackoverflow.com/questions/30732420/node-hid-not-showing-all-usb-devices

closed 2015-10-07 by todbot