node-hid/node-hid #98: Consider using node-pre-gyp

uksus70, 2015-06-12:
I have installed the package into my node-webkit project:

```
npm install node-hid --runtime=node-webkit --target=0.8.6
```

And during runtime I get the following error:
```
"Uncaught Error: /root/src/test/node_modules/node-hid/build/Release/HID.node: undefined symbol: _ZN2v86String3NewEPKci", source: module.js (359)
```

After enetering the `node_modules\node-hid` folder and running `nw-gyp rebuild --target=0.8.6` there everything works fine.

Please consider using [node-pre-gyp](https://www.npmjs.com/package/node-pre-gyp) in order to automate runtime detection during installation process.

todbot, 2015-10-15:
Great idea.  node-hid uses node-pre-gyp now for most popular combinations of OS and Node versions.  Let me know if you try it out.

closed 2015-10-15 by todbot

brandonros, 2015-12-30:
Just to clarify, node-webkit prebuilt binaries aren't available, right?

This is a real shame. Is there anyway to get around this? I'd really rather not have to download 3GB of Windows compiler data.