node-hid/node-hid #99: segfault on hid_write

atoikka, 2015-06-17:
Hi.

Please pardon my inexperience in the field of USB devices. It would be very convenient for me and it would benefit the structural integrity of the application that I'm developing if I could communicate with a HID through the node-hid Node.JS module.

Unfortunately that is not the case as I get a "segmentation fault" when hid_write is executed. An outtake of my code is this:
```javascript
var HID = require('node-hid');
var Config = {
	vendorId: 1240,
	productId: 64823
};

var device = false;
var devices = HID.devices();
console.log(devices);
if(devices.length === 0) {
	return false;
}
for(var i = 0; i != devices.length; i++)
{
	if(devices[i].vendorId == Config.vendorId && devices[i].productId == Config.productId) {
		try {
			device = new HID.HID(devices[i].path);
		}
		catch(err) {
			console.log("[ERROR] [measure] Device cannot be opened: " + err);
		}
		break;
	}
}
if(device !== false)
{
	device.on("data", onData.bind(this));
	// if I uncomment the following line (regardless of the data that is sent), I get a segfault
	//device.write([]);
}

//...

onData = function(data) {
	console.log(data);
}
```
I have tried disabling listening to the data event to see if that may have anything to do with it. I have also fiddled with the setNonBlocking option, but with no luck. A segfault happens when in HID.cc, hid_write is executed.

dmesg gives me this information:
```
[ 4090.840276] nodejs[3395]: segfault at 0 ip 00007fd572143832 sp 00007ffca7087510 error 4 in libv8.so.3.14.5[7fd5720a4000+3f1000]
[ 4145.085411] nodejs[3556]: segfault at 0 ip 00007f9366fb0832 sp 00007ffcabff2920 error 4 in libv8.so.3.14.5[7f9366f11000+3f1000]
[ 4320.961328] nodejs[3783]: segfault at 0 ip 00007ff1b7fbf832 sp 00007fffecdf90b0 error 4 in libv8.so.3.14.5[7ff1b7f20000+3f1000]
```

I wrote a small C++ program to communicate with the device and 2-way communication works. Here's an outtake from code that works for me:
```c++
handle = hid_open(0x04d8, 0xfd37, NULL);
unsigned char buf[8];
hid_set_nonblocking(handle, 1);
buf[0] = 0x0;
buf[1] = 'S';
res = hid_write(handle, buf, 2);
if(res < 0) {
	printf("Unable to write()!\n");
}
	
res = 0;
while(res == 0) {
	res = hid_read(handle, buf, sizeof(buf));
	if(res == 0) {
		printf("waiting...\n");
	} else if(res < 0) {
		printf("Unable to read()\n");
	}
	usleep(500*1000);
}
	
printf("Version data:\n");
for(i = 0; i < res; i++) {
	printf("%02hhx ", buf[i]);
}
```
The code above uses the same library, I execute it in myProjectDir/node_modules/node-hid/hidapi, and it works.

I am very sorry if this is wasting your time as it seems to me as an error on my device management or my scripting, but I'm puzzled that calling hid_write does ot crash from my application.

I would be very pleased if you could direct me to debug measures which could help me track down this issue. From debug steps, I have tried building (with gyp) node-hid in debug mode and running "node debug index.js" instead of "node index.js", but with no useful results as the crash comes from libv8.so according to my understanding.

Thank you!

breedx2, 2015-08-07:
:+1: for this issue.  I can reproduce it exactly on Linux.  The above C code (tweaked for my specific HID device) works fine, but I can't get node-hid to write to the device without segfaulting. Show-stopper.

todbot, 2015-10-11:
I believe this has been fixed in commit b8e7736c0d237830bd26915935cf3bf5a9c7f34e.  It's not been pushed to npm yet but if you want to try it from a checkout, please do.

breedx2, 2015-10-11:
Bummer I can't try it -- I ended up writing some C code for that project and no longer have the HID device to test with....but I appreciate the follow-up!

todbot, 2015-10-12:
:-) closing for now then

closed 2015-10-12 by todbot