node-hid/node-hid #100: node-hid for Windows?

giorgoc, 2015-06-19:
Guys,

Any chance to get it work for windows? Sorry, kinda new to node and not sure what i m doing wrong. I have installed lots of modules with no issues, but node-hid ... I have tried so many different node versions, and always getting the same error :( See below:

Any tips?

George

C:\Users\George\Desktop\notifier>node -v
v0.10.32

C:\Users\George\Desktop\notifier>npm install https://github.com/natevw/node-hid/releases/download/test-node-12/node-hid-0.3.2.tgz
/
> node-hid@0.3.2 install C:\Users\George\Desktop\notifier\node_modules\node-hid
> node-gyp rebuild

C:\Users\George\Desktop\notifier\node_modules\node-hid>node "C:\Program Files (x86)\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
C:\Users\George\Desktop\notifier\node_modules\node-hid\build\hidapi.vcxproj(18,3): error MSB4019: The imported project "C:\Microsoft.Cpp.De
fault.props" was not found. Confirm that the path in the <Import> declaration is correct, and that the file exists on disk.
gyp ERR! build error
gyp ERR! stack Error: `C:\Windows\Microsoft.NET\Framework\v4.0.30319\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Program Files (x86)\nodejs\node_modules\npm\node_modules\node-gyp\lib\build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:810:12)
gyp ERR! System Windows_NT 6.1.7601
gyp ERR! command "node" "C:\\Program Files (x86)\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\Users\George\Desktop\notifier\node_modules\node-hid
gyp ERR! node -v v0.10.32
gyp ERR! node-gyp -v v1.0.1
gyp ERR! not ok

npm ERR! node-hid@0.3.2 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the node-hid@0.3.2 install script.
npm ERR! This is most likely a problem with the node-hid package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls node-hid
npm ERR! There is likely additional logging output above.
npm ERR! System Windows_NT 6.1.7601
npm ERR! command "C:\\Program Files (x86)\\nodejs\\\\node.exe" "C:\\Program Files (x86)\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "install" "https://github.com/natevw/node-hid/releases/download/test-node-12/node-hid-0.3.2.tgz"
npm ERR! cwd C:\Users\George\Desktop\notifier
npm ERR! node -v v0.10.32
npm ERR! npm -v 1.4.28
npm ERR! code ELIFECYCLE
npm ERR! not ok code 0

giorgoc, 2015-06-19:
same with the latest node-hid version

C:\Users\George\Desktop\notifier>npm install node-hid
\
> node-hid@0.4.0 install C:\Users\George\Desktop\notifier\node_modules\node-hid
> node-gyp rebuild

C:\Users\George\Desktop\notifier\node_modules\node-hid>node "C:\Program Files (x86)\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
C:\Users\George\Desktop\notifier\node_modules\node-hid\build\hidapi.vcxproj(18,3): error MSB4019: The imported project "C:\Microsoft.Cpp.De
fault.props" was not found. Confirm that the path in the <Import> declaration is correct, and that the file exists on disk.
gyp ERR! build error
gyp ERR! stack Error: `C:\Windows\Microsoft.NET\Framework\v4.0.30319\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Program Files (x86)\nodejs\node_modules\npm\node_modules\node-gyp\lib\build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:810:12)
gyp ERR! System Windows_NT 6.1.7601
gyp ERR! command "node" "C:\\Program Files (x86)\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\Users\George\Desktop\notifier\node_modules\node-hid
gyp ERR! node -v v0.10.32
gyp ERR! node-gyp -v v1.0.1
gyp ERR! not ok

npm ERR! node-hid@0.4.0 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the node-hid@0.4.0 install script.
npm ERR! This is most likely a problem with the node-hid package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls node-hid
npm ERR! There is likely additional logging output above.
npm ERR! System Windows_NT 6.1.7601
npm ERR! command "C:\\Program Files (x86)\\nodejs\\\\node.exe" "C:\\Program Files (x86)\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "install" "node-hid"
npm ERR! cwd C:\Users\George\Desktop\notifier
npm ERR! node -v v0.10.32
npm ERR! npm -v 1.4.28
npm ERR! code ELIFECYCLE
npm ERR! not ok code 0

C:\Users\George\Desktop\notifier>

odingalt, 2015-09-17:
I get a lot of nanthrowerror

C:\TEMP\bootloader-jsx\node_modules\node-hid>node-gyp build --msvs_version=2013
| more
gyp info it worked if it ends with ok
gyp info using node-gyp@3.0.3
gyp info using node@4.0.0 | win32 | x64
gyp info spawn msbuild
gyp info spawn args [ 'build/binding.sln',
gyp info spawn args   '/clp:Verbosity=minimal',
gyp info spawn args   '/nologo',
gyp info spawn args   '/p:Configuration=Release;Platform=x64' ]
Building the projects in this solution one at a time. To enable parallel build,
please add the "/m" switch.
  hidapi.vcxproj -> C:\TEMP\bootloader-jsx\node_modules\node-hid\build\Release\\
hidapi.lib
  HID.cc
..\src\HID.cc(51): error C3861: 'NanThrowError': identifier not found [C:\TEMP\b
ootloader-jsx\node_modules\node-hid\build\HID.vcxproj]
..\src\HID.cc(67): warning C4290: C++ exception specification ignored except to
indicate a function is not __declspec(nothrow) [C:\TEMP\bootloader-jsx\node_modu
les\node-hid\build\HID.vcxproj]
..\src\HID.cc(70): warning C4290: C++ exception specification ignored except to
indicate a function is not __declspec(nothrow) [C:\TEMP\bootloader-jsx\node_modu
les\node-hid\build\HID.vcxproj]
..\src\HID.cc(92): error C2061: syntax error : identifier 'NanCallback' [C:\TEMP
\bootloader-jsx\node_modules\node-hid\build\HID.vcxproj]
..\src\HID.cc(106): error C2143: syntax error : missing ';' before '*' [C:\TEMP\
bootloader-jsx\node_modules\node-hid\build\HID.vcxproj]
.... (the list goes on)

C:\TEMP\bootloader-jsx\node_modules\node-hid>npm -v
2.14.3

C:\TEMP\bootloader-jsx\node_modules\node-hid>node -v
v4.0.0

C:\TEMP\bootloader-jsx\node_modules\node-hid>node-gyp -v
v3.0.3

todbot, 2015-10-06:
This has been addressed in pull request #111, it's just awaiting merging.  But you can use the patched version as mentioned in the comments for pull #111.

closed 2015-10-06 by todbot