node-hid/node-hid #101: Node-HID Lacks Support for Common HID Message Types?

Pecacheu, 2015-07-20:
Hello, I'm trying to make my gamepads vibrate using node-hid. Currently, I'm trying with a DS3, using code I (very slowly) ported from a C++ library for DualShock controllers. It claims to be able to make a DS3 vibrate by sending the same data I am trying to send using `hid.write`. Unfortunately, it doesn't seem to be working. It connects to controllers and even reads button data, but the DS3 just won't rumble.

So, I downloaded Apple's Bluetooth-sniffing app from the dev site (I'm on Mac OS X), and have discovered that node-hid is sending a different type of message than the DualShock 3 wants. The DS3 apparently needs transaction type **SET REPORT** and report type **OUTPUT**, which adds up to a starting byte of 0x52, in hexadecimal. The `write` command can only send 0xA2, or transaction type **DATA**. We should also have a `setReport` command, so that we can control devices like the DS3.

There is also **SET REPORT** combined with report type **FEATURE**, or 0x53, which the DS4 may use to initiate audio data streaming.

Pecacheu, 2015-09-26:
The report type being sent appears to be up to the "hid_write" function, which unfortunately is part of the OS-dependent `hid.c` file of node-hid. On a Mac, you can use one of 3 different possible report types as outlined in [Apple's IOHIDKeys referance](http://developer.apple.com/library/mac/documentation/Kernel/Reference/IOHIDKeys_header_reference/index.html#//apple_ref/c/tdef/IOHIDReportType).

Except, the hid_write function is using `kIOHIDFeatureTypeOutput`, which isn't one of the possibilities, and doesn't seem to appear anywhere in Apple's IOHID documentation. Strange...

todbot, 2015-09-26:
You may want to look at node-gamepad.  They purport to work with DualShock 3. https://github.com/carldanley/node-gamepad

Pecacheu, 2015-09-27:
I've tried several different node modules for gamepads. If you look at them, you'll find that any advanced features of the devices are listed as 'todo' or not mentioned. All gamepads I've seen use message types other than the basic **OUTPUT** type for control of any special features (with the exception of the DS4's back-LED, for some reason).

There was one I found that did support other message types because it actually had it's own HID implementation rather than relying on node-hid, but I couldn't get it to install on the latest Node.js.

todbot, 2015-10-11:
I've added a "test-ps3-rumbleled.js" example in src that shows how to set the rumble and LED of a PS3 dualshock controller.  It uses the node-hid `hid.write()` command.  Works on Bluetooth and USB connection modes.
https://github.com/node-hid/node-hid/blob/master/src/test-ps3-rumbleled.js

Since node-hid can do this, I'm closing this issue.  Thanks for bringing it up, I'm having fun playing with the PS3 controller.

closed 2015-10-11 by todbot

Pecacheu, 2015-10-15:
That is almost identical to the code I had before (based on reverse-engineering documentation of the PS3), except for that first byte in the write function. After some testing with the debuggers, I have discovered that the write function behaves differently with that extra 0x01 in front, and it formats the headers with correct report ID for the DS3.

This is not compliant with the HID spec. That being said, clearly I'm an idiot for not reading the node-hid documentation: "**Some devices use reportIds for OUTPUT reports. If that is the case, the first byte of the array to write() should be the reportId.**" Anyway, it's definitely rumbling now, so, thanks!

todbot, 2015-10-15:
You're not an idiot, I recently added that documentation.
node-hid is at the mercy of the underlying hidapi library, and that's how it works.