node-hid/node-hid #102: cannot open device with path 0001:001c:00

miguelmota, 2015-07-23:
When trying to open [big red button](http://dreamcheeky.com/big-red-button)

todbot, 2015-10-11:
Do you have more details on how it fails?

todbot, 2015-10-15:
Hi,
I have a Big Red Button and have tested it. Here's the test code I used:
https://github.com/node-hid/node-hid/blob/master/src/test-bigredbutton.js
It works with node-hid.

I suspect your problem is related to udev permissions.  You can test this by running your node program as 'root' or adding a udev rule for the Big Red Button.

To create a udev rule for Big Red Button:
* Create a file `/etc/udev/rules.d/99-bigredbutton.rules` and put into it two lines:
```
# Dream Cheeky - Big Red Button
ACTION=="add", SUBSYSTEM=="usb", ATTR{idVendor}=="1d34", ATTR{idProduct}=="000d", MODE="0666"
```
* Run `udevadm control --reload-rules`

(this stolen from: https://github.com/fidian/bonkers/blob/master/doc/99-bonkers.rules )

closed 2015-10-15 by todbot

miguelmota, 2017-08-30:
Awesome, thanks!