node-hid/node-hid #105: leak on HID.on('data', function(){ }) ?

4Taine, 2015-09-15:
Hi,
I run some code with your node-hid but I've a memory leak
When i do :
``` javascript
var HID = require('node-hid');
var devices = HID.devices(vendor, id);
var hid = new HID.HID(devices[0].path);
/*
hid.on("data", function(data){
   work(data);
   return;
});
*/
```

I have no memory leak, but when i do :
``` javascript
var HID = require('node-hid');
var devices = HID.devices(vendor, id);
var hid = new HID.HID(devices[0].path);
hid.on("data", function(data){
   /*
   work(data);
   */
   return;
});
```
I've an important one.
any idea ?
I use :
* node-hid version 0.4.0
* nodejs version 0.10.40

Thanks

4Taine, 2015-09-23:
hi,
I've fix my issue adding :
``` cpp
delete req;
```
in HID.cc at line 233

todbot, 2015-10-06:
Thanks for the patch, it's been incorporated in pull-request #111.

closed 2015-10-06 by todbot