node-hid/node-hid #112: npm install: build error

axelerator, 2015-10-01:
When I'm trying to `npm install node-hid` on my Mac I get a build error.

In file included from ../src/HID.cc:35:
../node_modules/nan/nan.h:324:27: error: redefinition of 'NanEnsureHandleOrPersistent'
  NAN_INLINE v8::Local<T> NanEnsureHandleOrPersistent(const v8::Local<T> &val) {

[Gist with npm-debug.log](https://gist.github.com/axelerator/5137a78050924efafbaa)

todbot, 2015-10-01:
node-hid currently only supports up to Node v0.12.7.  The pull-request #111 addresses this but it hasn't been merged yet by the node-hid owners.

If you want to use this patched node-hid with Node v4.x, you can do:
```
npm install 'https://github.com/lorenc-tomasz/node-hid.git#package'
```

axelerator, 2015-10-01:
Thx for the workaround!

closed 2015-10-01 by axelerator