node-hid/node-hid #113: Unexpected behavior with more than 4 equal devices

peterpunk, 2015-10-02:
I'm connecting 6 USB RFID readers and the on 'read' works normally up to 4 devices.
I tried with different hubs, combining them but the results are pretty the same.
Is something like this:

on device 1 -> no data
on device 2 -> no data
on device 3 -> data from 1 2 and 3
on device 4 -> no data
on device 5 -> no data
on device 6 -> data from 4, 5, 6
on device 1 -> no data
on device 1 -> no data
on device 1 -> no data
on device 1 -> no data
on device 1 -> no data
on device 3 -> no data
on device 4 -> data from 1, 3, 4

There's a pattern but I cannot figure out from the code where is the problem.

MacOs X Yosemite.

todbot, 2015-10-11:
Could you post the code you're using for this?  I have several identical HID devices I could use to try to replicate your results.
And what's the model number of the USB RFID readers?

todbot, 2016-02-23:
Closing as no response from submitter.  If these were USB HID devices (which likely if RFID readers), then results will be inconsistent due to the OS grabbing them.

closed 2016-02-23 by todbot