node-hid/node-hid #114: Issues while compiling under Yosemite (10.10.5) and node.js 4.1.1

dblas, 2015-10-03:
Here follows the result. An idea ? Please avoid "try node 0.10".
I'm very bad at C++ ?
Thanks,
db

> node-hid@0.4.0 install /Users/db/node/node-hid
> node-gyp rebuild

  CC(target) Release/obj.target/hidapi/hidapi/mac/hid.o
../hidapi/mac/hid.c:255:20: warning: comparison of integers of different signs: 'CFIndex' (aka 'long') and 'size_t' (aka 'unsigned long')
      [-Wsign-compare]
                if (chars_copied == len)
                    ~~~~~~~~~~~~ ^  ~~~
../hidapi/mac/hid.c:295:20: warning: comparison of integers of different signs: 'CFIndex' (aka 'long') and 'size_t' (aka 'unsigned long')
      [-Wsign-compare]
                if (used_buf_len == len)
                    ~~~~~~~~~~~~ ^  ~~~
2 warnings generated.
  LIBTOOL-STATIC Release/hidapi.a
  CXX(target) Release/obj.target/HID/src/HID.o
In file included from ../src/HID.cc:35:
../node_modules/nan/nan.h:324:27: error: redefinition of 'NanEnsureHandleOrPersistent'
  NAN_INLINE v8::Local<T> NanEnsureHandleOrPersistent(const v8::Local<T> &val) {
                          ^
../node_modules/nan/nan.h:319:17: note: previous definition is here
  v8::Handle<T> NanEnsureHandleOrPersistent(const v8::Handle<T> &val) {
                ^
../node_modules/nan/nan.h:344:27: error: redefinition of 'NanEnsureLocal'
  NAN_INLINE v8::Local<T> NanEnsureLocal(const v8::Handle<T> &val) {
                          ^
../node_modules/nan/nan.h:334:27: note: previous definition is here
  NAN_INLINE v8::Local<T> NanEnsureLocal(const v8::Local<T> &val) {
                          ^
../node_modules/nan/nan.h:757:13: error: no member named 'smalloc' in namespace 'node'
    , node::smalloc::FreeCallback callback
      ~~~~~~^
../node_modules/nan/nan.h:768:12: error: no matching function for call to 'New'
    return node::Buffer::New(v8::Isolate::GetCurrent(), data, size);
           ^~~~~~~~~~~~~~~~~
/Users/db/.node-gyp/4.1.1/include/node/node_buffer.h:31:40: note: candidate function not viable: no known conversion from 'uint32_t'
      (aka 'unsigned int') to 'enum encoding' for 3rd argument
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
/Users/db/.node-gyp/4.1.1/include/node/node_buffer.h:43:40: note: candidate function not viable: 2nd argument ('const char *') would lose const
      qualifier
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
/Users/db/.node-gyp/4.1.1/include/node/node_buffer.h:28:40: note: candidate function not viable: requires 2 arguments, but 3 were provided
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate, size_t length);
                                       ^
/Users/db/.node-gyp/4.1.1/include/node/node_buffer.h:36:40: note: candidate function not viable: requires 5 arguments, but 3 were provided
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
In file included from ../src/HID.cc:35:
../node_modules/nan/nan.h:772:12: error: no viable conversion from 'v8::MaybeLocal<v8::Object>' to 'v8::Local<v8::Object>'
    return node::Buffer::New(v8::Isolate::GetCurrent(), size);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/db/.node-gyp/4.1.1/include/node/v8.h:210:7: note: candidate constructor (the implicit copy constructor) not viable: no known conversion
      from 'v8::MaybeLocal<v8::Object>' to 'const v8::Local<v8::Object> &' for 1st argument
class Local {
      ^
/Users/db/.node-gyp/4.1.1/include/node/v8.h:210:7: note: candidate constructor (the implicit move constructor) not viable: no known conversion
      from 'v8::MaybeLocal<v8::Object>' to 'v8::Local<v8::Object> &&' for 1st argument
class Local {
      ^
/Users/db/.node-gyp/4.1.1/include/node/v8.h:214:13: note: candidate template ignored: could not match 'Local' against 'MaybeLocal'
  V8_INLINE Local(Local<S> that)
            ^
/Users/db/.node-gyp/4.1.1/include/node/v8.h:326:13: note: candidate template ignored: could not match 'S *' against
      'v8::MaybeLocal<v8::Object>'
  V8_INLINE Local(S* that)
            ^
In file included from ../src/HID.cc:35:
../node_modules/nan/nan.h:779:26: error: no member named 'Use' in namespace 'node::Buffer'
    return node::Buffer::Use(v8::Isolate::GetCurrent(), data, size);
           ~~~~~~~~~~~~~~^
../src/HID.cc:231:5: warning: 'FatalException' is deprecated: Use FatalException(isolate, ...) [-Wdeprecated-declarations]
    FatalException(tryCatch);
    ^
/Users/db/.node-gyp/4.1.1/include/node/node.h:283:29: note: 'FatalException' has been explicitly marked deprecated here
                inline void FatalException(const v8::TryCatch& try_catch) {
                            ^
/Users/db/.node-gyp/4.1.1/include/node/node.h:66:42: note: expanded from macro 'NODE_DEPRECATED'
    __attribute__((deprecated(message))) declarator
                                         ^
In file included from ../src/HID.cc:30:
/Users/db/.node-gyp/4.1.1/include/node/v8.h:221:5: error: assigning to 'v8::Primitive *volatile' from incompatible type 'v8::Value *'
    TYPE_CHECK(T, S);
    ^~~~~~~~~~~~~~~~
/Users/db/.node-gyp/4.1.1/include/node/v8.h:180:37: note: expanded from macro 'TYPE_CHECK'
    *(static_cast<T* volatile*>(0)) = static_cast<S*>(0);      \
                                    ^ ~~~~~~~~~~~~~~~~~~
../node_modules/nan/nan.h:501:12: note: in instantiation of function template specialization 'v8::Local<v8::Primitive>::Local<v8::Value>'
      requested here
    return NanEscapeScope(NanNew(v8::Undefined(v8::Isolate::GetCurrent())));
           ^
../node_modules/nan/nan.h:483:30: note: expanded from macro 'NanEscapeScope'
# define NanEscapeScope(val) scope.Escape(Nan::imp::NanEnsureLocal(val))
                             ^
In file included from ../src/HID.cc:30:
/Users/db/.node-gyp/4.1.1/include/node/v8.h:221:5: error: assigning to 'v8::Boolean *volatile' from incompatible type 'v8::Value *'
    TYPE_CHECK(T, S);
    ^~~~~~~~~~~~~~~~
/Users/db/.node-gyp/4.1.1/include/node/v8.h:180:37: note: expanded from macro 'TYPE_CHECK'
    *(static_cast<T* volatile*>(0)) = static_cast<S*>(0);      \
                                    ^ ~~~~~~~~~~~~~~~~~~
../node_modules/nan/nan.h:511:12: note: in instantiation of function template specialization 'v8::Local<v8::Boolean>::Local<v8::Value>'
      requested here
    return NanEscapeScope(NanNew(v8::True(v8::Isolate::GetCurrent())));
           ^
../node_modules/nan/nan.h:483:30: note: expanded from macro 'NanEscapeScope'
# define NanEscapeScope(val) scope.Escape(Nan::imp::NanEnsureLocal(val))
                             ^
In file included from ../src/HID.cc:30:
/Users/db/.node-gyp/4.1.1/include/node/v8.h:221:5: error: assigning to 'v8::Function *volatile' from incompatible type 'v8::Value *'
    TYPE_CHECK(T, S);
    ^~~~~~~~~~~~~~~~
/Users/db/.node-gyp/4.1.1/include/node/v8.h:180:37: note: expanded from macro 'TYPE_CHECK'
    *(static_cast<T* volatile*>(0)) = static_cast<S*>(0);      \
                                    ^ ~~~~~~~~~~~~~~~~~~
../node_modules/nan/nan.h:1645:12: note: in instantiation of function template specialization 'v8::Local<v8::Function>::Local<v8::Value>'
      requested here
    return NanEscapeScope(NanNew(handle)->Get(kCallbackIndex)
           ^
../node_modules/nan/nan.h:483:30: note: expanded from macro 'NanEscapeScope'
# define NanEscapeScope(val) scope.Escape(Nan::imp::NanEnsureLocal(val))
                             ^
In file included from ../src/HID.cc:30:
/Users/db/.node-gyp/4.1.1/include/node/v8.h:221:5: error: assigning to 'v8::Object *volatile' from incompatible type 'v8::Value *'
    TYPE_CHECK(T, S);
    ^~~~~~~~~~~~~~~~
/Users/db/.node-gyp/4.1.1/include/node/v8.h:180:37: note: expanded from macro 'TYPE_CHECK'
    *(static_cast<T* volatile*>(0)) = static_cast<S*>(0);      \
                                    ^ ~~~~~~~~~~~~~~~~~~
../node_modules/nan/nan.h:1776:12: note: in instantiation of function template specialization 'v8::Local<v8::Object>::Local<v8::Value>'
      requested here
    return NanEscapeScope(
           ^
../node_modules/nan/nan.h:483:30: note: expanded from macro 'NanEscapeScope'
# define NanEscapeScope(val) scope.Escape(Nan::imp::NanEnsureLocal(val))
                             ^
1 warning and 10 errors generated.
make: *** [Release/obj.target/HID/src/HID.o] Error 1

rdepena, 2015-10-03:
There is a pull request already in for 4.1.1 compatibility #111 you can target the following package https://github.com/lorenc-tomasz/node-hid#package in npm and preview the changes.

dblas, 2015-10-03:
OK, thank you.
I gonna close the issue since the pb I encountered is not well described 
by these errors.
Indeed, there are many errors due to Nan and its numerous flavours.
db

dblas, 2015-10-04:
The pb doesn't sum up in a few errors. 
It's linked to Nan which has too many versions depending on the node.js/io.js version themselves.
db

closed 2015-10-04 by dblas

dblas, 2015-10-04:
Good news,
Lorenc-tomasz version works with node.js v. 4.1.1!

Just 4 things to change in order to avoid warnings:

Replacing *FatalException(* with 
*FatalException(v8::Isolate::GetCurrent(), ...)* in HID.cc

as well as 2 times *char_copied == len* with *labs(char_copied) == 
(long) len* in hid.c

Best,
db

todbot, 2015-10-05:
this is great @dblas, could you create a pull-request for this change?

dblas, 2015-10-05:
I tried to but didn't manage to.
I'll try again tonight (french standard time).
Best,
db