node-hid/node-hid #115: variable 'bytes read' might be clobbered by.. when installing

brianehlert, 2015-10-07:
> node-hid@0.5.0 install /usr/local/lib/node_modules/node-hid
> node-gyp rebuild

make: Entering directory '/usr/local/lib/node_modules/node-hid/build'
  CC(target) Release/obj.target/hidapi/hidapi/libusb/hid.o
../hidapi/libusb/hid.c: In function âhid_read_timeoutâ:
../hidapi/libusb/hid.c:1078:6: warning: variable âbytes_readâ might be clobbered by âlongjmpâ or âvforkâ [-Wclobbered]
  int bytes_read = -1;
      ^
  AR(target) Release/obj.target/hidapi.a
  COPY Release/hidapi.a
  CXX(target) Release/obj.target/HID/src/HID.o
../src/HID.cc: In static member function âstatic void HID::recvAsyncDone(uv_work_t*)â:
../src/HID.cc:231:28: warning: âvoid node::FatalException(const v8::TryCatch&)â is deprecated (declared at /root/.node-gyp/4.0.0/include/node/node.h:282): Use FatalException(isolate, ...) [-Wdeprecated-declarations]
     FatalException(tryCatch);
                            ^
  SOLINK_MODULE(target) Release/obj.target/HID.node
  COPY Release/HID.node
make: Leaving directory '/usr/local/lib/node_modules/node-hid/build'
postinstall â unlock      
/usr/local/lib
âââ node-hid@0.5.0

todbot, 2015-10-07:
Hi, a couple of questions:
1. What OS are you on? (exact distro and version)
2. Could you copy the npm-debug.log instead of your screen?  Makes it hard to read because of the color codes showing up as garbage characters.
Thanks!

brianehlert, 2015-10-07:
Sorry about that.
The distro is Raspian Jessie.  Node 4, NPM 3
Linux raspberrypi 4.1.7-v7+ #817 SMP PREEMPT Sat Sep 19 15:32:00 BST 2015 armv7l GNU/Linux
Ugh.  I did a global install and having a heck of a time locating the npm-debug.log

brianehlert, 2015-10-07:
Ran sudo npm install -g -y node-hid --unsafe-perm 

again.  Still not finding where npm-debug.log might be .

> node-hid@0.5.0 install /usr/local/lib/node_modules/node-hid
> node-gyp rebuild

make: Entering directory '/usr/local/lib/node_modules/node-hid/build'
  CC(target) Release/obj.target/hidapi/hidapi/libusb/hid.o
../hidapi/libusb/hid.c: In function hid_read_timeout:
../hidapi/libusb/hid.c:1078:6: warning: variable bytes_read might be clobbered by longjmp or vfork [-Wclobbered]
  int bytes_read = -1;
      ^
  AR(target) Release/obj.target/hidapi.a
  COPY Release/hidapi.a
  CXX(target) Release/obj.target/HID/src/HID.o
../src/HID.cc: In static member function static void HID::recvAsyncDone(uv_work_t*):
../src/HID.cc:231:28: warning: void node::FatalException(const v8::TryCatch&) is deprecated (declared at /root/.node-gyp/4.0.0/include/node/node.h:282): Use FatalException(isolate, ...) [-Wdeprecated-declarations]
     FatalException(tryCatch);
                            ^
  SOLINK_MODULE(target) Release/obj.target/HID.node
  COPY Release/HID.node
make: Leaving directory '/usr/local/lib/node_modules/node-hid/build'

todbot, 2015-10-08:
Okay I've been able to replicate this on my Pi.
The problem is the default gcc & g++ on Raspbian is 4.6 and Node 4 expects 4.8.
Therefore, a workaround for now is to install gcc-/g++-4.8 and tell npm to to use it.  Like so:
```
$ sudo apt-get install gcc-4.8 g++-4.8
$ export CXX=g++-4.8
$ npm install node-hid
```
You'll get a warning, but it will compile and I've verified it does work on Raspbian 4.1.7-v7+.
I'll work on updating the docs to note this.

Two more questions:
1. How did you install Node4 on your system? 
2. Why are you installing node-hid globally?

brianehlert, 2015-10-08:
Thanks for that.  I will try that myself and tweak my instructions.

The global install of node-hid was to help me identify any issues with node-hid when it was installed as a deeply nested dependency of another node application.  That way I could get it out of the way and focus on other issues specific to that application.

I installed Node4 from the node-arm project.
echo "deb http://node-arm.herokuapp.com/ /" | sudo tee --append /etc/apt/sources.list
sudo apt-get update
sudo apt-get install node -y --force-yes

brianehlert, 2015-10-08:
Making notes as I go...
So I tried it again.
I updated the gcc.  (I did not run the command export CXX=g++-4.8)
I allowed the module to install node-hid.  (not installing node-hid globally)

And I get the following:

npm WARN deprecated node-hid@0.4.1: bad semver, NANv2 change means we shoudl be v0.5.0, NANv1 is 0.4.0

> node-hid@0.4.1 install /home/pi/gateblu-service/tmp/node_modules/meshblu-blink1/node_modules/node-blink1/node_modules/node-hid
> node-gyp rebuild

make: Entering directory '/home/pi/gateblu-service/tmp/node_modules/meshblu-blink1/node_modules/node-blink1/node_modules/node-hid/build'
  CC(target) Release/obj.target/hidapi/hidapi/libusb/hid.o
../hidapi/libusb/hid.c: In function hid_read_timeout:
../hidapi/libusb/hid.c:1078:6: warning: variable bytes_read might be clobbered by longjmp or vfork [-Wclobbered]
  int bytes_read = -1;
      ^
  AR(target) Release/obj.target/hidapi.a
  COPY Release/hidapi.a
  CXX(target) Release/obj.target/HID/src/HID.o
../src/HID.cc: In static member function static void HID::recvAsyncDone(uv_work_t*):
../src/HID.cc:231:28: warning: void node::FatalException(const v8::TryCatch&) is deprecated (declared at /home/pi/.node-gyp/4.0.0/include/node/node.h:282): 
     FatalException(tryCatch);
                            ^
  SOLINK_MODULE(target) Release/obj.target/HID.node
  COPY Release/HID.node
make: Leaving directory '/home/pi/gateblu-service/tmp/node_modules/meshblu-blink1/node_modules/node-blink1/node_modules/node-hid/build'

I am going to start fresh and install node-hid globally and do all the rest the same.

brianehlert, 2015-10-08:
If I install node-hid globally I get:

> node-hid@0.5.0 install /usr/local/lib/node_modules/node-hid
> node-gyp rebuild

install:nan               
make: Entering directory '/usr/local/lib/node_modules/node-hid/build'
  CC(target) Release/obj.target/hidapi/hidapi/libusb/hid.o
  AR(target) Release/obj.target/hidapi.a
  COPY Release/hidapi.a
  CXX(target) Release/obj.target/HID/src/HID.o
../src/HID.cc: In static member function static void HID::recvAsyncDone(uv_work_t*):
../src/HID.cc:231:28: warning: âvoid node::FatalException(const v8::TryCatch&) is deprecated (declared at /root/.node-gyp/4.0.0/include/node/node.h:282): Use FatalException(isolate, ...) [-Wdeprecated-declarations]
     FatalException(tryCatch);
                            ^
  SOLINK_MODULE(target) Release/obj.target/HID.node
  COPY Release/HID.node
make: Leaving directory '/usr/local/lib/node_modules/node-hid/build'

However, node-hid reports as being installed and actually works.
If I don't install using;  sudo npm install -g -y node-hid --unsafe-perm  it does not actually install on my RaspberryPi.

Obviously the issue is beyond my rudimentary Linux skills.

todbot, 2015-10-08:
In both of those cases, I believe node-hid is installing, you're just seeing warnings.  (and getting those warnings to go away is something I'm working on)

todbot, 2015-10-08:
Also, how are you getting Node v4.1.2 from node-arm?  
When I run through the steps you mention I get node v4.0.0.

brianehlert, 2015-10-08:
I'm not.
Just installing node-arm as I noted

https://github.com/node-hid/node-hid/issues/115#issuecomment-146716690

todbot, 2015-10-08:
oh duh, my mistake. I read the kernel version (4.1.7) as the Node version.

todbot, 2015-10-15:
I think this issue is closed.  Let me know if you have other issues.

closed 2015-10-15 by todbot