node-hid/node-hid #116: Issue in Node 4.1.2

rdepena, 2015-10-09:
Tried to Npm install and got the following:

gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/Users/rdepena/.nvm/versions/node/v4.1.2/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:270:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Darwin 14.5.0
gyp ERR! command "/Users/rdepena/.nvm/versions/node/v4.1.2/bin/node" "/Users/rdepena/.nvm/versions/node/v4.1.2/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/rdepena/Documents/Projects/node-hid
gyp ERR! node -v v4.1.2
gyp ERR! node-gyp -v v3.0.3
gyp ERR! not ok

npm ERR! Darwin 14.5.0
npm ERR! argv "/Users/rdepena/.nvm/versions/node/v4.1.2/bin/node" "/Users/rdepena/.nvm/versions/node/v4.1.2/bin/npm" "install"
npm ERR! node v4.1.2
npm ERR! npm  v2.14.4
npm ERR! code ELIFECYCLE
npm ERR! node-hid@0.5.0 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the node-hid@0.5.0 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the node-hid package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls node-hid
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /Users/rdepena/Documents/Projects/node-hid/npm-debug.log

todbot, 2015-10-09:
Hi, 
What OS is this?  It looks like Mac OS X but node-hid compiles on my Macs.
In a new empty directory, could you do "npm install node-hid" and post a pastebin of the entire "npm-debug.log"?

rdepena, 2015-10-09:
hey, I am on Mac OS 10.10.5, pastebin: http://pastebin.com/7nwtBHTE

todbot, 2015-10-09:
wow that's just not giving any useful error info.  Do you have Xcode installed?  (you need it to compile node packages with native code like node-hid)

rdepena, 2015-10-09:
I do have xcode 7.0.1

todbot, 2015-10-09:
very weird.  Our systems sound identical.  What happens when you type `g++ --version` on the commandline?

todbot, 2015-10-09:
oh wait, are you doing a git checkout or npm download?  That is, this:
```
git clone git@github.com:node-hid/node-hid.git
cd node-hid
npm install
```
or 
```
npm install node-hid
```
If you're doing the first one, you need to initialize the hidapi submodule:
```
git submodule update --init 
npm install
```

rdepena, 2015-10-09:
result of g++ --version:

Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 7.0.0 (clang-700.0.72)
Target: x86_64-apple-darwin14.5.0
Thread model: posix

Additionally I am trying both methods with the same result event post  "git submodule update --init "

todbot, 2015-10-09:
Unfortunately, the npm-debug.log isn't giving any useful output.  It seems like something is screwed up with node-gyp or the directory you're in isn't empty before doing `npm install node-hid`.

About the only thing I can think to try would be a "nuke from orbit": disable/remove nvm, get a fresh install of Node v4.1.2 from nodejs.org using their installer, and try installing node-hid again.

tommy7600, 2015-10-09:
Testes on 4.1.2 on all my machines - all works ok

rdepena, 2015-10-09:
I will try the "nuke from orbit" approach, I just tried on another machine and it worked.

todbot, 2015-10-09:
Another thing that might be an issue: perhaps there's something wrong with node-gyp or the python it's using.
In a new directory, try this:
```
git clone git@github.com:node-hid/node-hid.git
cd node-hid
git submodule update --init 
npm install    # this line should partially fail like before, but at least nan will get installed
npm install -g node-gyp 
node-gyp --verbose clean configure build
```

rdepena, 2015-10-09:
that sounds promising, I will try latter tonight. thank you.

rdepena, 2015-10-11:
hmm, nothing seems to work, i removed nvm from my system and installed a clean version of 4.1.2 and still nothing, on the other hand I can confirm that this is not a node-hid issue but seems like a node-gyp problem as other native backed node modules fail as well, hiredis for example, so I will close but will make sure to update this issue once I know what is up.

closed 2015-10-11 by rdepena

todbot, 2015-10-11:
That's so strange. If you ever figure it out let me know, I'm really curious.

rdepena, 2015-10-20:
Apparently the issue was that for some strange reason GNU libtool was being used instead of the XcodeDefault.xctoolchain libtool. works now, but not sure what swapped them initially.