node-hid/node-hid #117: Cannot open device on Windows 10 64bit

stone18x, 2015-10-14:
Hey I am trying to use your library for connecting to a xbox 360 controller but I'm always running in errors. When showing a list of all connected devices everything is working fine and my controller is shown:

vendorId: 1118,
    productId: 654,
    path: '\\\\?\\hid#vid_045e&pid_028e&ig_00#7&1c37ca02&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}',
    product: 'Controller (XBOX 360 For Windows)',
    release: 0,
    interface: -1 

But when I try to access the controller this error occurs.
var device = new hid.HID(1118, 654);

C:\Users\stefa\AppData\Roaming\npm\node_modules\node-hid\index.js:24
        this._raw = new (Function.prototype.bind.apply(binding.HID,
                    ^

Error: cannot open device with vendor id 0x45e and product id 0x28e
    at Error (native)
    at new HID (C:\Users\stefa\AppData\Roaming\npm\node_modules\node-hid\index.j
s:24:14)
    at Object.<anonymous> (C:\Users\stefa\Dropbox\AIM15\Semester 1\MMPE\lib_pres
entation\src\show_devices.js:6:14)
    at Module._compile (module.js:434:26)
    at Object.Module._extensions..js (module.js:452:10)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
    at Function.Module.runMain (module.js:475:10)
    at startup (node.js:118:18)
    at node.js:952:3

I'm running node as adminstrator and node-gyp is installed.

todbot, 2015-10-14:
I'll try this out myself. How are you connecting the controller to your PC? Is it a wired controller or via the wireless dongle thing?

stone18x, 2015-10-14:
It's a wired controller. 
https://www.microsoft.com/hardware/en-us/d/xbox-360-controller-for-windows

tommy7600, 2015-10-15:
I have the same issues with Windows 10. They probably changes something within Windows itself.

tommy7600, 2015-10-16:
On Windows 10 I cannot open any device :(

simonbooth, 2015-12-31:
I had the same issue with all devices on win 10 and came across this upstream fix in hidapi https://github.com/signal11/hidapi/issues/231 so far I just patched my local version and did a rebuild and its working so far. I am using a wiimote, not an xbox360 controller, so ymmv.

MadLittleMods, 2016-01-03:
Probably related to https://github.com/node-hid/node-hid/issues/73

todbot, 2016-03-28:
If you have the ability to do a rebuild of node-hid on your Windows 10 box, try the current checkout.  The underlying `hidapi` library that node-hid uses was recently updated to potentially fix this issues.

closed 2019-11-23 by todbot