node-hid/node-hid #118: Error: could not send feature report to device

folx, 2015-10-18:
Hi there,

Trying wearefractal/msi-keyboard on a fresh Funtoo install, using nodejs-4.1.1 & libusb-1.0.19, got me following stack trace:
.../msi-keyboard/examples $ node sample.js
.../msi-keyboard/lib/setColor.js:24
keyboard.sendFeatureReport(activate);
^

Error: could not send feature report to device
at Error (native)
at module.exports (.../msi-keyboard/lib/setColor.js:24:11)
at HID.board.color (.../msi-keyboard/lib/findKeyboard.js:12:3)
at Object. (.../msi-keyboard/examples/sample.js:3:10)
at Module._compile (module.js:434:26)
at Object.Module._extensions..js (module.js:452:10)
at Module.load (module.js:355:32)
at Function.Module._load (module.js:310:12)
at Function.Module.runMain (module.js:475:10)
at startup (node.js:117:18)

While I reported issue to wearefractal/msi-keyboard stevelacy points out that error occured at node-hid interface.
Other nodejs version worked just fine for me : tryed 0.12.6 & 0.12.7, both working perfectly.

dimitriscsd, 2016-03-11:
Same problem for me on node 5.8.0

todbot, 2016-03-11:
Are you running as root or have udev rules installed?  Also, if this is a keyboard, then you'll have to configure the kernel to release its hold on the device.

stevelacy, 2016-03-16:
The feature report works with the following versions:
`node v0.10.43`
`node v0.12.12`
`iojs v2.5.0`
All using `node-hid v 0.5.1`
Fails on `iojs v3` and all node versions after with the message `could not send feature report to device`. Root is required.

todbot, 2016-03-16:
Hi Steve,
I don't quite understand your statement.  It works with those node versions but also fails on those versions?

stevelacy, 2016-03-16:
It works on the versions prior to iojs v3

stevelacy, 2016-05-26:
Fixed - can close

folx, 2016-05-30:
Working with both node-0.12.6 & node-4.4.0 ,

Thx

closed 2016-05-30 by folx