node-hid/node-hid #119: Can't build node-hid on OSX El Capitan, node v4.2.1

OmgImAlexis, 2015-10-26:
[npm-debug.log](https://gist.github.com/OmgImAlexis/e3f0f88c2a9f508f471e)

````
➜  u2f-client git:(master) ✗ npm install
\
> node-hid@0.3.2 preinstall /Users/xo/code/u2f-client/node_modules/node-hid
> sh get-hidapi.sh

Cloning into 'hidapi'...
remote: Counting objects: 1965, done.
remote: Total 1965 (delta 0), reused 0 (delta 0), pack-reused 1965
Receiving objects: 100% (1965/1965), 2.80 MiB | 30.00 KiB/s, done.
Resolving deltas: 100% (1086/1086), done.
Checking connectivity... done.

> node-hid@0.3.2 install /Users/xo/code/u2f-client/node_modules/node-hid
> sh install.sh

  CC(target) Release/obj.target/hidapi/hidapi/mac/hid.o
../hidapi/mac/hid.c:255:20: warning: comparison of integers of different signs: 'CFIndex' (aka 'long') and 'size_t'
      (aka 'unsigned long') [-Wsign-compare]
                if (chars_copied == len)
                    ~~~~~~~~~~~~ ^  ~~~
../hidapi/mac/hid.c:295:20: warning: comparison of integers of different signs: 'CFIndex' (aka 'long') and 'size_t'
      (aka 'unsigned long') [-Wsign-compare]
                if (used_buf_len == len)
                    ~~~~~~~~~~~~ ^  ~~~
2 warnings generated.
  LIBTOOL-STATIC Release/hidapi.a
  CXX(target) Release/obj.target/HID/src/HID.o
../src/HID.cc:50:79: error: no member named 'New' in 'v8::String'
  virtual Handle<Value> asV8Exception() const { return ThrowException(String::New(message().c_str())); }
                                                                      ~~~~~~~~^
../src/HID.cc:57:12: error: expected class name
  : public ObjectWrap
           ^
../src/HID.cc:61:38: error: unknown type name 'Arguments'; did you mean 'v8::internal::Arguments'?
  static Handle<Value> devices(const Arguments& args);
                                     ^~~~~~~~~
                                     v8::internal::Arguments
/Users/xo/.node-gyp/4.2.1/include/node/v8.h:139:7: note: 'v8::internal::Arguments' declared here
class Arguments;
      ^
../src/HID.cc:76:34: error: unknown type name 'Arguments'; did you mean 'v8::internal::Arguments'?
  static Handle<Value> New(const Arguments& args);
                                 ^~~~~~~~~
                                 v8::internal::Arguments
/Users/xo/.node-gyp/4.2.1/include/node/v8.h:139:7: note: 'v8::internal::Arguments' declared here
class Arguments;
      ^
../src/HID.cc:77:35: error: unknown type name 'Arguments'; did you mean 'v8::internal::Arguments'?
  static Handle<Value> read(const Arguments& args);
                                  ^~~~~~~~~
                                  v8::internal::Arguments
/Users/xo/.node-gyp/4.2.1/include/node/v8.h:139:7: note: 'v8::internal::Arguments' declared here
class Arguments;
      ^
../src/HID.cc:78:36: error: unknown type name 'Arguments'; did you mean 'v8::internal::Arguments'?
  static Handle<Value> write(const Arguments& args);
                                   ^~~~~~~~~
                                   v8::internal::Arguments
/Users/xo/.node-gyp/4.2.1/include/node/v8.h:139:7: note: 'v8::internal::Arguments' declared here
class Arguments;
      ^
../src/HID.cc:79:36: error: unknown type name 'Arguments'; did you mean 'v8::internal::Arguments'?
  static Handle<Value> close(const Arguments& args);
                                   ^~~~~~~~~
                                   v8::internal::Arguments
/Users/xo/.node-gyp/4.2.1/include/node/v8.h:139:7: note: 'v8::internal::Arguments' declared here
class Arguments;
      ^
../src/HID.cc:80:45: error: unknown type name 'Arguments'; did you mean 'v8::internal::Arguments'?
  static Handle<Value> setNonBlocking(const Arguments& args);
                                            ^~~~~~~~~
                                            v8::internal::Arguments
/Users/xo/.node-gyp/4.2.1/include/node/v8.h:139:7: note: 'v8::internal::Arguments' declared here
class Arguments;
      ^
../src/HID.cc:81:47: error: unknown type name 'Arguments'; did you mean 'v8::internal::Arguments'?
  static Handle<Value> getFeatureReport(const Arguments& args);
                                              ^~~~~~~~~
                                              v8::internal::Arguments
/Users/xo/.node-gyp/4.2.1/include/node/v8.h:139:7: note: 'v8::internal::Arguments' declared here
class Arguments;
      ^
../src/HID.cc:83:48: error: unknown type name 'Arguments'; did you mean 'v8::internal::Arguments'?
  static Handle<Value> sendFeatureReport(const Arguments& args);
                                               ^~~~~~~~~
                                               v8::internal::Arguments
/Users/xo/.node-gyp/4.2.1/include/node/v8.h:139:7: note: 'v8::internal::Arguments' declared here
class Arguments;
      ^
../src/HID.cc:86:25: error: unknown type name 'uv_work_t'
  static void recvAsync(uv_work_t* req);
                        ^
../src/HID.cc:87:29: error: unknown type name 'uv_work_t'
  static void recvAsyncDone(uv_work_t* req);
                            ^
../src/HID.cc:177:16: error: unknown type name 'uv_work_t'
HID::recvAsync(uv_work_t* req)
               ^
../src/HID.cc:195:40: error: no member named 'New' in 'v8::String'
    argv[0] = Exception::Error(String::New(iocb->_error->message().c_str()));
                               ~~~~~~~~^
../src/HID.cc:200:16: error: no member named 'GetCurrent' in 'v8::Context'
      Context::GetCurrent()->Global()->Get(String::New("Buffer") )
      ~~~~~~~~~^
../src/HID.cc:200:52: error: no member named 'New' in 'v8::String'
      Context::GetCurrent()->Global()->Get(String::New("Buffer") )
                                           ~~~~~~~~^
../src/HID.cc:203:68: error: too few arguments to function call, expected 2, have 1
    Handle<Value> nodeBufferArgs[1] = { Integer::New(message.size()) };
                                        ~~~~~~~~~~~~               ^
/Users/xo/.node-gyp/4.2.1/include/node/v8.h:2499:3: note: 'New' declared here
  static Local<Integer> New(Isolate* isolate, int32_t value);
  ^
../src/HID.cc:215:20: error: unknown type name 'uv_work_t'
HID::recvAsyncDone(uv_work_t* req)
                   ^
../src/HID.cc:217:15: error: calling a protected constructor of class 'v8::HandleScope'
  HandleScope scope;
              ^
/Users/xo/.node-gyp/4.2.1/include/node/v8.h:885:13: note: declared protected here
  V8_INLINE HandleScope() {}
            ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make: *** [Release/obj.target/HID/src/HID.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/Users/xo/.nvm/versions/node/v4.2.1/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:270:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Darwin 15.0.0
gyp ERR! command "/Users/xo/.nvm/versions/node/v4.2.1/bin/node" "/Users/xo/.nvm/versions/node/v4.2.1/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "configure" "build" "install"
gyp ERR! cwd /Users/xo/code/u2f-client/node_modules/node-hid
gyp ERR! node -v v4.2.1
gyp ERR! node-gyp -v v3.0.3
gyp ERR! not ok 
npm ERR! Darwin 15.0.0
npm ERR! argv "/Users/xo/.nvm/versions/node/v4.2.1/bin/node" "/Users/xo/.nvm/versions/node/v4.2.1/bin/npm" "install"
npm ERR! node v4.2.1
npm ERR! npm  v2.14.7
npm ERR! code ELIFECYCLE

npm ERR! node-hid@0.3.2 install: `sh install.sh`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-hid@0.3.2 install script 'sh install.sh'.
npm ERR! This is most likely a problem with the node-hid package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     sh install.sh
npm ERR! You can get their info via:
npm ERR!     npm owner ls node-hid
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /Users/xo/code/u2f-client/npm-debug.log
````

todbot, 2015-10-26:
It looks like you're trying to build a very old version of node-hid (v0.3.2).   The current version is v0.5.1.

OmgImAlexis, 2015-10-27:
@todbot ah, I'm trying to update `u2f-client`, that'd explain the errors. I'll try updating it and see if that fixes it all.

closed 2015-10-27 by OmgImAlexis