node-hid/node-hid #120: Is there a reason usage and usagePage was was taken out?

OmgImAlexis, 2015-10-27:
UsagePage is used in a repo I'm trying to update and I don't see any reason why it was taken out. 

Ref: https://github.com/node-hid/node-hid/commit/3cc5b0a968633c5b302c57d0df970619071b4f1c

todbot, 2015-10-27:
Good question.  I don't know (before I was working on node-hid). I'll try adding it in, do some minor tests, and check in to master (but not publish to npm) so you and others can try it out.

todbot, 2015-10-27:
Oh I see your commit where you did the same thing I was about to do.

It looks like usage and usagePage might be Mac & Windows only. I'm poking around the underlying hidapi C library now to see what's up.

OmgImAlexis, 2015-10-27:
@todbot okay sounds good. Also from what I can tell my crash is from when it attempts to send bytes to the device. I actually get a full kernel panic that crashes my Macbook. :cry:

amadsen, 2016-09-24:
I'm dealing with the same challenge @OmgImAlexis was. I'm still mostly in the dark, but this is something I have found so far:

Looking at [hidapi's libusb](https://github.com/signal11/hidapi/blob/master/libusb/hid.c), there's an explanation of why enumeration may not contain usage page and usage (lines 598-611):

    This section is removed because it is too invasive on the system. Getting a Usage Page
    and Usage requires parsing the HID Report descriptor. Getting a HID Report descriptor
    involves claiming the interface. Claiming the interface involves detaching the kernel driver.
    Detaching the kernel driver is hard on the system because it will unclaim interfaces (if another
    app has them claimed) and the re-attachment of the driver will sometimes change /dev entry
    names. It is for these reasons that this section is #if 0. For composite devices, use the interface
    field in the hid_device_info struct to distinguish between interfaces.

It seems to be a compile time flag, but the concern makes sense. Also - if I understand correctly -getting the first usage page and usage takes a lot more work during every enumerate and isn't isn't entirely accurate for some multifunction devices. If the values are not initialized they might point at anything by default; I could imagine sending bytes to them causing a kernel panic.

Unfortunately for me (and @OmgImAlexis, unless you've given up) the FIDO U2F spec requires checking the usage page (and usage) to see if it matches the one defined for U2F. I'm looking for information on how to get the usage page so I can cache the results of device enumerations and then pull the HID Report Descriptor just for new devices. This is very much a learn as I go sort of thing, so if you have any pointers I would appreciate it.

amadsen, 2016-11-12:
Just FYI, the recent release with the updated hidapi version does provide a unique path for multifunction devices, but it does not seem to address the missing usage and usagePage. The release notes sort of make it sound like it does because the underlying hidapi issue had a pull request (that ultimately was not used) that employed the usage and usagePage to make the path more unique.

Do you know if adding the usage and usagePage back (as in @OmgImAlexis' pull request) happens to work now?

todbot, 2016-11-12:
I'm testing adding it back in now. From what I can tell, `usage` and `usagePage` will only be available on Mac & Windows.  They are not available on Linux, as the underlying hidapi implementation for Linux (really libusb) doesn't access that info.

Also, thank you about 0.5.2 release notes. I've updated them to be more accurate. Apologies for that.

amadsen, 2016-11-13:
I'm might put a conditional around the usage and usagePage similar to some of the other items
~~~
if (dev->usage) {
      deviceInfo->Set(Nan::New<String>("usage").ToLocalChecked(), Integer::New(dev->usage_page));
    }
~~~

That way `usage` and `usagePage` will be `undefined` and fail a `hasOwnProperty()` check if it isn't there.

I'm curious if we can figure out how to get the `usage` and `usagePage` on linux - I'm going to look at how Yubico's u2f-host library does it. It's also using hidapi.

I also wish hidapi had a version of enumerate that let me specify characteristics of the devices I'm looking for (similar to OSX's [IOHIDManagerSetDeviceMatching](https://developer.apple.com/reference/iokit/1438371-iohidmanagersetdevicematching?language=objc)). While we could do something like that here, it probably belongs in the underlying library. The primary reason being you can pass the usage under the [kIOHIDDeviceUsagePageKey](https://developer.apple.com/reference/iokit/kiohiddeviceusagepagekey) and it will find all devices that support that key, rather than just returning the [kIOHIDPrimaryUsageKey](https://developer.apple.com/reference/iokit/kiohidprimaryusagekey) as hidapi currently does.

amadsen, 2016-11-13:
https://github.com/signal11/hidapi/issues/298 records that wish for posterity.

amadsen, 2016-11-13:
https://github.com/Yubico/libu2f-host/blob/master/u2f-host/devs.c uses `hidraw.h` to patch in the `usage` and `usagePage` on linux. They appear to be using the hidraw backend to hidapi as well though, while node-hid is using libusb.

In theory we could consider doing something similar for kernels that support hidraw - we're 5 or 6 years in to having linux kernel support. The code they are using to parse the usage and usage page from the report descriptor is very similar to code from the hidapi libusb backend - it varies mostly in how it retrieves the report descriptor. I don't *think* the hidraw-based code has to detach the kernel driver because it is using OS provided data, but I could be wrong about that.

As far as I can tell, the hid_device_info struct has usage and usagePage entries regardless of the OS and backend, it's just a matter of whether these are populated. Hopefully that means we can easily add them back for the environments that support it.

amadsen, 2016-11-13:
I noticed you committed to master. I can verify that on OSX I can now use the usage and usagePage if I depend on that latest code. I've done no testing of other platforms though.

todbot, 2016-11-13:
Yes, and thanks for the suggestion about the conditionals on usage & usage_page.

I did some research into adding optional hidraw support to the Linux build of node-hid and found a nice surprise: it's already in the binding.gyp.  

There is a node-gyp variable called `driver` that defaults to `libusb` but can be set to `hidraw`.

So all you have to do to get usage & usagePage on Linux is to build with the hidraw driver. And you do that with:
```
   npm install node-hid --driver=hidraw 
```

Until I publish 0.5.3 with the usagePage additions, you'll need to install the repo-direct version with:
```
npm install github:node-hid/node-hid --driver=hidraw
```

If using node-hid in a project, you can also use `--driver=hidraw` in your main `npm install`.  As far as I can tell, there's no way to persist this setting in package.json.

(the above are notes mostly to me for when I update the README)

closed 2017-05-22 by OmgImAlexis