node-hid/node-hid #122: sendFeatureReport does not support Buffer as argument

bminer, 2015-11-06:
`sendFeatureReport` works just fine when an Array is passed, but when a Node Buffer is passed, node-hid chokes.  This should probably be fixed.

bminer, 2015-11-06:
Any thoughts on the best way to do this?  Just convert the Buffer to an Array before passing to the native addon?  What's the overhead penalty for doing so?

underbluewaters, 2016-06-08:
Same api usage is relevant for `write`. I'm finding it more natural to use buffers when modifying messages. It's an easy thing to create an array of bytes, but it can complicate code that I'd prefer to focus on protocol logic.

todbot, 2016-06-08:
Is there another library that uses Buffers in a way you find particularly good? I'm not too familiar with them and would like to follow best practice.

underbluewaters, 2016-06-08:
[node-crc](https://github.com/alexgorbatchev/node-crc/blob/master/src/crc16.js) converts all inputs to buffers and uses them internally. I can't really offer any constructive ideas because I'm not familiar with using c++ bindings.

What I'm doing, is using this function to convert my buffer before passing to write.
```javascript
module.exports = function(buffer) {
  return buffer.toString("hex").match(/.{1,2}/g).map(function(val){
    return parseInt("0x" + val)
  });
}
```

todbot, 2016-06-08:
This is mostly just a note for myself, but it looks like you can also convert a Buffer to an array of numbers by doing `buffer.toJSON.data`, e.g.: 
```
var a = [0xcc, 0xaa, 0xff, 0xee];
var b = new Buffer( a );
var c = b.toJSON().data;
```

underbluewaters, 2016-06-09:
Nice. I'll use that. I've also seen `toString().charCodeAt(n)` work with the right encoding.

amadsen, 2016-11-23:
A project I am trying to use attempts to use node-hid with node Buffers instead of Arrays. Buffers provide a nice API for preparing binary data, so I'd prefer to continue to use them. Consequently, I am working on a pull request to address this issue.

I see 3 possible paths forward:
1. convert Buffers to Arrays before passing the data to C++, as discussed here
2. convert Arrays to buffers before passing data to C++
3. provide separate, but equivalent C++ methods for Arrays and Buffers

Option 3 feels messy and likely to cause maintenance headaches. Option 1 reuses existing native code, but may be less efficient both in terms of code an memory. However, these disadvantages are not really likely to be that significant for most Human Interface Devices. I am pursuing Option 3, since node has Buffer constructors (and factories) for creating a Buffer from and array or a Buffer. Also, the process of converting a C++ v8::Buffer object to a char* buffer appears to be very simple. I'll be very interested in getting feedback and testing when the pull request is ready.

todbot, 2016-11-23:
Hi @amadsen,

That sounds great!  I welcome help on this. I looked into it when this issue was first opened and it was too big a change at the time.  I came to similar conclusions as yourself but since I had both no experience with Buffers nor a project that needed them, I didn't pursue any changes to node-hid. Let me know how I can help going forward.

amadsen, 2016-11-23:
Thanks @todbot!

I've got an version that I think works, but it isn't tested for backward compatibility (not did I write tests - I just used it with my fork/branch of u2f-client. I don't have the devices to test the existing test cases.

https://github.com/node-hid/node-hid/pull/171

There is still a bunch of commented out code and some unnecessary logging in there too. We should clean this sort of thing up before merging.

amadsen, 2016-12-23:
Following up on this - https://github.com/node-hid/node-hid/pull/171 continues to work well for me on OSX (Sierra) using buffers. I'd very much appreciate people testing on other platforms, particularly using Arrays and providing feedback.

mcuee, 2023-05-04:
Since #357 has been merged, I think this issue can be closed.

closed 2023-05-26 by bminer