node-hid/node-hid #125: Different results on mac and raspberry

greenkohl23, 2015-12-01:
Hi,
I use a mac to build my node applications so i paired a remote shutter (for selfies) over bluetooth with my mac. Its  connected as a keyboard. If I click it sends  4 commands to the mac. 
So I take node-hid to read its commands. No problem everything works.

So I switched to the raspberry pi and after I pair the shutter with my raspberry via bluetooth, but if i perform hid.hid.devices() it shows me nothing. 

Is there anything I missed?
I also posted a question on stackoverflow: https://stackoverflow.com/questions/34028229/node-hid-get-different-result-on-raspbian-and-macos-for-hid-devices

Hope you can help me. I struggle with this since three days now ;(

todbot, 2015-12-01:
Have you tried running as 'root'?

greenkohl23, 2015-12-02:
Hi,
thx for your reply. No I started my node code with sudo and i tried to set up a udev rule. i will try it later as sudo.

greenkohl23, 2015-12-02:
I started the node script as root, but nothing changed.

greenkohl23, 2015-12-06:
Has anyone an idea, what could be wrong? I tested already 2 different bluetooth adapters and i setup my pi from scratch again. I see an output, when i use cat /dev/input/event0 and click on the device.

todbot, 2016-03-28:
node-hid only really works with USB HID devices, not Bluetooth HID devices.  Check out the noble and related packages for Bluetooth connectivity. Closing this issue for now.

closed 2016-03-28 by todbot