node-hid/node-hid #126: No devices returned Windows 8.1

Thelmos, 2015-12-10:
`node-hid\src\show-devices.js` sample returns no devices, Windows device manager lists the two HID devices I have currently connected, an optical mouse and a magnetic card reader.

**System:** 
* Windows 8.1 Enterprise 64 bits
* Node v4.2.2
* npm 3.4.0

It's easy to reproduce, install node-hid in a new folder:
> C:\test>npm install node-hid
> ...
> C:\test
>└─┬ node-hid@0.5.1
>     └── nan@2.1.0

And launch sample:

>C:\test\node_modules\node-hid>node src/show-devices.js
>devices: []

It seems binaries are correctly compiled:

>c:\test\node_modules\node-hid\build\HID\v0.5.1\Release\node-v46-win32-x64\HID.node 

And `binding_path` in `nodejid.js` correctly resolved.

Also tested in other system with same result:
* Windows 10 Home 32 bits
* Node v4.2.1
* npm 2.14.7

I can't find the problem and I'm totally stuck with this, any help would be appreciated.

Thanks.

Thelmos, 2015-12-11:
Problem tested directly with *signal11 hidapi library*, provided tests also fail listing devices.

So it's not really a node-hid problem.

Thelmos, 2015-12-11:
**Solved!**

Hidapi mantainer coments in this issue: https://github.com/signal11/hidapi/issues/247

>"HIDAPI won't work with keyboards and mice on Windows."

It seems there are security restrictions with keyboards on Windows, and my magnetic card reader was acting as a keyboard for the OS.

Changed device mode from *HID Keyboard* to *HID Compatilble* using card reader software and now node-hid works like a charm.

Hope this can help someone with same problem.

Thanks.

closed 2015-12-11 by Thelmos

todbot, 2015-12-11:
Thanks for doing that debugging.  I'll add a note about keyboards & mice to the README.  This has been a problem with other OSes as well.

sintaru, 2016-04-29:
@Thelmos  I am having a similar issue with my device 

hid_enumerate() returns NULL for my device. I used my device's VID and PID . 

if (hid_init())
		return -1;

	devs = hid_enumerate(0x0483, 0x09d7);
	cur_dev = devs;	

hid_open() throws the error saying "unable to open device"

Any thoughts ? 

Thanks.

todbot, 2016-04-29:
What kind of device is this?  If it's something that identifies as mouse or keyboard, node-hid can't get at it on Windows because Windows owns all HID devices of those types.  If your device is your own creation, I recommend giving it a HID usage page different from one of these:
https://msdn.microsoft.com/en-us/library/windows/hardware/ff543477(v=vs.85).aspx