node-hid/node-hid #130: Error handler instead of throwing exception when open device fails

pravincar, 2016-01-23:
Can you add a error handler as the last parameter in the constructor so that the exception can be handled when open device fails?

todbot, 2016-01-25:
Hi,

This sounds like a good idea.  I'll look into how to add it.

amadsen, 2016-11-15:
Perhaps this would be most easily implemented as a factory function that simply wraps the constructor with a try/catch. That way the constructor doesn't require significant modification, but the feature request can be met.

Something like (untested code):

~~~javascript
exports.open = function () {
	var args = [].slice.call(arguments),
		// the last argument MUST be a callback
		cb = args.pop(),
		device;

	if('function' !== typeof cb){
		throw new TypeError('HID.open() requires a callback!');
	}

	setImmediate(function(){
		try {
			// HID is a function. Use bind to make it a more speciallized function
			// by applying our args to it with a null context. Then call that as a
			// constructor using new.
			device = new HID.bind.apply(null, args);
			return cb(null, device);
		} catch(e) {
			return cb(e);
		}
	});
}
~~~