node-hid/node-hid #132: Strings like the product string should be available on an opened device

j-waechter, 2016-01-25:
Hello,

I would like to get the strings like the product string from an opened device. Right now, you can only retrieve those strings from the `devices()` function.

[`HID.cc`, line 468](https://github.com/node-hid/node-hid/blob/master/src/HID.cc#L468) shows that these strings should be:

- `path`
- `serialNumber`
- `manufacturer`
- `product`
- `release`
- `interface`

todbot, 2016-01-25:
Hi,

This is a good suggestion, something I've wanted myself.  I forget why but I thought that info wasn't easily available on a hidapi device handle (and thus would mean caching the result of hid_enumerate())

I might see a way of doing this now so I'll look into it.  (or if you're good with C and want to take a stab at it yourself, please do)

j-waechter, 2016-01-25:
I've taken a look into the code and decided to rather create an issue than try it myself. I don't find it straightforward.

PS: Thanks for this great and simple library!

todbot, 2016-02-23:
Hi @j-waechter, I'm attempting to implement this now, but I see that only some of the information is available on an open device.  (The underlying API that node-hid doesn't provide the functionality).  

Currently I know I can get manufacturer string, product string, and serial number.  
What specific information were you looking for?

kbohm, 2016-02-23:
Hello! The manufacturer and product strings were what I was actually looking for.

If there's more to get that's coming along the way while opening a device, couldn't this just get cached (e.g. the device path)?

todbot, 2016-02-23:
The deviceInfos can't get cached efficiently, at least with the current underlying hid-api library.  The `HID.devices()` call causes an explicit USB bus enumeration, which can be costly on some OSes. A direct `new HID.HID(path)` does not cause a bus enumeration and so is much faster for some uses.  (But unfortunately, `new HID.HID(vid,pid)` *does* cause an underlying enumeration in hid-api, but hid-api doesn't save or pass back the result)

So we could re-architect node-hid a bit to support deviceInfo caching for the very common `new HID.HID(vid,pid)` case but that's a bit of work (do we need to re-cache on device exception? only on 'new HID.HID(vid,pid)`?)  I'll need to think on the caching issue a bit more.  The above change I made was a slam dunk and doesn't affect existing use cases.

j-waechter, 2016-02-23:
Okay, I didn't mean to trigger so much work. To be honest, the manufacturer and product strings would be sufficient for me, as I can tell now.

todbot, 2016-02-23:
Not a problem.  This aspect of hid-api has been bugging me in my C projects for a few years now. :-)

mcuee, 2023-05-04:
#499 has been merged. I guess this issue can be closed now.

Julusian, 2023-05-05:
#499 isn't merged yet

mcuee, 2023-05-05:
Oops, sorry I made a mistake.

PR #499 (2023-05-05): Feat/expand getdeviceinfo
Closes #132

This expands on #474 with some new functionality. I thought it better to do as a separate PR to avoid making #474 even larger.

Only the last couple of commits are relevant here, once #474 is merged the changes involved here will become a lot clearer.

This update hidapi to 0.13.1, and uses the new `hid_get_device_info` method to add more properties to the `getDeviceInfo` methods. `getDeviceInfo` will now return the same content as is provided by a call to `HID.devices()`.

My usecase for this is that in https://github.com/Julusian/node-elgato-stream-deck [... cut]
closed 2023-05-05 by todbot