node-hid/node-hid #134: Error: cannot open device with vendor id 0x458 and product id 0x16c

ivanchoff, 2016-02-16:
Hi!!... i have this problem that i don't know hot to resolve.
i can list my hid devices:
```
devices: [ { vendorId: 1112,
    productId: 364,
    path: '0002:0004:00',
    release: 821,
    interface: 0 },
  { vendorId: 1112,
    productId: 364,
    path: '0002:0004:01',
    release: 821,
    interface: 1 },
  { vendorId: 1112,
    productId: 364,
    path: '0002:0004:02',
    release: 821,
    interface: 2 } ]
```

so i can see that vendorId is 1112 and productId is 364, i run this simple code: 

```
var HID = require('./node-hid/nodehid.js');
var devices = new HID.HID(1112,364);
console.log('devices:', HID.devices());
```

and get this error:

```
/home/choff/hid/node-hid/nodehid.js:28
    this._raw = new (Function.prototype.bind.apply(binding.HID,
                ^

Error: cannot open device with vendor id 0x458 and product id 0x16c
    at Error (native)
    at new HID (/home/choff/hid/node-hid/nodehid.js:28:17)
    at Object.<anonymous> (/home/choff/hid/test_hid.js:5:15)
    at Module._compile (module.js:413:34)
    at Object.Module._extensions..js (module.js:422:10)
    at Module.load (module.js:357:32)
    at Function.Module._load (module.js:314:12)
    at Function.Module.runMain (module.js:447:10)
    at startup (node.js:139:18)
    at node.js:999:3
```

todbot, 2016-02-17:
Hi,
What kind of device is it you are trying to open and what OS are you on?  On Linux you need to run the script as 'root' unless you have a udev entry for the device

ivanchoff, 2016-02-19:
hi!!, my device is a "Genius" presenter runing on GNU/linux... runing script with sudo works!!! 
thanks you :+1:

closed 2016-02-19 by ivanchoff

toxaq, 2016-05-06:
Seems related https://github.com/signal11/hidapi/issues/266 Particularly https://github.com/signal11/hidapi/issues/266#issuecomment-203097328

SalMinoz, 2019-11-05:
@todbot Hi! I am trying to open QR Code Scanner device but not able to open successfully. My current platform is Window. Could you please help me to figure it out???