node-hid/node-hid #138: Node 5

ForestMist, 2016-03-28:
Has anyone tried using Node 5?

Here is what happens with Node 5.7.0 on Mac OS 10.11.4 with the latest version of Xcode.

```
> node-hid@0.5.1 install /Users/daniel/Dropbox/Projects/node/logitech-g29/node_modules/node-hid
> node-pre-gyp install --fallback-to-build

  CC(target) Release/obj.target/hidapi/hidapi/mac/hid.o
../hidapi/mac/hid.c:255:20: warning: comparison of integers of different
      signs: 'CFIndex' (aka 'long') and 'size_t' (aka 'unsigned long')
      [-Wsign-compare]
                if (chars_copied == len)
                    ~~~~~~~~~~~~ ^  ~~~
../hidapi/mac/hid.c:295:20: warning: comparison of integers of different
      signs: 'CFIndex' (aka 'long') and 'size_t' (aka 'unsigned long')
      [-Wsign-compare]
                if (used_buf_len == len)
                    ~~~~~~~~~~~~ ^  ~~~
2 warnings generated.
  LIBTOOL-STATIC Release/hidapi.a
  CXX(target) Release/obj.target/HID/src/HID.o
  SOLINK_MODULE(target) Release/HID.node
  COPY /Users/daniel/Dropbox/Projects/node/logitech-g29/node_modules/node-hid/build/HID/v0.5.1/Release/node-v47-darwin-x64/HID.node
  TOUCH Release/obj.target/action_after_build.stamp
```

todbot, 2016-03-28:
Hi,
Those are warnings, not errors, and are in `hidapi`, the underlying C-library `node-hid` uses.  They've existed forever and do not affect `node-hid`.  They're just annoying to see.

ForestMist, 2016-03-29:
Ah, that makes perfect sense and with that in mind I found an issue in my code that I thought was related to the `hidap` warnings but is not. Oopsie.

Thank you so much for explaining this Tod. I really appreciate it. ^_^

closed 2016-03-29 by ForestMist