node-hid/node-hid #139: Can not work on Window 10 64bit, node 5.6.0

coolinker, 2016-03-30:
Hi,

I'm using thinkpad laptop, win10 64bit and node5.6.0. When I plug my Logitech usb wireless mouse/keyboard, HID.devices() can list 6 more devices(Not sure why this many?):
[ { vendorId: 1133,
    productId: 50478,
    path: '\\\\?\\hid#vid_046d&pid_c52e&mi_00#7&e246acd&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}',
    manufacturer: 'Logitech',
    product: 'USB Receiver',
    release: 5376,
    interface: 0 },
  { vendorId: 1133,
    productId: 50478,
    path: '\\\\?\\hid#vid_046d&pid_c52e&mi_01&col01#7&31fba88f&3&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}',
    manufacturer: 'Logitech',
    product: 'USB Receiver',
    release: 5376,
    interface: 1 },
  { vendorId: 1133,
    productId: 50478,
    path: '\\\\?\\hid#vid_046d&pid_c52e&mi_01&col02#7&31fba88f&3&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}',
    manufacturer: 'Logitech',
    product: 'USB Receiver',
    release: 5376,
    interface: 1 },
  { vendorId: 1133,
    productId: 50478,
    path: '\\\\?\\hid#vid_046d&pid_c52e&mi_01&col03#7&31fba88f&3&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}',
    manufacturer: 'Logitech',
    product: 'USB Receiver',
    release: 5376,
    interface: 1 },
  { vendorId: 1133,
    productId: 50478,
    path: '\\\\?\\hid#vid_046d&pid_c52e&mi_01&col04#7&31fba88f&3&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}',
    manufacturer: 'Logitech',
    product: 'USB Receiver',
    release: 5376,
    interface: 1 },
  { vendorId: 1133,
    productId: 50478,
    path: '\\\\?\\hid#vid_046d&pid_c52e&mi_01&col05#7&31fba88f&3&0004#{4d1e55b2-f16f-11cf-88cb-001111000030}',
    manufacturer: 'Logitech',
    product: 'USB Receiver',
    release: 5376,
    interface: 1 } ]

Then I found the first 3 devices can not be opened, I got:
       this._raw = new (Function.prototype.bind.apply(binding.HID,
                    ^
Error: cannot open device with path \\?\hid#vid_046d&pid_c52e&mi_01&col02#7&31fba88f&3&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
    at Error (native)
    at new HID (C:\Users\Jake\AppData\Roaming\npm\node_modules\node-hid\nodehid.js:28:14)
    at Object.<anonymous> (D:\works\bitbucket\tako\ffitest.js:7:26)
    at Module._compile (module.js:413:34)
    at Object.Module._extensions..js (module.js:422:10)
    at Module.load (module.js:357:32)
    at Function.Module._load (module.js:314:12)
    at Function.Module.runMain (module.js:447:10)
    at startup (node.js:140:18)
    at node.js:1001:3

But for the last 3 devices, no output when I run code:
var deviceInstance = new HID.HID(device.path);
    deviceInstance.on("data", function(data) {
        console.log(data);
    });
    deviceInstance.on("error", function(error) {
        console.log("Error on device "+device.product+": ");
        console.log(error);
    });

But when I run:
var device = new HID.HID('\\\\?\\hid#vid_046d&pid_c52e&mi_01&col05#7&31fba88f&3&0004#{4d1e55b2-f16f-11cf-88cb-001111000030}');
device.write([0, 0, 22, 0, 0, 0, 0, 0]);

I got error:
device.write([0, 0, 22, 0,0,0,0,0]);
       ^
Error: Cannot write to HID device
    at Error (native)
    at Object.<anonymous> (D:\works\bitbucket\tako\ffitest.js:9:8)
    at Module._compile (module.js:413:34)
    at Object.Module._extensions..js (module.js:422:10)
    at Module.load (module.js:357:32)
    at Function.Module._load (module.js:314:12)
    at Function.Module.runMain (module.js:447:10)
    at startup (node.js:140:18)
    at node.js:1001:3

I'm running in the administrator command line window, can you please help?

Thanks,
Jake

coolinker, 2016-03-30:
Sorry, just got some new clue, let me do more test and I will update it here later.

Thanks,
Jake

closed 2016-04-04 by coolinker

xiajingren, 2020-09-29:
I have similar problems, what is the final solution?

todbot, 2020-09-29:
Hi @xiajingren, you cannot access mice or keyboards with `node-hid`, as those devices are owned by the OS, for security reasons.  What is it you're trying to accomplish?

xiajingren, 2020-09-29:
Hi @todbot 
Thank you very much for your response. We have developed a keyboard device. I want to use node hid to communicate with the device, read and write. Can't I?

xiajingren, 2020-09-29:
At the moment, I just want to test the library with my Logitech mouse. Is this allowed?

zlt995413176, 2020-09-29:
@xiajingren

todbot, 2020-09-29:
Hi @xiajingren if you want to "spy" on the mouse as it moves or clicks, then no.  You generally cannot do that as the OS prevents it.  If you want to control some non-pointer features of the mouse (e.g. mouse LED lighting), then maybe. 
 The device's HID Report Descriptor may have specified separate report described with non-pointer `usagePage`/`usage` attributes.  These other reports may not be taken by the OS and thus usable by `node-hid`.

The USB HID Usage Pages are described here: https://www.usb.org/sites/default/files/documents/hut1_12v2.pdf

The `usagePage` and `usage` properties are shown in `hid-showdevices`, or with something like this:
```js
var HID = require('node-hid')
var devices = HID.devices()
for (d of devices) {
    console.log("vid/pid:usagePage/usage",d.vendorId,"/",d.productId, ": ", d.usagePage,",",d.usage)
}
```

xiajingren, 2020-09-29:
thank you @todbot 
Sorry, I just started hid. I want to use hid to send some configuration information to my keyboard and read the keyboard configuration. I need to create another `usagePage` and `usage` on the device for communication. Is this allowed?

Please forgive my poor English :)

todbot, 2020-09-29:
Hi @xiajingren, Not a problem!  Some of these HID concepts are difficult.  

Each USB HID device provides to the computer a "USB HID Report Descriptor" that describes the type of data packets ("reports") the device can send and receive.  These "reports" are classified by a "usagePage" and "usage" attribute.  Some of these data packets, based on their usagePage, will be owned by the OS for security reasons (e.g. you don't want random apps snooping on keypresses)

The list of defined usagePages is in that "hut1_12v2.pdf" document above. When you use `node-hid` to list devices, you will actually see multiple entries for a single device, one per usage. 

So in `node-hid` programs, you will often see a `filter` being applied to get the specific usagePage & usage "device", using something like:
```js
var devices = HID.devices(); // get all devices
var deviceInfo = devices.find( function(d) {
    return (d.usagePage===0xFFAB && d.usage===0x2000); // I'm looking for usagePage/usage: 0xFFAB/0x2000
});
if( deviceInfo ) { 
    device = new HID.HID( deviceInfo.path );
}

```
See for example, https://github.com/node-hid/node-hid/blob/master/src/test-teensyrawhid.js

In order to know what usagePage/usage you want to open, you need to know the USB HID implementation of your device.

xiajingren, 2020-09-30:
Hi @todbot , Thank you very much! Your answer is very helpful to me. I will try it. If there are any questions, I will come back to ask you.