node-hid/node-hid #140: Cannot open device despite udev configuration

pimartin, 2016-04-02:
Hi,
I've seen a few issues about this problem but every time the answer was to use `sudo` or to configure the rights using udev.

In my case, `sudo` isn't really an option so I tried to configure udev with this rule:
> SUBSYSTEM=="input", ATTRS{idVendor}=="256c", ATTRS{idProduct}=="006e", MODE="0666"

(I know 666 is unsafe)
It seems to work and I can see the correct rights on the corresponding /dev/input/event* :
> crw-rw-rw-   1 root input 13, 80 Apr  2 18:50 event16
> crw-rw-rw-   1 root input 13, 81 Apr  2 18:50 event17
> crw-rw-rw-   1 root input 13, 82 Apr  2 18:50 event18
> crw-rw-rw-   1 root input 13, 83 Apr  2 18:50 event19
> crw-rw-rw-   1 root input 13, 84 Apr  2 18:50 event20

Using `cat` on them works without `sudo` but when I try to open this device (identified with the vendorId and productId) using node-hid, I get this error :
> Error: cannot open device with path 0009:0003:02

(I also placed my user in the input group for good measure.)

Could you yell me where this comes from or what I screwed up please ?

reinzor, 2016-05-26:
I have the same issue; cat the device with current user works find but my node program has to run as root in order to get it to work.

maghis, 2016-05-27:
Same here, gave a device read permission from my group with an udev rule.
/dev/ link shows correct permissions and is accessible without sudo but node-hid says "Error: cannot open device with path 0003:0025:00".

todbot, 2016-11-13:
when changing the udev rules, are you then running `sudo udevadm control --reload-rules`

hphde, 2017-02-17:
Try
`SUBSYSTEM=="input", ATTRS{idVendor}=="256c", ATTRS{idProduct}=="006e", OWNER="root", GROUP="users", MODE="0660", SYMLINK+="huion540"`

Are you running libusb or hidraw? Try `npm install node-hid --driver=hidraw`

closed 2019-07-25 by todbot