node-hid/node-hid #142: node-webkit build does not work

brandonros, 2016-05-15:
Windows 10, 64-bit

Python 2.7.11

Visual Studio C++ 2015 installed

```
C:\Users\Brandon\Downloads\nwjs-sdk-v0.15.0-rc2-win-x64\nwjs-sdk-v0.15.0-rc2-win-x64\node_modules\node-hid>node-pre-gyp rebuild --runtime=node-webkit --target=0.15.0-rc2
node-pre-gyp info it worked if it ends with ok
node-pre-gyp info using node-pre-gyp@0.6.28
node-pre-gyp info using node@6.1.0 | win32 | x64
[node-hid] Removing "C:\Users\Brandon\Downloads\nwjs-sdk-v0.15.0-rc2-win-x64\nwjs-sdk-v0.15.0-rc2-win-x64\node_modules\node-hid\build\HID\v0.5.1\Release\node-webkit-v0.15.0-rc2-win32-x64"
gyp info it worked if it ends with ok
gyp info using node-gyp@3.3.1
gyp info using node@6.1.0 | win32 | x64
gyp info ok
gyp info it worked if it ends with ok
gyp info using nw-gyp@0.13.0
gyp info using node@6.1.0 | win32 | x64
(node:180) fs: re-evaluating native module sources is not supported. If you are using the graceful-fs module, please update it to a more recent version.
gyp http GET http://node-webkit.s3.amazonaws.com/v0.15.0-rc2/nw-headers-v0.15.0-rc2.tar.gz
gyp http 200 http://node-webkit.s3.amazonaws.com/v0.15.0-rc2/nw-headers-v0.15.0-rc2.tar.gz
gyp http GET http://node-webkit.s3.amazonaws.com/v0.15.0-rc2/node.lib
gyp http GET http://node-webkit.s3.amazonaws.com/v0.15.0-rc2/x64/node.lib
gyp http GET http://node-webkit.s3.amazonaws.com/v0.15.0-rc2/nw.lib
gyp http GET http://node-webkit.s3.amazonaws.com/v0.15.0-rc2/x64/nw.lib
gyp http 200 http://node-webkit.s3.amazonaws.com/v0.15.0-rc2/node.lib
gyp http 200 http://node-webkit.s3.amazonaws.com/v0.15.0-rc2/x64/node.lib
gyp http 200 http://node-webkit.s3.amazonaws.com/v0.15.0-rc2/nw.lib
gyp http 200 http://node-webkit.s3.amazonaws.com/v0.15.0-rc2/x64/nw.lib
gyp info spawn python
gyp info spawn args [ 'C:\\Users\\Brandon\\AppData\\Roaming\\npm\\node_modules\\nw-gyp\\gyp\\gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'msvs',
gyp info spawn args   '-G',
gyp info spawn args   'msvs_version=auto',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\Brandon\\Downloads\\nwjs-sdk-v0.15.0-rc2-win-x64\\nwjs-sdk-v0.15.0-rc2-win-x64\\node_modules\\node-hid\\build\\config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\Brandon\\AppData\\Roaming\\npm\\node_modules\\nw-gyp\\addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\Brandon\\.nw-gyp\\0.15.0-rc2\\common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=C:\\Users\\Brandon\\.nw-gyp\\0.15.0-rc2',
gyp info spawn args   '-Dmodule_root_dir=C:\\Users\\Brandon\\Downloads\\nwjs-sdk-v0.15.0-rc2-win-x64\\nwjs-sdk-v0.15.0-rc2-win-x64\\node_modules\\node-hid',
gyp info spawn args   '--depth=.',
gyp info spawn args   '-Dv13=1',
gyp info spawn args   '--generator-output',
gyp info spawn args   'C:\\Users\\Brandon\\Downloads\\nwjs-sdk-v0.15.0-rc2-win-x64\\nwjs-sdk-v0.15.0-rc2-win-x64\\node_modules\\node-hid\\build',
gyp info spawn args   '-Goutput_dir=.' ]
(node:180) DeprecationWarning: child_process: options.customFds option is deprecated. Use options.stdio instead.
Traceback (most recent call last):
  File "C:\Users\Brandon\AppData\Roaming\npm\node_modules\nw-gyp\gyp\gyp_main.py", line 18, in <module>
    sys.exit(gyp.script_main())
  File "C:\Users\Brandon\AppData\Roaming\npm\node_modules\nw-gyp\gyp\pylib\gyp\__init__.py", line 534, in script_main
    return main(sys.argv[1:])
  File "C:\Users\Brandon\AppData\Roaming\npm\node_modules\nw-gyp\gyp\pylib\gyp\__init__.py", line 527, in main
    return gyp_main(args)
  File "C:\Users\Brandon\AppData\Roaming\npm\node_modules\nw-gyp\gyp\pylib\gyp\__init__.py", line 503, in gyp_main
    options.circular_check)
  File "C:\Users\Brandon\AppData\Roaming\npm\node_modules\nw-gyp\gyp\pylib\gyp\__init__.py", line 98, in Load
    generator.CalculateVariables(default_variables, params)
  File "C:\Users\Brandon\AppData\Roaming\npm\node_modules\nw-gyp\gyp\pylib\gyp\generator\msvs.py", line 1798, in CalculateVariables
    generator_flags.get('msvs_version', 'auto'))
  File "C:\Users\Brandon\AppData\Roaming\npm\node_modules\nw-gyp\gyp\pylib\gyp\MSVSVersion.py", line 400, in SelectVisualStudioVersion
    versions = _DetectVisualStudioVersions(version_map[version], 'e' in version)
KeyError: '2015'
gyp ERR! configure error
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (C:\Users\Brandon\AppData\Roaming\npm\node_modules\nw-gyp\lib\configure.js:361:16)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:204:12)
gyp ERR! System Windows_NT 10.0.10586
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\Brandon\\AppData\\Roaming\\npm\\node_modules\\nw-gyp\\bin\\nw-gyp.js" "configure" "--runtime=node-webkit" "--target=0.15.0-rc2" "--module=C:\\Users\\Brandon\\Downloads\\nwjs-sdk-v0.15.0-rc2-win-x64\\nwjs-sdk-v0.15.0-rc2-win-x64\\node_modules\\node-hid\\build\\HID\\v0.5.1\\Release\\node-webkit-v0.15.0-rc2-win32-x64\\HID.node" "--module_name=HID" "--module_path=C:\\Users\\Brandon\\Downloads\\nwjs-sdk-v0.15.0-rc2-win-x64\\nwjs-sdk-v0.15.0-rc2-win-x64\\node_modules\\node-hid\\build\\HID\\v0.5.1\\Release\\node-webkit-v0.15.0-rc2-win32-x64"
gyp ERR! cwd C:\Users\Brandon\Downloads\nwjs-sdk-v0.15.0-rc2-win-x64\nwjs-sdk-v0.15.0-rc2-win-x64\node_modules\node-hid
gyp ERR! node -v v6.1.0
gyp ERR! nw-gyp -v v0.13.0
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute 'nw-gyp.cmd configure --runtime=node-webkit --target=0.15.0-rc2 --module=C:\Users\Brandon\Downloads\nwjs-sdk-v0.15.0-rc2-win-x64\nwjs-sdk-v0.15.0-rc2-win-x64\node_modules\node-hid\build\HID\v0.5.1\Release\node-webkit-v0.15.0-rc2-win32-x64\HID.node --module_name=HID --module_path=C:\Users\Brandon\Downloads\nwjs-sdk-v0.15.0-rc2-win-x64\nwjs-sdk-v0.15.0-rc2-win-x64\node_modules\node-hid\build\HID\v0.5.1\Release\node-webkit-v0.15.0-rc2-win32-x64' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (C:\Users\Brandon\AppData\Roaming\npm\node_modules\node-pre-gyp\lib\util\compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:850:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:5)
node-pre-gyp ERR! System Windows_NT 10.0.10586
node-pre-gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\Brandon\\AppData\\Roaming\\npm\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "rebuild" "--runtime=node-webkit" "--target=0.15.0-rc2"
node-pre-gyp ERR! cwd C:\Users\Brandon\Downloads\nwjs-sdk-v0.15.0-rc2-win-x64\nwjs-sdk-v0.15.0-rc2-win-x64\node_modules\node-hid
node-pre-gyp ERR! node -v v6.1.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.28
node-pre-gyp ERR! not ok
Failed to execute 'nw-gyp.cmd configure --runtime=node-webkit --target=0.15.0-rc2 --module=C:\Users\Brandon\Downloads\nwjs-sdk-v0.15.0-rc2-win-x64\nwjs-sdk-v0.15.0-rc2-win-x64\node_modules\node-hid\build\HID\v0.5.1\Release\node-webkit-v0.15.0-rc2-win32-x64\HID.node --module_name=HID --module_path=C:\Users\Brandon\Downloads\nwjs-sdk-v0.15.0-rc2-win-x64\nwjs-sdk-v0.15.0-rc2-win-x64\node_modules\node-hid\build\HID\v0.5.1\Release\node-webkit-v0.15.0-rc2-win32-x64' (1)
```

brandonros, 2016-05-15:
I tried 0.14.5 as well.

brandonros, 2016-05-16:
I ran npm install -g node-gyp, copied the newer gyp directory from that module into the outdated as shit nw-gyp directory, and it worked...

8 hours into this...

closed 2016-05-16 by brandonros

sant123, 2016-06-10:
I was trying to compile Edge.js with nw.js but I got the following error:

```
gyp info it worked if it ends with ok
gyp info using nw-gyp@0.13.0
gyp info using node@6.2.1 | win32 | x64
(node:9024) fs: re-evaluating native module sources is not supported. If you are using the graceful-fs module, please update it to a more recent version.
gyp info spawn py
gyp info spawn args [ 'C:\\Users\\santiagoah\\AppData\\Roaming\\npm\\node_modules\\nw-gyp\\gyp\\gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'msvs',
gyp info spawn args   '-G',
gyp info spawn args   'msvs_version=2013',
gyp info spawn args   '-I',
gyp info spawn args   'D:\\santiagoah\\informacion\\Desktop\\Testing Edge\\node_modules\\edge\\build\\config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\santiagoah\\AppData\\Roaming\\npm\\node_modules\\nw-gyp\\addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\santiagoah\\.nw-gyp\\0.15.1\\common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=C:\\Users\\santiagoah\\.nw-gyp\\0.15.1',
gyp info spawn args   '-Dmodule_root_dir=D:\\santiagoah\\informacion\\Desktop\\Testing Edge\\node_modules\\edge',
gyp info spawn args   '--depth=.',
gyp info spawn args   '-Dv13=1',
gyp info spawn args   '--generator-output',
gyp info spawn args   'D:\\santiagoah\\informacion\\Desktop\\Testing Edge\\node_modules\\edge\\build',
gyp info spawn args   '-Goutput_dir=.' ]
(node:9024) DeprecationWarning: child_process: options.customFds option is deprecated. Use options.stdio instead.
  File "C:\Users\santiagoah\AppData\Roaming\npm\node_modules\nw-gyp\gyp\gyp_main.py", line 12
    except ImportError, e:
                      ^
SyntaxError: invalid syntax
gyp ERR! configure error
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (C:\Users\santiagoah\AppData\Roaming\npm\node_modules\nw-gyp\lib\configure.js:361:16)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:204:12)
gyp ERR! System Windows_NT 6.3.9600
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\santiagoah\\AppData\\Roaming\\npm\\node_modules\\nw-gyp\\bin\\nw-gyp.js" "configure" "--target=v0.15.1" "--msvs_version=2013" "--arch=x64" "--python=py"
gyp ERR! cwd D:\santiagoah\informacion\Desktop\Testing Edge\node_modules\edge
gyp ERR! node -v v6.2.1
gyp ERR! nw-gyp -v v0.13.0
gyp ERR! not ok
```

I'm using Node.js v6.2.1 and nw v0.15.1 in Windows 8.1 x64.

todbot, 2016-06-10:
Hi,
There isn't a a pre-built native library for that combination of Node, NW.js, & Windows.  You will have to install a compiler and build it yourself.  See the [Compiling from source](https://github.com/node-hid/node-hid#compiling-from-source) section of the README and then the [Using node-hid in NW.js projects](https://github.com/node-hid/node-hid/blob/master/README.md#using-node-hid-in-nwjs-projects) below that.

sant123, 2016-06-10:
@todbot thanks for the answer, I've followed the steps above and I found this error: 

```
D:\santiagoah\informacion\Desktop\Testing Edge>node_modules\.bin\node-pre-gyp rebuild --runtime=node-webkit --target=0.15.1 --python=py
node-pre-gyp info it worked if it ends with ok
node-pre-gyp info using node-pre-gyp@0.6.28
node-pre-gyp info using node@6.2.1 | win32 | x64
node-pre-gyp ERR! UNCAUGHT EXCEPTION
node-pre-gyp ERR! stack Error: testing_edge package.json is not node-pre-gyp ready:
node-pre-gyp ERR! stack package.json must declare these properties:
node-pre-gyp ERR! stack binary.module_name
node-pre-gyp ERR! stack binary.module_path
node-pre-gyp ERR! stack binary.host
node-pre-gyp ERR! stack     at validate_config (D:\santiagoah\informacion\Desktop\Testing Edge\node_modules\node-pre-gyp\lib\util\versioning.js:203:15)
node-pre-gyp ERR! stack     at Object.module.exports.evaluate (D:\santiagoah\informacion\Desktop\Testing Edge\node_modules\node-pre-gyp\lib\util\versioning.js:261:5)
node-pre-gyp ERR! stack     at clean (D:\santiagoah\informacion\Desktop\Testing Edge\node_modules\node-pre-gyp\lib\clean.js:14:27)
node-pre-gyp ERR! stack     at Object.self.commands.(anonymous function) [as clean] (D:\santiagoah\informacion\Desktop\Testing Edge\node_modules\node-pre-gyp\lib\node-pre-gyp.js:48:37)
node-pre-gyp ERR! stack     at run (D:\santiagoah\informacion\Desktop\Testing Edge\node_modules\node-pre-gyp\bin\node-pre-gyp:79:30)
node-pre-gyp ERR! stack     at _combinedTickCallback (internal/process/next_tick.js:67:7)
node-pre-gyp ERR! stack     at process._tickCallback (internal/process/next_tick.js:98:9)
node-pre-gyp ERR! stack     at Function.Module.runMain (module.js:577:11)
node-pre-gyp ERR! stack     at startup (node.js:160:18)
node-pre-gyp ERR! stack     at node.js:456:3
node-pre-gyp ERR! System Windows_NT 6.3.9600
node-pre-gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "D:\\santiagoah\\informacion\\Desktop\\Testing Edge\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "rebuild" "--runtime=node-webkit" "--target=0.15.1" "--python=py"
node-pre-gyp ERR! cwd D:\santiagoah\informacion\Desktop\Testing Edge
node-pre-gyp ERR! node -v v6.2.1
node-pre-gyp ERR! node-pre-gyp -v v0.6.28
node-pre-gyp ERR! This is a bug in `node-pre-gyp`.
node-pre-gyp ERR! Try to update node-pre-gyp and file an issue if it does not help:
node-pre-gyp ERR!     <https://github.com/mapbox/node-pre-gyp/issues>
``` 

I do not know what did I do wrong :disappointed:

todbot, 2016-06-10:
Hmm. I'm not too familiar with NW.js builds. I'll see if I can re-create your setup and try building.

sant123, 2016-06-11:
Thanks @todbot :smile: