node-hid/node-hid #144: Failed at the node-hid@0.5.1 install script 'node-pre-gyp install --fallback-to-build'.

yas-nyan, 2016-06-24:
npm ERR! Linux 4.4.0-24-generic
npm ERR! node v6.2.2
npm ERR! npm  v3.9.5
npm ERR! code ELIFECYCLE
npm ERR! node-hid@0.5.1 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-hid@0.5.1 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the node-hid package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs node-hid
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls node-hid
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:

todbot, 2016-06-25:
Hi, 
1. Which Linux distro are you on?  
2. Do you have libusb installed?  On Linux, you will always be compiling from source, so make sure you have the dependencies mentioned in https://github.com/node-hid/node-hid#compiling-from-source
3. What did you type to attempt to install node-hid?

yas-nyan, 2016-06-27:
@todbot 
Thank you for the quick response.

1. Ubuntu 16.04
2. Maybe already installed. 
3."npm install node-hid" 

 When use "node v 5.6.0", this err does not occur.

todbot, 2016-07-17:
Hi, 
I just spun up Ubuntu 16.04 and was able to get node-hid to install with no problem.
I suspect the problem you experienced was that Ubuntu doesn't install `libusb-1.0-0-dev` by default.  So do this:
```
sudo apt install libusb-1.0-0-dev 
```
and then try installing node-hid again.

closed 2016-07-18 by todbot

sergy8612, 2017-01-01:
Hi,

this is what I actually get after npm install node-hid: 

root@sergio-GE72-6QD:/home/sergio/msi-klm# npm install node-hid

> node-hid@0.5.2 install /home/sergio/msi-klm/node_modules/node-hid
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download(404): https://github.com/node-hid/node-hid/releases/download/0.5.2/HID-v0.5.2-node-v51-linux-x64.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for node-hid@0.5.2 and node@7.3.0 (node-v51 ABI) (falling back to source compile with node-gyp) 
gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/7.3.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/sergio/msi-klm/node_modules/node-hid/.node-gyp"
make: se entra en el directorio '/home/sergio/msi-klm/node_modules/node-hid/build'
make: *** No hay ninguna regla para construir el objetivo '../.node-gyp/7.3.0/include/node/common.gypi', necesario para 'Makefile'.  Alto.
make: se sale del directorio '/home/sergio/msi-klm/node_modules/node-hid/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Linux 4.4.0-57-generic
gyp ERR! command "/usr/bin/nodejs" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/home/sergio/msi-klm/node_modules/node-hid/build/HID/Release/HID.node" "--module_name=HID" "--module_path=/home/sergio/msi-klm/node_modules/node-hid/build/HID/Release"
gyp ERR! cwd /home/sergio/msi-klm/node_modules/node-hid
gyp ERR! node -v v7.3.0
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/sergio/msi-klm/node_modules/node-hid/build/HID/Release/HID.node --module_name=HID --module_path=/home/sergio/msi-klm/node_modules/node-hid/build/HID/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/sergio/msi-klm/node_modules/node-hid/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:885:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
node-pre-gyp ERR! System Linux 4.4.0-57-generic
node-pre-gyp ERR! command "/usr/bin/nodejs" "/home/sergio/msi-klm/node_modules/node-hid/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/sergio/msi-klm/node_modules/node-hid
node-pre-gyp ERR! node -v v7.3.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.31
node-pre-gyp ERR! not ok 
Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/sergio/msi-klm/node_modules/node-hid/build/HID/Release/HID.node --module_name=HID --module_path=/home/sergio/msi-klm/node_modules/node-hid/build/HID/Release' (1)
npm WARN enoent ENOENT: no such file or directory, open '/home/sergio/msi-klm/package.json'
npm WARN msi-klm No description
npm WARN msi-klm No repository field.
npm WARN msi-klm No README data
npm WARN msi-klm No license field.
npm ERR! Linux 4.4.0-57-generic
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install" "node-hid"
npm ERR! node v7.3.0
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE

npm ERR! node-hid@0.5.2 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-hid@0.5.2 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the node-hid package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs node-hid
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls node-hid
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/sergio/msi-klm/npm-debug.log

this is the ubuntu distro I'm using:

root@sergio-GE72-6QD:/home/sergio/msi-klm# lsb_release -a
LSB Version:	core-9.20160110ubuntu0.2-amd64:core-9.20160110ubuntu0.2-noarch:security-9.20160110ubuntu0.2-amd64:security-9.20160110ubuntu0.2-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.1 LTS
Release:	16.04
Codename:	xenial

and of course I have installed libusb-1.0-0-dev 

It seems as if all this is due to https://github.com/node-hid/node-hid/releases/download/0.5.2/HID-v0.5.2-node-v51-linux-x64.tar.gz is no longer available, right? I don't quite understand any of this. 

Can anyone help me?

Thank you!

todbot, 2017-01-01:
Hi Sergio, 
What is the English translation of:
```
make: se entra en el directorio '/home/sergio/msi-klm/node_modules/node-hid/build'
make: *** No hay ninguna regla para construir el objetivo '../.node-gyp/7.3.0/include/node/common.gypi', necesario para 'Makefile'. Alto.
make: se sale del directorio '/home/sergio/msi-klm/node_modules/node-hid/build'
```
as I'm suspecting that's where your error is starting.

The warning about not finding "HID-v0.5.2-node-v51-linux-x64.tar.gz" is node-pre-gyp trying to find a pre-built binary for your OS. Since it cannot find it, node-pre-gyp hands off to node-gyp to compile.  (As a rule there won't be pre-built binaries for Linuxes because they vary so much)

Also, did you install the other prerequisites mentioned on:  https://github.com/node-hid/node-hid#compiling-from-source
Most specifically:
```
apt-get install build-essential git 
apt-get install gcc-4.8 g++-4.8 && export CXX=g++-4.8
```

Finally, one thing to try if you're still having issues to is `npm install --verbose node-hid`, posting the output to gist.github.com and referencing it here.  (verbose output is way too big for issues conversations)