node-hid/node-hid #145: Support for bluetooth HID?

rodzzlessa24, 2016-07-01:
I have a griffin powermate bluetooth controller and it doesn't get recognized when listing all devices. Is this package for usb-hid devices only?

todbot, 2016-07-01:
node-hid should be able to see Bluetooth HID devices (at least on some OSes in some circumstances), but I believe the Griffen PowerMate Bluetooth is not Bluetooth HID but rather a Bluetooth Low Energy (BLE) device. 

Instead, you should check out the `noble` package: https://github.com/sandeepmistry/noble
And I couldn't find a nice stand-alone package showing how to use noble with the Bluetooth PowerMate, but here's a plugin for Node-RED that should give you some tips:
  https://github.com/circuitbeard/node-red-contrib-powermateble

closed 2016-07-01 by todbot

rodzzlessa24, 2016-07-01:
@todbot Thank you!