node-hid/node-hid #146: Can't build on windows 10 following the instructions in the readme file

cwssoftwarejoshwright, 2016-07-02:
I am on windows 10 and have the following installed:

Python 2.7
node-gyp installed globally
Visual Studio 2015

I use the following commands in the node-hid directory:
git submodule update --init   # done on publish automatically
npm install       # rebuilds the module

So far so good.  Then I navigate to "node-hid/node_modules/.bin/" and execute the following command:
node-pre-gyp rebuild

I get the following error:

C:\Users\Josh\Desktop\TEst\node-hid\node_modules\.bin>node-pre-gyp rebuild
node-pre-gyp info it worked if it ends with ok
node-pre-gyp info using node-pre-gyp@0.6.29
node-pre-gyp info using node@4.4.5 | win32 | x64
node-pre-gyp ERR! UNCAUGHT EXCEPTION
node-pre-gyp ERR! stack Error: ENOENT: no such file or directory, open 'C:\Users\Josh\Desktop\TEst\node-hid\node_modules\.bin\package.json'
node-pre-gyp ERR! stack     at Error (native)
node-pre-gyp ERR! stack     at Object.fs.openSync (fs.js:549:18)
node-pre-gyp ERR! stack     at Object.fs.readFileSync (fs.js:397:15)
node-pre-gyp ERR! stack     at clean (C:\Users\Josh\Desktop\TEst\node-hid\node_modules\node-pre-gyp\lib\clean.js:13:38)
node-pre-gyp ERR! stack     at Object.self.commands.(anonymous function) [as clean] (C:\Users\Josh\Desktop\TEst\node-hid\node_modules\node-pre-gyp\lib\node-pre-gyp.js:48:37)
node-pre-gyp ERR! stack     at run (C:\Users\Josh\Desktop\TEst\node-hid\node_modules\node-pre-gyp\bin\node-pre-gyp:79:30)
node-pre-gyp ERR! stack     at nextTickCallbackWith0Args (node.js:420:9)
node-pre-gyp ERR! stack     at process._tickCallback (node.js:349:13)
node-pre-gyp ERR! stack     at Function.Module.runMain (module.js:443:11)
node-pre-gyp ERR! stack     at startup (node.js:139:18)
node-pre-gyp ERR! System Windows_NT 10.0.10586
node-pre-gyp ERR! command "E:\\Program Files\\nodejs\\node.exe" "C:\\Users\\Josh\\Desktop\\TEst\\node-hid\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "rebuild"
node-pre-gyp ERR! cwd C:\Users\Josh\Desktop\TEst\node-hid\node_modules\.bin
node-pre-gyp ERR! node -v v4.4.5
node-pre-gyp ERR! node-pre-gyp -v v0.6.29
node-pre-gyp ERR! This is a bug in `node-pre-gyp`.
node-pre-gyp ERR! Try to update node-pre-gyp and file an issue if it does not help:
node-pre-gyp ERR!     <https://github.com/mapbox/node-pre-gyp/issues>

pfirpfel, 2016-09-02:
I'm pretty sure that you launched the build the wrong way. You used the following:

`C:\Users\Josh\Desktop\TEst\node-hid\node_modules.bin>node-pre-gyp rebuild`

The final directory "node_modules.bin" looks like a product of a typo and should not exist. Anyway, your base path should be the node-hid directory. In this case "C:\Users\Josh\Desktop\TEst\node-hid\". From there you launch node-pre-gyp.

Try it this way:

```
cd C:\Users\Josh\Desktop\TEst\node-hid\
node_modules\.bin\node-pre-gyp rebuild
```

This _should_ work as long as the rest is setup correctly. I managed to build it on Windows 10 as well, but I used Visual Studio 2013.

McClellandLaboratories, 2016-09-13:
I tried it the way pfirpfel specified:

cd C:\Users\Josh\Desktop\TEst\node-hid\
node_modules\.bin\node-pre-gyp rebuild

and it worked :+1: 

Thanks for the clarification!

GeoMetJosh, 2016-09-13:
This worked for me (Thanks pfirpfel):

cd C:\Users\Josh\Desktop\TEst\node-hid\
node_modules\.bin\node-pre-gyp rebuild

note that the readme file indicates:  ./node_modules/.bin/node-pre-gyp rebuild

Should the readMe file read "node_modules\.bin\node-pre-gyp rebuild" instead?

todbot, 2016-09-13:
On most dev environments (including Windows using MinGW-based ones like the git bash shell and even Windows PowerShell), doing: ` ./node_modules/.bin/node-pre-gyp rebuild` will work.

I forgot that backslash paths are only necessary when using Windows CMD shell.  I've updated the README to reflect this.

closed 2019-11-23 by todbot