node-hid/node-hid #147: Issues with electron

rodzzlessa24, 2016-07-06:
I'm running node 6.2.2 and electron 1.2. I have a griffin powermate connected to my machine. I'm running trying to just run a simple "data" event function

NOTE: I followed the instructions in the README to use this package with electron

```
var HID = require('node-hid');
var device = new HID.HID("USB_077d_0410_14200000");

device.on("data", function(data) {
  console.log(data);
});
```

And I get this error with any "gesture" I do with the Powermate 

```
#
# Fatal error in , line 0
# unreachable code
#

==== C stack trace ===============================

 1: V8_Fatal
 2: v8::internal::ToBooleanICStub::Types::UpdateStatus(v8::internal::Handle<v8::internal::Object>)
 3: v8::internal::ToBooleanICStub::UpdateStatus(v8::internal::Handle<v8::internal::Object>)
 4: v8::internal::Runtime_ToBooleanIC_Miss(int, v8::internal::Object**, v8::internal::Isolate*)
 5: 0x1f12626079a7
 6: 0x1f126262ea65
```

todbot, 2016-07-06:
Hi,
Can you try either Node 6 in a non-Electron environment or Node 5 in Electron?

rodzzlessa24, 2016-07-06:
@todbot I tried node 6 in a none electron env and it worked and Node5 with electron same error

todbot, 2016-07-18:
Hi, another question: What OS are you on?  I've created a demo showing node-hid working inside of Electron 1.2.6 here: https://www.npmjs.com/package/electron-hid-toy

closed 2016-07-18 by todbot