node-hid/node-hid #149: self.on is not a function

runn1ng, 2016-07-15:
Code I try to run:
    
    var HID = require('node-hid');
    var device = HID.HID(21324, 1);

The resulting error:

```
        self.on("newListener", function(eventName, listener) {
             ^

TypeError: self.on is not a function
    at Object.HID (/home/peepee/dev/trezor-connection/node_modules/node-hid/nodehid.js:49:7)
    at Object.<anonymous> (/home/peepee/dev/trezor-connection/lib/index.js:43:18)
    at Module._compile (module.js:399:26)
    at Object.Module._extensions..js (module.js:406:10)
    at Module.load (module.js:345:32)
    at Function.Module._load (module.js:302:12)
    at Function.Module.runMain (module.js:431:10)
    at startup (node.js:141:18)
    at node.js:977:3

```

When I digged deeper, it seems the HID never inherits eventemitter correctly. I am not sure why.

runn1ng, 2016-07-15:
I have node version 5.2.0, node-hid from npm (0.5.1)

todbot, 2016-07-15:
node-hid does inherit from EventListeneer (line 53 in nodehis.js)  I'm not sure how you're getting that error.  What OS are you on?  Do you have the 'util' or an 'EventEmitter' package installed globally?  How did you install node-hid?  Thanks!

runn1ng, 2016-07-15:
Oh. That's because I call

`var foo = HID.HID(21324, 1)`

the correct call is

`var foo =new HID.HID(21324, 1)`

That's about it.

runn1ng, 2016-07-15:
Not sure if I should call this. It's possible to check if the function is called correctly, right?

todbot, 2016-07-17:
I don't think so.  I'll close this issue anyway, if you find out anything, please update enhancement issue #150.  Thanks!

closed 2016-07-17 by todbot