node-hid/node-hid #152: Get device path inside data event?

gordysc, 2016-07-30:
Sorry I'm a bit new to this event-driven world of javascript... If I have a device and want to log the device's path inside the device.on('data', function(data){}) method, how would I go about this? I have multiple devices I want to read data from simultaneously, but can't figure out how to tell which device the data being read came from.

todbot, 2016-07-30:
Hi,

If you are creating a new inner function for each `on()`, then each handler function can be customized for each device, like:
```
device1.on('data', function(data) { 
  //do something with device 1
  console.log("this is device 1");
}));
device2.on('data', function(data) { 
  //do something with device 2
  console.log("this is device 2");
}));

```
If you are iterating over a list of devices or have created a single separate function, you can pass in extra arguments to that function to be whatever you like, e.g.:
```
var handleData = function(data, mydeviceid) {
  console.log("here is data from "+mydeviceid+": "+data);
  if( mydeviceid === 'myfirstdevice' ) { 
    console.log("I like this device the best");
  } 
};

device1.on('data', handleData(data,'myfirstdevice'));
device2.on('data', handleData(data,'myseconddevice'));

gordysc, 2016-07-30:
Ah, beautiful, thank you very much for the super fast response!!

closed 2016-11-16 by todbot