node-hid/node-hid #153: Error in electron,when i pull out device

qqnp1100, 2016-08-16:
use node-hid in electron app,when i pull out my device,my electron app interface into blank,and DevTools log :
[image]
but use device.close() before pull out device,it's ok.
I think code                     `
device.on("data", (data) => { console.log(data); });`
is the problem,how to solve it ,thanks

todbot, 2016-08-16:
Hi,
Without seeing your code, I can't offer useful debugging advice. It sounds like a problem with DevTools though, not node-hid.

qqnp1100, 2016-08-16:
@todbot 
my code is very easy. this is all my code:
```
import React, {Component} from 'react';
import RaisedButton from 'material-ui/RaisedButton';

var myDevice;
var HID = require('node-hid');

export default class Pager1 extends Component {
    constructor(props) {
        super(props)
        usbDetect.on('add', (device) => {
            console.log('add', device);
            if (device.vendorId == 1155 && device.productId == 22352) {
                myDevice = new HID.HID(1155, 22352);
                myDevice.on("data", (data) => {
                    console.log(data);
                });
            }
        });
    }

    _hehe() {
        myDevice.write([0xB0, 0x03, 0xAF, 0x01, 0x00, 0xAF]);
    }
    render() {
        return (
            <div>
                <RaisedButton
                    onTouchTap={() => this._hehe() }
                    label="Primary"
                    primary={true} />
            </div>
        )
    }
}
```
it's run ok.but pull out device make app interface into blank.

if i run twice `myDevice.close();` is happened same problem.
 it is not judge device connect in nodehid c++ code?

todbot, 2016-08-16:
It sounds like you're not checking if `myDevice` is not null before calling methods on it.

qqnp1100, 2016-08-17:
can you add checking device is connected in c/c++ method?
in `device.on("data", function(data) {});`
are you create a new thread continues  run hid_read() method in c?
In that case,i think when device pull out,the read thread still run,so make error in c code.

in my js code,i don.t konw how to slove this problem.
not run  `
myDevice.on("data", (data) => {
                    console.log(data);
                });
`
There is no problem

todbot, 2016-08-17:
You can either capture USB disconnect events using the `usb-detection` package, as you're doing for connect.  Or you can wrap calls to the HID device object in a try/catch, like:
```
try {
  myDevice.write(...);
} catch(err) {
  console.log('device disconnected');
}
```

qqnp1100, 2016-08-17:
I use         `
device.read((e,d)=>{console.log(d); })` now, it's ok.
I give up the `device.on("data", function(data) {});` method.
I still think there are some error in hid.c when run read thread.

Thank you for your help.(ง •̀_•́)ง

closed 2016-08-17 by qqnp1100

todbot, 2016-08-17:
Sorry I don't understand what you mean by "when run read thread".

qqnp1100, 2016-08-17:
when run `device.on("data", function(data) {});`
is you create a new thread loop read device data?
such as:
while(flag){
Thread.sleep(time);
hid_read();
} 
when the device pull out,the loop have some error.
just i think.
so I use `device.read();`

todbot, 2016-08-17:
If I understand you, that's not how hidapi (the underlying C library that node-hid uses), does on reads. Instead, it uses platform-specific event methods depending on the OS. (On Mac it uses the IOHIDDeviceScheduler; on Windows it uses ReadFile() on the threaded HID device driver; on Linux, it does use pthreads and waiting)  In any case, it is not the duty of node-hid to deal with threads, but with hidapi. 

node-hid works as intended, I'm unclear what issues you're still having.

qqnp1100, 2016-08-17:
sorry,I am chinese,most of reply depends on translation。
i don't know how to express my meaning，but now the problem have sorted.
thanks~