node-hid/node-hid #154: Asynchronous HID.devices()?

runn1ng, 2016-08-23:
Hello,

HID.devices is synchronous. That can cause problems, because the whole node locks down, and it takes a while to list the devices (about 10 seconds on my PC).

Would it be possible to add some asynchronous call?

todbot, 2016-08-23:
node-hid uses the underlying hidapi C library to accomplish its tasks, including enumeration.  Unfortunately, it's single threaded.

Enumeration is normally very quick (<500msec, often <100msec).  I would look into why it's taking so long on your PC.  Do you have a mis-behaving device that the OS's USB stack is issuing retries on?  Or many devices or hubs?

runn1ng, 2016-08-23:
Interesting. Once I unplugged the device and plugged it again, it's quick. It seems like a "Heisenbug". I will keep this open in case I found what exactly triggered it. (It might be something with more programs using HID API at the same time.)

runn1ng, 2016-08-23:
Oh. It seems like it's triggered when I start to write and read into the device a lot, in one process through node-hid, and simultaneously listing repeatedly. Then suddenly it takes about 10 seconds, even after I kill the process and start it again.

runn1ng, 2016-08-23:
It seemed to be a bug in the hardware firmware that slowed down reading of the device info when being in inconsistent state.

closed 2016-08-23 by runn1ng

todbot, 2016-08-23:
Interesting. I've seen enumerations in the OS take several seconds when one of the devices I was developing had bad firmware that was making it emit wrong data in response to enumeration. The OS would retry about 10 times every 500ms or so and finally give up, marking the device as "does not respond to enumeration".

runn1ng, 2016-08-25:
If you are interested, it was caused probably by this issue in library that's used in the firmware

https://github.com/libopencm3/libopencm3/issues/668

dcerisano, 2018-04-12:
If your requirements are for writing concurrently to multiple hid devices, you can use the [threads](https://www.npmjs.com/package/threads) package to spawn a separate worker for each device.

A bit like swatting a fly with a Buic, but it does effectively give you asynchronous hid writes that do not block the parent node.

As @todbot mentioned, all writes in node-hid are synchronous. It would be nice if asynchronous writes were supported at some point.