node-hid/node-hid #157: Increase buffer size to 2048 or something that's sensible

danieltian, 2016-09-12:
The Aquacomputer Aquero 6 will send a input report of 1038 bytes:

[image]

This is larger than the 1024-byte buffer set in `HID.cc`, so only the first 1024 bytes of the report will be received. I suggest increasing this to something reasonably large to account for devices that will send large amounts of data in one report.

todbot, 2016-09-12:
Thanks Daniel, this looks like a good change. I've never seen a HID report so large! I looked a bit into the underlying hidapi C library and found no glaring issues with upping it beyond 1024.

I created a local define READ_BUFF_MAXSIZE in HID.cc and upped it to 2048.  I did some preliminary tests and it doesn't seem to have any adverse effects.  If you'd like to try it out now (since I won't be able to do a proper release for several weeks), you can reference it in your package.json as 
```
"dependencies": {
    "node-hid": "node-hid/node-hid"
}
```

danieltian, 2016-09-14:
Yeah, the Aquaero has a lot of sensors on it and reports the full state of everything on every update. I found out recently that only the first 868 bytes are used, but it reserves 1038 bytes. Better safe than sorry I say. Thanks for the fix!

closed 2016-09-14 by danieltian