node-hid/node-hid #158: Trackpad Force touch problem

bushev, 2016-09-25:
Hello!

Code above disables my Trackpad clicks (but I still can move my mouse pointer). I use a Macbooc 2015 with force touch.

```javascript
var HID = require('node-hid');

var device = new HID.HID('USB_05ac_0274_14400000');

device.on("data", function(data) {
     console.log(data);
});
```

todbot, 2016-09-25:
What device is "USB_05ac_0274_14400000"?

bushev, 2016-09-25:
@todbot Hi! It's my HID device:

```javascript
HID.devices(): 

{ vendorId: 1452,
    productId: 628,
    path: 'USB_05ac_0274_14400000',
    serialNumber: 'D3H5175F1W1FTV4AX3JS',
    manufacturer: 'Apple Inc.',
    product: 'Apple Internal Keyboard / Trackpad',
    release: 1572,
    interface: -1 }
```

todbot, 2016-11-16:
Hi,
The problem you're running into I think is that you can't just open a composite USB device like the trackpad/keyboard on Macs, since it shows up as multiple devices to node-hid.   So instead you need to get the path of the current sub-device and do `var device = new HID.HID( path )`. 

One way to get the correct path is to dump all the devices with "HID.devices()" and keep trying one until you get the trackpad.  (This is kinda dangerous as you both need to be the "root" user with `sudo` and you could grab the keyboard instead, meaning all key input goes to the program, not to the OS.  To get control back, you'll have to restart your box or ssh into it and kill the node process)

Another way is to use the `usage` and `usagePage properties when searching devices. In the github master version of node-hid there is now `usagePage` & `usage` field support for Mac & Windows.  (use `npm install github:node-hid/node-hid` to get it)

I've checked in an example of how to use this that works on my Macbook Pro here:
https://github.com/node-hid/node-hid/blob/master/src/test-macbookprotrackpad.js
My `productId` is a bit different from yours, so you'll have to adjust that.

closed 2019-07-25 by todbot