node-hid/node-hid #159: Segmentation fault (core dumped)

agirorn, 2016-10-13:
When using the read callback I can't close the device without getting a Segmentation fault.
This only happens when the reader function reassigns its self to the read.

```javascript
var HID=require('node-hid');
var device = new HID.HID(3118, 3233);
var reader = function(err, data) { 
  console.log('error: ', err);
  console.log('data:', data);
  device.read(reader); // Only if this is done.
};
device.read(reader); 
device.close();
```

```shell
>error:  [Error: could not read from HID device]
data: undefined
Segmentation fault (core dumped)
```

todbot, 2016-10-13:
You are calling the reader function from inside the reader function.  This leads to undefined behavior.

agirorn, 2016-10-15:
My guess is that when you call close on the device it calls the callback to read with the error "Error: could not read from HID device" then when I try to call device.read after closing then I get the Segmentation fault. I think I would get the same error with this code.

```javascript
var HID=require('node-hid');
var device = new HID.HID(3118, 3233);
var reader = function(err, data) { 
  console.log('error: ', err);
  console.log('data:', data);
};
device.read(reader); 
device.close();
device.read(reader); // I thnk this will cause a Segmentation fault.
```

If that is the case then we might need to add a check in the read function to check if the device is closed and throw and exception, not Segmentation.

````javascript
throw new Error('Unable to read from a closed HID device');
```

I'm going to test that on Monday when I have the USB device.

agirorn, 2016-10-24:
I have tried this now and it is the read after `close()` that does cause the segfault.

```shell
$> node
> var HID=require('node-hid');
undefined
> var device = new HID.HID(3118, 3233);
undefined
> device.close()
undefined
> device.read(function(d) { console.log(d); });
[1]    11019 segmentation fault (core dumped)  node
```

todbot, 2016-11-08:
Thanks!  This will be in v0.52.

closed 2016-11-08 by todbot