node-hid/node-hid #161: Unable to require node-hid in Travis CI

agirorn, 2016-10-22:
When requiring node-hid in Travis-CI I get this error "cannot initialize hidapi (hid_init failed)".

I am unable to run unit tests on modules that use node-hid on Travis-CI because this error is thrown when node-hid is required. I have tried to stub out the module in my tests without any luck. I have traced this to https://github.com/node-hid/node-hid/blob/master/src/HID.cc#L583. This code is being run when the module is required and I think this should be done when the HID instance is created. 

```javascript
const HID = require('node-hid'); // Is thrown her.

const device = new HID.HID(path); // Should be thrown her.
```

todbot, 2016-10-22:
Hi,
I'm not sure hidapi's `hid_init()` function could be moved into an object instance's data. I think there are hidapi-global data that needs to be available to all node-hid instances.  I'll look a bit more, but I think `hid_init()` must remain on class load.

todbot, 2016-10-22:
Why is `hid_init()` failing on your system?

agirorn, 2016-10-22:
I have no idea why hid_init is failing on [https://travis-ci.org/](https://travis-ci.org/agirorn/node-hid-stream/builds/169731455). I am trying to get the CI server to run the unit tests for [node-hid-stream](https://github.com/agirorn/node-hid-stream) and I can't get it to run them because of this error.

I have installed build-essential git libudev-dev libusb-1.0-0-dev gcc-4.8 g++-4.8 and that is all needed to install node-hid on travis-co I got some other errors before I did that.

As I see it I should be able to start node.js and require the module before I start to get runtime errors from the module. I guess this could be moved into an init call that would be run on first use.

According to this http://www.signal11.us/oss/hidapi/hidapi/doxygen/html/group__API.html#ga142ffc1b0b7a7fa412d3862b2a17164b hid_init could be either skipped or called for before any usage.

agirorn, 2016-10-22:
I guess hid_init is failing because there is not USB controller on the CI server.

rdepena, 2016-10-22:
Hey @agirorn, you could Mock node-hid altogether, this is my travis-cli setup /mocking for example:

Travis setup:
https://github.com/rdepena/node-dualshock-controller/blob/master/.travis.yml#L4

Tests that mock node-hid:
https://github.com/rdepena/node-dualshock-controller/blob/master/test/dualshock.tests.js#L57

This is old code so I assume there are now better ways to mock but that might be a good starting point.

agirorn, 2016-10-22:
Thanks, that does work.

Only because mockery is so overly aggressive when it does it's mocking.

You can get node-hid to install on travice by adding this to travis.yml.
```yaml
before_install:
  - sudo apt-get update
  - sudo apt-get install build-essential git -y
  - sudo apt-get install libudev-dev -y
  - sudo apt-get install libusb-1.0-0-dev -y
  - sudo apt-get install gcc-4.8 g++-4.8 -y
  - if [[ `npm -v` != 3* ]]; then npm i -g npm@3; fi
```

I still think that node-hid should not start executing this code when the module is required.

todbot, 2017-01-29:
I believe this has been fixed in the latest release.  We're currently doing builds in Travis too. 
Thanks for your patience.

closed 2017-01-29 by todbot