node-hid/node-hid #164: install error in windows

qqnp1100, 2016-11-09:
I use electron. the project in mac is build success,but int window is error.

I am already add  `"postinstall": "electron-rebuild --pre-gyp-fix --force"`

when i run `npm install` ,in windows is log:
[image]

I find  it's try to get [https://github.com/node-hid/node-hid/releases/download/0.5.2/HID-v0.5.2-electron-v1.4-win32-x64.tar.gz](https://github.com/node-hid/node-hid/releases/download/0.5.2/HID-v0.5.2-electron-v1.4-win32-x64.tar.gz)
but in mac it's get [https://github.com/node-hid/node-hid/releases/download/0.5.2/HID-v0.5.2-node-v48-darwin-x64.tar.gz](https://github.com/node-hid/node-hid/releases/download/0.5.2/HID-v0.5.2-node-v48-darwin-x64.tar.gz)

It may be causing the problem。

Does anyone here know how to solve it？thanks

todbot, 2016-11-09:
You need a compiler to rebuild node-hid for Electron.  An abbreviated list of requirements is mentioned here: https://github.com/node-hid/node-hid#compiling-from-source
and a full description is mentioned here: https://github.com/nodejs/node-gyp#installation

qqnp1100, 2016-11-11:
Thank you very much . it's build success.
but when i new HID,it's log `Uncaught Error: cannot open device with vendor id 0x1c4f and product id `

my code :
```
    var devices = HID.devices()
    var device = new HID.HID(devices[0].vendorId,devices[0].productId);
```

my system: win8 x64
Ask for your help again. thanks

qqnp1100, 2016-11-11:
when i opening my HID device(not keyboard or mouse).
```
var device = new HID.HID(devices[4].vendorId,devices[4].productId);
console.log(device);
device.write([0xB0, 0x03, 0xAF, 0x01, 0x00, 0xAF]);
```
it's log :
[image]

qqnp1100, 2016-11-11:
oh,I found a way to solve it. =.=
[Cannot write to HID device error #70](https://github.com/node-hid/node-hid/issues/70)

closed 2016-11-11 by qqnp1100