node-hid/node-hid #165: Randomly "Cannot write to HID device"

vincentdieltiens, 2016-11-14:
I'm using node-hid (master branch) on a mac 10.10 with a teensy 2++ programmed as RawHID, 16 Mhz using Teensyduino (Teensy plugin for Arduino).

Sometimes everything is working as expected, sometimes writing a message to the device hangs. On Ctrl+C,  I get the following message : 

> device.write(message);
>       ^
>
>Error: Cannot write to HID device
>    at Error (native)
>    at Object.<anonymous> (/Users/Vincent/Documents/code/test_64.js:11:8)
>    at Module._compile (module.js:570:32)
>    at Object.Module._extensions..js (module.js:579:10)
>    at Module.load (module.js:487:32)
>    at tryModuleLoad (module.js:446:12)
>    at Function.Module._load (module.js:438:3)
>    at Module.runMain (module.js:604:10)
>    at run (bootstrap_node.js:394:7)
>    at startup (bootstrap_node.js:149:9)

I think that the problem does not occur on linux (I will test again today).

NB 1 : As the following codes are very simples and shorts, I don't think the problem is on the teensy side. Also, It works very well with the following tester (in C) : http://www.pjrc.com/teensy/rawhid.html
NB 2 : Everything is ok for Receiving data from the teensy,.

When I run the test_64.js several times (I test with and without a delay), the results are different : 

- Works as expected (message is sent, "Received" is printed in Serial and the LED blinks
- test_64.js hangs on writing, "Received" is never printed in Serial and the LED doesn't blink (as if no data were received)
- test_64.js doesn't hang but "Received" is never printed in Serial and the LED doesn't blink (as if no data were received)

Here are the codes for the tester and teensy

test_64.js
```javascript
var HID = require('node-hid');
var device = new HID.HID(0x16C0, 0x0486);

var message = [];
for(var i=0; i < 64; i++) {
	message[i] = 0x00;
}
message[1] = 122;

console.log('message : ', message);
device.write(message);
```

Teensy
```c
void setup() {
  Serial.begin(9600);
  Serial.println(F("RawHID Example"));
  digitalWrite(6, HIGH);
}

// RawHID packets are always 64 bytes
byte buffer[64];

void loop() {
  int n;
  n = RawHID.recv(buffer, 0); // 0 timeout = do not wait
  if (n > 0) {
     // the computer sent a message.  Blink the led
    Serial.println(F("Receive"));
    for(int i=0; i < 5; i++) {
      digitalWrite(6, LOW);
      delay(100);
      digitalWrite(6, HIGH);
      delay(100);
    }
  }
}
```

Any Idea on what's going on ?

Thanks

vincentdieltiens, 2016-11-14:
I confirm that the node-hid works on Ubuntu 16.04 (in a VirtualBox on my Mac).

todbot, 2016-11-14:
As a general rule I've learned when doing HID writes to custom HID class devices: try inserting the report descriptor at the beginning of your message array.  This is because of historical weirdness, but try this in your JS:
```
var message = [];
for(var i=0; i < 65; i++) {
	message[i] = 0x00;
}
message[0] = 1;
message[1] = 122;
```
(From the Teensy code, it looks like the report descriptor's id is '1'.  Also, note the buffer size is increased by one.  This may or may not be the right thing to do)

To figure out what's really going on, I'll need to pull out one of my Teensy's and start poking.  Since node-hid is just a wrapper on the C hidapi library, I think we need to see what hidapi is doing. 

On Linux, hidapi's `hid_write()` is does some cleverness to determine whether or not to send via a USB control transfer or interrupt transfer, but the Mac code for `hid_write()` is always doing an output transfer. I need to look in my notes.

Apologies for the vagueness, but I wanted to give some quick pointers in case you're blocked on this.

vincentdieltiens, 2016-11-14:
I had made a lot of digging before posting this issue and already tried setting a report descriptor ID. I just retry now, and same problem.

If it is a report descriptor problem, I'm not sure the teensy will receive any data. But in my case sometimes it just works.

I tried with https://github.com/gbishop/cython-hidapi and have the same problem using the python equivalent tester (also tried with an array of size 65 + descriptor to 1 at the beginning of the array): 

So, the problem seems to be in hidapi...

```python
import hid

def tostr(num):
	return str(num)

device = hid.device(0x16C0, 0x0486)

message = []
for i in range(64):
	message.append(0x00)
message[0] = 122

print 'message : ' + ','.join(map(tostr, message))
device.write(message) 
```

todbot, 2016-11-14:
I've verified this issue on a Teensy 2 running RawHID and connected to Mac OS X 10.11.6.
As you've mentioned, it appears to not be a reportId issue or a buffer size issue.

I'm poking around the hidapi source some more to see what's wrong.  To explore hidapi, I've created a small program demonstrating the problem:
https://gist.github.com/todbot/8cd6c3ebdd7357dabd346ac16d8735c8

The above program also occasionally hangs on hidapi's `hid_write()`.   It seems to *not* hang however, if `hid_read()` is also called. (by uncommenting `#define ENABLE_READ` in the gist)

todbot, 2016-11-14:
Okay I think I figured out the problem.

Teensy RawHID advertises itself as a compound HID device.  Basically it shows up as *two* devices to node-hid (and hidapi): "Teensyduino RawHID" and "Emulated Arduino Serial".   Both have the same vendorId & productId, so we need to filter on something else.  In the example PJRC provides, he filters on usagePage & usage.

I think what is happening on the Mac is we're randomly getting the other device instead of the "Teensyduino RawHID". 

I've added a new test script in 'src' that shows how to filter on usagePage & usage and it seems to reliably send data to the Teensy.  You can view it here:
https://github.com/node-hid/node-hid/blob/master/src/test-teensyrawhid.js

vincentdieltiens, 2016-11-15:
You got the point ! It seems to work fine with usagePage and usage :)

I did not see that teensy had shows up as two devices (same under linux btw). It's weird that the order in Mac OS is not consistent.

Anyway, thanks for your help :)

I will close this issue and the one on hidapi.

closed 2016-11-15 by vincentdieltiens