node-hid/node-hid #166: Packaging for Visual Studio Code Extension

JamesCaska, 2016-11-14:
I am developing a VSCode extension that uses node-hid. In server 'debug' mode everything works very nicely but when deploying as an extension the native module doesn't get loaded.

This seems to be a known and difficult problem
https://github.com/electron/electron/issues/3243
https://github.com/Microsoft/vscode/issues/658

It seems to relate to repacking somehow the native module output with appropriate headers. It's all a bit of a muddle because it seems several versions need to be aligned.

VSCode is from what I can tell an electron project and there are some comments in the readme about how to configure node-hid to output for an electron project eg  "postinstall": "electron-rebuild --force" etc

However this is for an electron project from the electron development environment I assume not for binding to the output of an electron project ie VSCode

I am working with Windows 10 64 bit as my development platform but want cross-platform capability for the extension I am working on

It seems for my development platform the npm creates a 64 bit version of node-hid but VSCode can only load 32bit. 

I have tried several things, and I am relatively new to nodejs - this is my first project

A few observations
* node-hid seems to not find any prebuilt nodes and 'fallsback' to build which does work
* Checking out from source and building seems to fail
* Checking out electron-rebuild seems to fail so I haven't been able to figure out how to apply the repackaging
* Running my extension in Windows 7 32 bit VSCode gives the error 'LaunchingThe system cannot find message text for message number 0x%1 in the message file for %2'
* Running my extension in Windows 10 64 bit VSCode gives the error 'Launching%1 is not a valid Win32 application.'

From the research it looks like I need to somehow use the electron-rebuild to repackage the node-hid generated output to work with the specific version of node bundled with VSCode.

So what I would like is instructions on how to use the above workarounds to package node-hid as a visual studio extension on Windows 10 64bit. Particularly I would like to know how to maintain these workarounds as new versions are created. This could be tricky because I don't think VSCode extension manager runs any post installation scripts (yet). However, I don't mind creating multiple versions of my extension to account for the different configurations as long as it's easy for my end users.

Thanks for a great module.

todbot, 2016-11-15:
When you're building and you see it fail, what errors are you seeing?
Unfortunately, npm can disguise what's really going on when handling native modules so I would try `npm install --verbose` to get better visibility.  Be prepared for a wall of text though.

I think the crux of the issue is what is the version of electron-rebuild (or equivalent) and version of Electron you need to work with VSCode? 

`electron-rebuild` uses the headers and code of the specific version of Electron to recompile native modules like `node-hid`.  If you have the wrong ones, `electron-rebuild` will generate a native Node module unusable with the version of Node inside of Electron.  In previous versions of `electron-rebuild` you had to specify by hand the exact version of Electron to compile against.  Now it can figure it out usually.

From the comment https://github.com/Microsoft/vscode/issues/658#issuecomment-240616389 it sounds like VSCode ships with an older version of Electron that newer `electron-rebuild` can't handle.  In which case, perhaps you can try getting an older version of `electron-rebuild` and trying that? (I am a little suspect of that, as I think I've compiled older versions of Electron with electron-rebuild no problem, but still, it's a potential lead)

JamesCaska, 2016-11-15:
To reproduce the issue - you need Visual Studio Code
Follow the development setup here 
https://code.visualstudio.com/Docs/extensions/example-debuggers
ie
```
git clone https://github.com/Microsoft/vscode-mock-debug.git
cd vscode-mock-debug
npm install
```
Open the that vscode-mock-debug folder in VS Code and press F5 to launch the extension
Then (From above reference)
* In VS Code, create a test project with a new "program" file readme.md and enter several lines of arbitrary text.
* Switch to the Debug view and press the gear icon.
* VS Code will let you select a debug environment (select "Mock Debugger") and creates a default launch configuration for it.
* Launch debug mode and test out the mock debug stepping etc

Next install node-hid into the same project

`npm install node-hid`
And finally edit the src/MockDebug.ts source code adding adding var HID = require('node-hid'); 

```
..
import {basename} from 'path';
var HID = require('node-hid');
```

Run the extension again and this time when launching the mock debugger session the extension crashes with 'debug adapter process has terminated unexpectedly'. At least that's whats happening on my WIndows 10 x64 machine.
VSCode 1.71.
node v6.9.1
npm 3.10.9

librae8226, 2018-06-21:
Any progress?

todbot, 2018-06-21:
Maybe try a technique similar to that shown here for `node-serialport`?  Both `node-hid` and `node-serialport` have native components, so they have similar issues. https://github.com/node-serialport/node-serialport/issues/1410

JamesCaska, 2018-06-21:
As a work around I ended up not integrating HID and using a SocketServer instead and hosting the HID access in a separate application. It's not ideal and an extra setup step so I will revisit this problem eventually. Hopefully by then it will have resolved itself.

closed 2019-11-23 by todbot