node-hid/node-hid #167: Using node-hid with electron on mac

ovsokolov, 2016-11-14:
Im getting an error on Mac OS X Yosemite 10.10.5
` 
Error: Module version mismatch. Expected 50, got 46.
    at Error (native)
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:178:20)
    at Object.Module._extensions..node (module.js:583:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:178:20)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (module.js:483:17)
    at require (internal/module.js:20:19)
`

Also getting this error during npm install
`[node-hid] Success: "/Users/olegsokolov/Documents/Navtool Projects/navtool-mac/electron-quick-start/node_modules/node-hid/build/HID/Release/HID.node" is installed via remote
npm ERR! Darwin 14.5.0
npm ERR! argv "/Users/olegsokolov/.nvm/versions/node/v4.4.7/bin/node" "/Users/olegsokolov/.nvm/versions/node/v4.4.7/bin/npm" "install"
npm ERR! node v4.4.7
npm ERR! npm  v2.15.8
npm ERR! path ../electron-prebuilt/cli.js
npm ERR! code EEXIST
npm ERR! errno -17
npm ERR! syscall symlink
`

amadsen, 2016-11-14:
So far as I can tell - from similar errors I have dealt with - you are using node-hid with an electron app, but have not rebuilt it for electron as instructed in the README. There is a very good chance your node-hid has been built against the headers for node v4.4.7, rather than the version of node that is integrated with the version of electron you are using.

ovsokolov, 2016-11-14:
Not sure what im missing
i have  "postinstall": "electron-rebuild --force" in my script
and have electron-rebuild and electron-prebuilt in devDependencies
here is my package.json file
```
{
  "name": "electron-quick-start",
  "productName": "DashConnectPlus",
  "version": "1.0.0",
  "description": "A minimal Electron application",
  "main": "main.js",
  "scripts": {
    "start": "electron .",
    "postinstall": "electron-rebuild --force",
    "watch": "./node_modules/.bin/webpack-dev-server"
  },
  "repository": "https://github.com/electron/electron-quick-start",
  "keywords": [
    "Electron",
    "quick",
    "start",
    "tutorial",
    "demo"
  ],
  "author": "GitHub",
  "license": "CC0-1.0",
  "devDependencies": {
    "babel": "^5.6.10",
    "babel-core": "^5.6.11",
    "babel-loader": "^5.2.2",
    "css-loader": "^0.15.1",
    "electron-packager": "^4.1.3",
    "electron-rebuild": "^1.3.0",
    "less": "^2.5.1",
    "less-loader": "^2.2.0",
    "node-libs-browser": "^0.5.2",
    "style-loader": "^0.12.3",
    "webpack": "^1.9.12",
    "webpack-dev-server": "^1.9.0",
    "electron": "^1.4.1",
    "electron-prebuilt": "^1.3.8"
  },
  "dependencies": {
    "electron-log": "^1.2.2",
    "node-hid": "^0.5.1",
    "node-usb-detection": "^1.0.2",
    "react": "^15.3.2",
    "react-dom": "^15.3.2"
  }
}
```
Same file works on widows machine

ovsokolov, 2016-11-14:
I also tried with node 6.9.1 and its failing as well.
I removed node_module folder and cleaned npm cache as well

ovsokolov, 2016-11-15:
I have tried to compile with multiple version of node. I used 6.9.1, 7.0.0, 7.1.0
If its < 7.0.0 im getting expected module version 50 got 48. If its > 7.0 we are getting expected module version 50 got 51.

amadsen, 2016-11-15:
I don't know exactly what you are dealing with. A couple of things to try:

1. is there any difference between running `npm start` and running `electron .` directly? (I doubt it, but it occurred to me to test it.)
2. Have you looked at https://github.com/electron/electron/blob/master/docs/tutorial/using-native-node-modules.md#modules-that-rely-on-node-pre-gyp? node-hid relies on node-pre-gyp, so those instructions might apply.

todbot, 2016-11-15:
Try this project:
  https://github.com/todbot/electron-hid-toy

ovsokolov, 2016-11-15:
Thanks,

I eventually did these steps and it start working

# Electron's version.
export npm_config_target=1.2.3
# The architecture of Electron, can be ia32 or x64.
export npm_config_arch=x64
export npm_config_target_arch=x64
# Download headers for Electron.
export npm_config_disturl=https://atom.io/download/electron
# Tell node-pre-gyp that we are building for Electron.
export npm_config_runtime=electron
# Tell node-pre-gyp to build module from source code.
export npm_config_build_from_source=true
# Install all dependencies, and store cache to ~/.electron-gyp.
HOME=~/.electron-gyp npm install

ovsokolov, 2016-11-15:
this https://github.com/todbot/electron-hid-toy also failing npm install on my machine at least

todbot, 2016-11-15:
That's very strange about electron-hid-toy. Here is a gist of the exact commands I used and their output:
https://gist.github.com/todbot/c0e6a385bba825dfe3ef395e7d556b05

I was on node v6.9.1, npm 3.10.8, Xcode 8.1, node-gyp and node-pre-gyp installed globally, and nothing in my npm config

ovsokolov, 2016-11-15:
> electron-hid-toy@0.1.2 postinstall /Users/olegsokolov/Documents/Navtool Projects/electron-hid-toy
> npm run electron-rebuild
> electron-hid-toy@0.1.2 electron-rebuild /Users/olegsokolov/Documents/Navtool Projects/electron-hid-toy
> electron-rebuild --pre-gyp-fix --force --module_dir . -w node-hid -e node_modules/electron-prebuilt
> node-hid@0.5.2 install /Users/olegsokolov/Documents/Navtool Projects/electron-hid-toy/node_modules/node-hid
> node-pre-gyp install --fallback-to-build

Here is log from my npm install

  CC(target) Release/obj.target/hidapi/hidapi/mac/hid.o
Failed to execute '/Users/olegsokolov/.nvm/versions/node/v6.9.1/bin/node /Users/olegsokolov/Documents/Navtool Projects/electron-hid-toy/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/Users/olegsokolov/Documents/Navtool Projects/electron-hid-toy/node_modules/node-hid/build/HID/Release/HID.node --module_name=HID --module_path=/Users/olegsokolov/Documents/Navtool Projects/electron-hid-toy/node_modules/node-hid/build/HID/Release' (1)

node-pre-gyp ERR! Tried to download(404): https://github.com/node-hid/node-hid/releases/download/0.5.2/HID-v0.5.2-electron-v1.4-darwin-x64.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for node-hid@0.5.2 and electron@1.4.6 (electron-v1.4 ABI) (falling back to source compile with node-gyp) 
gyp WARN download NVM_IOJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use IOJS_ORG_MIRROR
gyp WARN download NVM_IOJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use IOJS_ORG_MIRROR
gyp WARN download NVM_IOJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use IOJS_ORG_MIRROR
clang: error: no such file or directory: 'Projects/electron-hid-toy/node_modules/electron-rebuild/lib/headers/.node-gyp/iojs-1.4.6/include/node'
clang: error: no such file or directory: 'Projects/electron-hid-toy/node_modules/electron-rebuild/lib/headers/.node-gyp/iojs-1.4.6/src'
clang: error: no such file or directory: 'Projects/electron-hid-toy/node_modules/electron-rebuild/lib/headers/.node-gyp/iojs-1.4.6/deps/uv/include'
clang: error: no such file or directory: 'Projects/electron-hid-toy/node_modules/electron-rebuild/lib/headers/.node-gyp/iojs-1.4.6/deps/v8/include'
make: *** [Release/obj.target/hidapi/hidapi/mac/hid.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/Users/olegsokolov/Documents/Navtool Projects/electron-hid-toy/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Darwin 14.5.0
gyp ERR! command "/Users/olegsokolov/.nvm/versions/node/v6.9.1/bin/node" "/Users/olegsokolov/Documents/Navtool Projects/electron-hid-toy/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/Users/olegsokolov/Documents/Navtool Projects/electron-hid-toy/node_modules/node-hid/build/HID/Release/HID.node" "--module_name=HID" "--module_path=/Users/olegsokolov/Documents/Navtool Projects/electron-hid-toy/node_modules/node-hid/build/HID/Release"
gyp ERR! cwd /Users/olegsokolov/Documents/Navtool Projects/electron-hid-toy/node_modules/node-hid
gyp ERR! node -v v6.9.1
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/Users/olegsokolov/.nvm/versions/node/v6.9.1/bin/node /Users/olegsokolov/Documents/Navtool Projects/electron-hid-toy/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/Users/olegsokolov/Documents/Navtool Projects/electron-hid-toy/node_modules/node-hid/build/HID/Release/HID.node --module_name=HID --module_path=/Users/olegsokolov/Documents/Navtool Projects/electron-hid-toy/node_modules/node-hid/build/HID/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/Users/olegsokolov/Documents/Navtool Projects/electron-hid-toy/node_modules/node-hid/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:877:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
node-pre-gyp ERR! System Darwin 14.5.0
node-pre-gyp ERR! command "/Users/olegsokolov/.nvm/versions/node/v6.9.1/bin/node" "/Users/olegsokolov/Documents/Navtool Projects/electron-hid-toy/node_modules/node-hid/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /Users/olegsokolov/Documents/Navtool Projects/electron-hid-toy/node_modules/node-hid
node-pre-gyp ERR! node -v v6.9.1
node-pre-gyp ERR! node-pre-gyp -v v0.6.31
node-pre-gyp ERR! not ok 

npm ERR! Darwin 14.5.0
npm ERR! argv "/Users/olegsokolov/.nvm/versions/node/v6.9.1/bin/node" "/Users/olegsokolov/Documents/Navtool Projects/electron-hid-toy/node_modules/npm/bin/npm-cli.js" "rebuild" "node-hid" "--runtime=electron" "--target=1.4.6" "--arch=x64" "--update-binary"
npm ERR! node v6.9.1
npm ERR! npm  v3.10.9
npm ERR! code ELIFECYCLE
npm ERR! node-hid@0.5.2 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-hid@0.5.2 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the node-hid package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs node-hid
[... cut ...]
npm ERR! Tell the author that this fails on your system:
npm ERR!     npm run electron-rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs electron-hid-toy
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls electron-hid-toy
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /Users/olegsokolov/Documents/Navtool Projects/electron-hid-toy/npm-debug.log

ovsokolov, 2016-11-15:
my  Xcode older 7.1.
I sent you my log in email

todbot, 2016-11-15:
Is the directory you're building the project special in some way, like a file share or a FAT filesystem?  It looks like electron-rebuild can't find node-gyp's headers.

Another thing to try, install "node-gyp" and "node-pre-gyp" globally with:
`npm install node-gyp -g` and `npm install node-pre-gyp -g`

(also in the future, please create a gist and link to that. It's very hard to follow github issues with huge walls of text)

ovsokolov, 2016-11-15:
No nothing special about directory and "node-gyp" and "node-pre-gyp" are global

ovsokolov, 2016-11-15:
Ok looks like an issue caused by space in a folder name. One of the parent folders had space in it. Once removed everything works as expected. Thanks a lot!

todbot, 2016-11-15:
oh! wow, yeah that shouldn't be an issue but node-gyp has historically had issues with filesystems so it doesn't surprise me.  Thanks for figuring it out.

closed 2016-11-15 by todbot