node-hid/node-hid #168: Inconsistent usagePage and usage with rawHID teensy and raspbian

vincentdieltiens, 2016-11-16:
I'm testing my teensy under raspbian (on a raspberry pi 3). I've installed node-hid using  : 

> $ npm install https://github.com/node-hid/node-hid.git#master --driver=hidraw

Running multiples times, I get different values for usage and usagePage

> $ node test.js
> [ { vendorId: 5824,
>     productId: 1158,
>     path: '/dev/hidraw0',
>     serialNumber: '',
>     product: 'Teensyduino RawHID Device',
>     release: 258,
>     interface: 0,
>     **usagePage: 22016,**
>     **usage: 498** },
>   { vendorId: 5824,
>     productId: 1158,
>     path: '/dev/hidraw1',
>     serialNumber: '',
>     product: 'Teensyduino RawHID Device',
>     release: 258,
>     interface: 1 } ]

> $ node test.js
> [ { vendorId: 5824,
>     productId: 1158,
>     path: '/dev/hidraw0',
>     serialNumber: '',
>     product: 'Teensyduino RawHID Device',
>     release: 258,
>     interface: 0,
>     **usagePage: 43960**,
>     **usage: 636** },
>   { vendorId: 5824,
>     productId: 1158,
>     path: '/dev/hidraw1',
>     serialNumber: '',
>     product: 'Teensyduino RawHID Device',
>     release: 258,
>     interface: 1 } ]

Sometimes I've no usagePage nor usage :

> [ { vendorId: 5824,
>     productId: 1158,
>     path: '/dev/hidraw0',
>     serialNumber: '',
>     product: 'Teensyduino RawHID Device',
>     release: 258,
>     interface: 0 },
>   { vendorId: 5824,
>     productId: 1158,
>     path: '/dev/hidraw1',
>     serialNumber: '',
>     product: 'Teensyduino RawHID Device',
>     release: 258,
>     interface: 1 } ]

NB :  same result with root access using sudo

The used code is pretty simple : 
```javascript
evices = HID.devices();

var deviceInfo = devices.find( function(d) {
    return d.vendorId===0x16C0 
        && d.productId===0x0486 
        && d.usagePage===0xFFAB 
        && d.usage===0x200;
});
if (!deviceInfo) {
    console.log(devices);
    console.log("Could not find RawHID device in device list");
    process.exit(1);
}
var device = new HID.HID( deviceInfo.path );
// do something with device
```

todbot, 2016-11-16:
What is the "test.js" program you mention above?  (It doesn't look like the sample code you provide)

Do you see the same thing if you run just `node ./src/show-devices.js`?

todbot, 2016-11-16:
I just verified it with `showdevices.js`. It looks like hidapi's hidraw code isn't even populating the usagePage & usage fields.  Sigh.

vincentdieltiens, 2016-11-16:
Whoops, sorry. Problem with copy-paste : 

test.js : 
```javascript
var HID = require('node-hid');

var devices = HID.devices();

var deviceInfo = devices.find( function(d) {
    return d.vendorId===0x16C0 
        && d.productId===0x0486 
        && d.usagePage===0xFFAB 
        && d.usage===0x200;
});
if (!deviceInfo) {
    console.log(devices);
    console.log("Could not find RawHID device in device list");
    process.exit(1);
}
var device = new HID.HID( deviceInfo.path );
// do something with device
```

Same result with show-devices.js

Retry more than once with show-devices and sometimes usagePage and usage are populated (with wrong values), sometimes not.

todbot, 2016-11-16:
Don't bother. I just verified that hidapi's hidraw code contains nothing about the usagePage and usage fields.  We are at their mercy.  The (now outdated) pull request to add this was submitted several years ago but never merged:  https://github.com/signal11/hidapi/pull/6

vincentdieltiens, 2016-11-17:
My knowledge of hidapi (I'm pretty new to HID protocol, only two weeks that I have my teensy) is not sufficient to update the PR. Maybe one day... :)

I've updated my test.js so that he tries to get with usage and usagePage and in case of failure tries without the two properties. As the order of devices in linux seems to be consistent (see #165) , it works.

But it will be cleaner to use usagePage and usage on linux too :)

closed 2019-11-23 by todbot