node-hid/node-hid #169: blink1 path empty when connected to some hubs

gzphreak, 2016-11-23:
I recently upgrade to 0.5.2 and my blink1 is no longer found on the HUB its plugged into. It works when plugged directly into the computer as the path for the device is found. blink1control2 finds, and is able to control the blink1 just fine on said HUB. I can provide any other technical details if asked. Thanks :)

`➜  src node test-blink1.js 
blink(1) devices found: [ { vendorId: 10168,
    productId: 493,
    path: '',
    serialNumber: '200065C4',
    manufacturer: 'ThingM',
    product: 'blink(1) mk2',
    release: 2,
    interface: -1 } ] 

/Users/dan/work/slack-notifier/node_modules/node-hid/nodehid.js:28
	this._raw = new (Function.prototype.bind.apply(binding.HID,
	            ^

Error: cannot open device with vendor id 0x27b8 and product id 0x1ed
    at Error (native)
    at new HID (/Users/dan/work/slack-notifier/node_modules/node-hid/nodehid.js:28:14)
    at Object.<anonymous> (/Users/dan/work/slack-notifier/node_modules/node-hid/src/test-blink1.js:21:17)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.runMain (module.js:604:10)
    at run (bootstrap_node.js:394:7)
`

gzphreak, 2016-11-23:
I went back though the commit log and it looks like PR #163 is what broke it.

todbot, 2016-11-23:
Hi,
I just tried testing this and can't replicate your results.  The commands I used were:
```
mkdir node-hid-test
cd node-hid-test
npm init -y
npm install node-hid@0.5.2
node ./node_modules/node-hid/src/test-blink1.js
```

With a blink(1) plugged either through a hub or directly, everything works as expected.  I'm using node v6.9.1 on Mac OS X 10.11.6.

If you're still having issues, what is the output of `node ./node_modules/node-hid/src/show-devices.js`?

gzphreak, 2016-11-23:
I've bolded the blink1 in the list. It's actually on the last hub in a chain of hubs. It's a problem with `hidapi` and how they're creating the `path`. I rolled back the `hidapi` submodule to before they switched macOS to use IOKit. So I don't think it's a `node-hid` specific issue.

I'm using node 6.9.1 and macOS 10.12.1.

➜  src node show-devices.js 
devices: [ { vendorId: 1241,
    productId: 617,
    path: 'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/PEG1@1,1/IOPP/UPSB@0/IOPP/DSB1@3/IOPP/UPS0@0/IOPP/pci-bridge@0/IOPP/pci-bridge@0/IOPP/pci-bridge@3/IOPP/pci12d8,400f@0,2/AppleUSBEHCIPI7C9X440SL@40000000/PRT1@40100000/IOUSBHostDevice@40100000/AppleUSB20InternalHub@40100000/PRT2@40120000/USB Keyboard@40120000/IOUSBHostInterface@0/IOUSBHostHIDDevice@40120000,0',
    serialNumber: '',
    manufacturer: '',
    product: 'USB Keyboard',
    release: 4617,
    interface: -1 },
  { vendorId: 1452,
    productId: 610,
    path: 'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS12@14400000/Apple Internal Keyboard / Trackpad@14400000/Touchpad@1/AppleUSBMultitouchDriver@14400000,1',
    serialNumber: '',
    manufacturer: 'Apple Inc.',
    product: 'Apple Internal Keyboard / Trackpad',
    release: 549,
    interface: -1 },
  { vendorId: 1452,
    productId: 610,
    path: 'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS12@14400000/Apple Internal Keyboard / Trackpad@14400000/Apple Internal Keyboard@0/AppleUSBTCKeyboard@14400000,0',
    serialNumber: '',
    manufacturer: 'Apple Inc.',
    product: 'Apple Internal Keyboard / Trackpad',
    release: 549,
    interface: -1 },
  { vendorId: 3468,
    productId: 4,
    path: 'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/PEG1@1,1/IOPP/UPSB@0/IOPP/DSB1@3/IOPP/UPS0@0/IOPP/pci-bridge@0/IOPP/pci-bridge@0/IOPP/pci-bridge@3/IOPP/pci12d8,400f@0,2/AppleUSBEHCIPI7C9X440SL@40000000/PRT1@40100000/IOUSBHostDevice@40100000/AppleUSB20InternalHub@40100000/PRT1@40110000/IOUSBHostDevice@40110000/AppleUSB20Hub@40110000/AppleUSB20HubPort@40111000/Schiit USB Audio Gen 2@40111000/IOUSBHostInterface@2/IOUSBHostHIDDevice@40111000,2',
    serialNumber: '',
    manufacturer: 'Schiit Audio',
    product: 'Schiit USB Audio Gen 2',
    release: 258,
    interface: -1 },
  { vendorId: 5426,
    productId: 76,
    path: 'IOService:/IOResources/AppleUSBHostResources/AppleUSBLegacyRoot/AppleUSBEHCI@40000000/Razer Diamondback@40132100/AppleUSBInterface@1/RzHidInterface',
    serialNumber: '',
    manufacturer: 'Razer',
    product: 'Razer Diamondback',
    release: 512,
    interface: -1 },
  { vendorId: 1452,
    productId: 37415,
    path: 'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/PEG1@1,1/IOPP/UPSB@0/IOPP/DSB1@3/IOPP/UPS0@0/IOPP/pci-bridge@0/IOPP/pci-bridge@0/IOPP/pci-bridge@3/IOPP/pci12d8,400f@0,2/AppleUSBEHCIPI7C9X440SL@40000000/PRT1@40100000/IOUSBHostDevice@40100000/AppleUSB20InternalHub@40100000/PRT7@40170000/Apple Thunderbolt Display@40170000/IOUSBHostInterface@0/AppleUSBDisplays@40170000,0',
    serialNumber: '17101952',
    manufacturer: 'Apple Inc.',
    product: 'Apple Thunderbolt Display',
    release: 312,
    interface: -1 },
  { vendorId: 1241,
    productId: 617,
    path: 'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/PEG1@1,1/IOPP/UPSB@0/IOPP/DSB1@3/IOPP/UPS0@0/IOPP/pci-bridge@0/IOPP/pci-bridge@0/IOPP/pci-bridge@3/IOPP/pci12d8,400f@0,2/AppleUSBEHCIPI7C9X440SL@40000000/PRT1@40100000/IOUSBHostDevice@40100000/AppleUSB20InternalHub@40100000/PRT2@40120000/USB Keyboard@40120000/IOUSBHostInterface@1/IOUSBHostHIDDevice@40120000,1',
    serialNumber: '',
    manufacturer: '',
    product: 'USB Keyboard',
    release: 4617,
    interface: -1 },
  { vendorId: 1452,
    productId: 610,
    path: 'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS12@14400000/Apple Internal Keyboard / Trackpad@14400000/Touchpad@2/AppleUSBTCButtons@14400000,2',
    serialNumber: '',
    manufacturer: 'Apple Inc.',
    product: 'Apple Internal Keyboard / Trackpad',
    release: 549,
    interface: -1 },
  { vendorId: 5426,
    productId: 76,
    path: 'IOService:/IOResources/AppleUSBHostResources/AppleUSBLegacyRoot/AppleUSBEHCI@40000000/Razer Diamondback@40132100/AppleUSBInterface@2/RzHidInterface',
    serialNumber: '',
    manufacturer: 'Razer',
    product: 'Razer Diamondback',
    release: 512,
    interface: -1 },
  **{ vendorId: 10168,
    productId: 493,
    path: '',
    serialNumber: '200065C4',
    manufacturer: 'ThingM',
    product: 'blink(1) mk2',
    release: 2,
    interface: -1 },**
  { vendorId: 5426,
    productId: 76,
    path: 'IOService:/IOResources/AppleUSBHostResources/AppleUSBLegacyRoot/AppleUSBEHCI@40000000/Razer Diamondback@40132100/AppleUSBInterface@0/RzHidInterface',
    serialNumber: '',
    manufacturer: 'Razer',
    product: 'Razer Diamondback',
    release: 512,
    interface: -1 } ]

todbot, 2016-11-23:
Hmmm.  It may be something as dumb as the path string isn't allocated long enough.

How many hubs down is the blink(1) module? 

For instance, what does System Information say?  Here's an example from my Macbook Pro
[image]

gzphreak, 2016-11-23:
Perhaps. I did not start digging into `hidapi` other than to see what they changed recently.

[image]

todbot, 2016-11-23:
Looks like you have 4 hubs in between the host and the blink(1) (Some are built-in or if you have a >4 port USB hub, that can actually count as two).  This is close to the [maximum of 5 hubs](https://en.wikipedia.org/wiki/USB#System_design). (for @todbot: [also see this by pluggable](http://support.plugable.com/plugable/topics/daisy_chaining_usb3_hub7_81x_seven_port_usb_3_0_hub)).

From looking at commit https://github.com/signal11/hidapi/commit/59a076705b1bc5316fb4513fc28033b662a23675, hidapi is using the OS X standard [IORegistryEntryGetPath()](https://developer.apple.com/reference/iokit/1514229-ioregistryentrygetpath?language=objc) and passing the recommended `io_string_t` type for path. And `io_string_t` is [defined as 512 characters long](https://github.com/darwin-on-arm/xnu/blob/master/osfmk/device/device_types.h).  

On my system, when I daisychain hubs, I can only get up to 3 external hubs (If I try any more I get "AppleUSBHostPort::setPortLocation: Hub count exceeded" errors in dmesg) and the resulting path is 460 characters long.  I'm not sure of the naming of the elements in `IORegistryEntryGetPath` but it seems like it could get longer than 512.  So this almost seems like it could be a bug in Mac OS X.

As a workaround, I would say reduce the number of USB hubs used with hidapi / node-hid projects.
This isn't a really node-hid issue but rather a hidapi one. I think we should open up an issue there.  I'm still doing some tests to see if I can characterize it a bit more before I open up a hidapi issue.

I'll leave this issue open for now as a reminder to me.

todbot, 2016-11-23:
[Notes for @todbot] Ahah, via swapping hubs around I was able to get a daisychain hub configuration that allows `blink1-tool` to work but node-hid & hidapi return blank path on:
[image]

gzphreak, 2016-11-23:
Indeed. It would appear that one of the hubs I have has two built into it.

mcuee, 2023-05-04:
This has been fixed in upstream HIDAPI project. So this can be closed.

Julusian, 2023-05-05:
@mcuee I think that hidapi fix hasnt made its way into this project yet

mcuee, 2023-05-05:
I see.

@todbot 
Maybe you want to update the HIDAPI latets release version 0.13.1. Thanks.