node-hid/node-hid #170: A dynamic link library

koeshiro, 2016-11-23:
Hello. Trying use you node-hid modul, but all the time get this Error: (full loh)

>  App threw an error during load
> Error: A dynamic link library (DLL) initialization routine failed.
> \\?\C:\Users\koesh\Desktop\test\electron\node_modules\node-hid\build\HID\Release\HID.node
>     at Error (native)
>     at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:20)
>     at Object.Module._extensions..node (module.js:583:18)
>     at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:173:20)
>     at Module.load (module.js:473:32)
>     at tryModuleLoad (module.js:432:12)
>     at Function.Module._load (module.js:424:3)
>     at Module.require (module.js:483:17)
>     at require (internal/module.js:20:19)
>     at Object.<anonymous> (C:\Users\koesh\Desktop\test\electron\node_modules\node-hid\nodehid.js:9:15)
Node js v7.2.0
electron v1.4.8
Windows 10

todbot, 2016-11-23:
You need to run electron-rebuild and rebuild node-hid from source for electron.
See: https://github.com/node-hid/node-hid#using-node-hid-in-electron-projects
and see: https://github.com/node-hid/node-hid#compiling-from-source

todbot, 2017-01-29:
I believe this has been fixed in the latest release.

closed 2017-01-29 by todbot

eugeneshemonaev, 2017-07-25:
Hello. have the same issue. this solution 

> See: https://github.com/node-hid/node-hid#using-node-hid-in-electron-projects
> and see: https://github.com/node-hid/node-hid#compiling-from-source

doesnt help. what  else can I do?

todbot, 2017-07-25:
Hi,
Please give a detailed log as a gist showing the errors you're receiving. You must have a full and working compiler setup on your system to use node-hid with Electron.