node-hid/node-hid #172: Read with timeout without blocking JS thread.

Cierpliwy, 2016-11-28:
By looking at API and testing it I have a problem to get asynchronous read with a timeout.
There are three functions:

- `read(callback)` - this call doesn't block UI but don't allow to cancel it (and I would like to do it after certain period).
- `readSync()` - blocks thread so it's not useful.
- `readTimeout(timeout)` - blocks thread so it's not useful.

I would like to use method such as `read(callback, timeout)` or have an ability to cancel such `read` in some way. Is there any technique to achieve that?

todbot, 2017-01-29:
To close pending node-hid EventEmitters, you need to close the device and re-open.

limpkin, 2017-03-30:
I second this need for feature.
On mooltipass we're getting random "cannot write to HID device" events, which trigger a device disconnect. However, because the read() call is ongoing we are not able to connect to the device ever again.

Julusian, 2025-06-03:
This is possible with the async api in v3.0+

closed 2025-06-03 by Julusian